RoboStack/ros-noetic #3: Fix sparse_bundle_adjustment build (required for slam_toolbox)

Tobias-Fischer, 2020-10-11:
Cannot `find_package(BLAS)` at the moment: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=702&view=logs&jobId=96baafb5-a962-5442-c60b-eeb71310fecc&j=96baafb5-a962-5442-c60b-eeb71310fecc&t=b23dd469-b816-56d1-7268-91c4c260ef1f

Weirdly, BLAS can be found okay in ceres-solver which uses the same cmake script to find SuiteSparse: https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=210040&view=logs&jobId=a70f640f-cc53-5cd3-6cdc-236a1aa90802&j=a70f640f-cc53-5cd3-6cdc-236a1aa90802&t=f5d15007-a01c-5ad8-c9ce-4d519d3b275f

Differences are gcc vs MSVS & blas mkl vs blas netlib

Tobias-Fischer, 2020-10-25:
Hi @wolfv,
I am pretty convinced that this issue is due to a bug in `boa`. 

I am pinning down blas+lapack exactly now on Windows (https://github.com/RoboStack/ros-noetic/blob/master/conda-forge.yaml#L425-L432) and yet the mkl version is pulled in (https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=790&view=logs&j=96baafb5-a962-5442-c60b-eeb71310fecc&t=b23dd469-b816-56d1-7268-91c4c260ef1f).

Unfortunately I don't have a Windows machine available for testing, so I can't provide a minimum example. It would be great if you could look into this.

Tobi

Tobias-Fischer, 2021-01-14:
Hi @wolfv - this still happens with the most recent boa.
The variant clearly shows that netlib should be used:
```
+------------------------------------+
| Package         | Variant versions |
|-----------------+------------------|
| target_platform | win-64           |
| libblas         | 3.8 *netlib      |
| liblapack       | 3.8 *netlib      |
| python          | 3.8.* *_cpython  |
| suitesparse     | 5.4              |
| cxx_compiler    | vs2017           |
| c_compiler      | vs2017           |
+------------------------------------+
```

Also the recipe pins netlib:
```
...
  host:
    - libblas 3.8 *netlib
    - liblapack 3.8.0 *netlib
 ...
```

And yet mkl is installed. Is there anything I can help to debug this? Should I open an issue in boa?

wolfv, 2021-03-14:
@Tobias-Fischer indeed, this might just be a bug in mamba / libsolv: https://github.com/mamba-org/mamba/issues/725

The fix is here: https://github.com/openSUSE/libsolv/issues/435

Unfortunately we don't have a release yet, but I think I can apply that patch on the libsolv-feedstock on Monday.

Sorry that it took so long to get this resolved :/

Tobias-Fischer, 2021-03-14:
I just created a PR in the libsolv feedstock :)

closed 2021-03-15 by Tobias-Fischer

wolfv, 2021-03-15:
I'm really glad this worked!