RoboStack/ros-noetic #9: Add run_constraints for osx

Tobias-Fischer, 2020-11-01:
See https://conda-forge.org/docs/maintainer/knowledge_base.html#requiring-newer-macos-sdks - we require osx 10.14 already (see https://github.com/RoboStack/ros-noetic/blob/master/conda_build_config.yaml#L632-L635). I guess we need to alter vinca to add this.

/cc @traversaro @wolfv

Tobias-Fischer, 2020-11-30:
I had a play around with this by changing vinca as follows:
```
diff --git a/vinca/main.py b/vinca/main.py
index d71146e..226b299 100644
--- a/vinca/main.py
+++ b/vinca/main.py
@@ -170,7 +170,9 @@ def generate_output(pkg_shortname, vinca_conf, distro, version):
                 "cmake",
             ],
             "host": [],
-            "run": [],
+            "run": [
+                'sel(osx and x86_64): __osx >={{ MACOSX_DEPLOYMENT_TARGET|default(\'10.14\') }}'
+            ],
         },
         "build": {"script": ""},
     }

```

This generates recipes with the following run dep:
```
  run:
    - python
    - "sel(osx and x86_64): __osx >={{ MACOSX_DEPLOYMENT_TARGET|default(10.14) }}"
```

However boa doesn't like this:
```
Traceback (most recent call last):
  File "/home/tobias/anaconda3/bin/boa", line 11, in <module>
    load_entry_point('boa', 'console_scripts', 'boa')()
  File "/home/tobias/robot/boa/boa/cli/boa.py", line 94, in main
    run_build(args)
  File "/home/tobias/robot/boa/boa/core/run_build.py", line 429, in run_build
    build_recipe(args, recipe["recipe_file"], cbc, config)
  File "/home/tobias/robot/boa/boa/core/run_build.py", line 362, in build_recipe
    o.finalize_solve(sorted_outputs)
  File "/home/tobias/robot/boa/boa/core/recipe_output.py", line 572, in finalize_solve
    self._solve_env("run", all_outputs)
  File "/home/tobias/robot/boa/boa/core/recipe_output.py", line 519, in _solve_env
    t = solver.solve(specs, [pkg_cache])
  File "/home/tobias/robot/boa/boa/core/solver.py", line 136, in solve
    api_solver.add_jobs(_specs, mamba_api.SOLVER_INSTALL)
RuntimeError: Invalid spec, no package name found:  __osx >=10.14
```

When I manually remove the quotes around the `sel(..)` line it works. However I can't get ruamel to not insert those quotes. Is there a way to adapt boa to properly handle those lines containing a quote, @wolfv?

wolfv, 2020-11-30:
ha, yeah, the trick is that `sel(xxx): blabla` is actually a dictionary.

so adding `run: [{'sel(osx and x86_64)': '__osx >={{ MACOSX_DEPL...}}'}]` should work!

Tobias-Fischer, 2020-11-30:
Fixed with https://github.com/RoboStack/vinca/commit/059c294e24db41e8e5980de789d9b8921451f2ea

Thanks for the hint @wolfv

Tobias-Fischer, 2020-12-04:
This doesn't work quite yet - any idea how to resolve this @wolfv? https://github.com/RoboStack/vinca/commit/d8b13f25f85b6e7799cb7021b50103c6c9feb028

closed 2021-03-06 by wolfv