RoboStack/ros-noetic #10: Patch catkin_tools

Tobias-Fischer, 2020-11-05:
Last time I tried, 'catkin build' failed in a catkin_ws

PeterMitrano, 2020-11-22:
I tried this, using the catkin_tools in the conda-forge channel, and it fails with 

```
Traceback (most recent call last):
  File "/home/pmitrano/miniconda3/envs/noetic-merrrt/lib/python3.6/site-packages/catkin_tools/verbs/catkin_build/build.py", line 552, in build_isolated_workspace
    continue_without_deps=False))
  File "/home/pmitrano/miniconda3/envs/noetic-merrrt/lib/python3.6/site-packages/catkin_tools/execution/executor.py", line 377, in run_until_complete
    return loop.run_until_complete(coroutine)
  File "/home/pmitrano/miniconda3/envs/noetic-merrrt/lib/python3.6/asyncio/base_events.py", line 488, in run_until_complete
    return future.result()
  File "/home/pmitrano/miniconda3/envs/noetic-merrrt/lib/python3.6/site-packages/catkin_tools/execution/executor.py", line 278, in execute_jobs
    done_job_fs, active_job_fs = yield asyncio.From(asyncio.wait(
AttributeError: module 'asyncio' has no attribute 'From'
```

So we should create python 3.7 and 3.8 versions I believe.

Tobias-Fischer, 2020-11-23:
The issue is that catkin tools for python 3 has not yet been properly released :(. See https://github.com/catkin/catkin_tools/issues/594
I'm on holidays for a few days, I'll try and patch it after that. Or, feel free to open a PR if you want to contribute @PeterMitrano

Tobias-Fischer, 2020-11-30:
That should be fixed by https://github.com/conda-forge/catkin_tools-feedstock/pull/15

@PeterMitrano can you please let me know if you still have issues after upgrading catkin tools?

closed 2020-11-30 by Tobias-Fischer

PeterMitrano, 2020-11-30:
Not sure if this is related, but the following env.yml is incompatible:

```
channels:
- defaults
- conda-forge
- robostack
dependencies:
- ros-noetic-desktop-full
- catkin_tools
name: test_catkin_tools_env
```

Trying to create this environment gives
```
peter@armstorm:~/conda_ws2$ conda  env create -f env.yml 
Collecting package metadata (repodata.json): done
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                             
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                             

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python_abi conflicts for:
ros-noetic-desktop-full -> python_abi=3.6
ros-noetic-desktop-full -> python=3.6 -> python_abi==3.6[build=*_pypy36_pp73]

Package libstdcxx-ng conflicts for:
ros-noetic-desktop-full -> libstdcxx-ng[version='>=7.5.0']
ros-noetic-desktop-full -> python=3.6 -> libstdcxx-ng[version='>=4.9|>=7.2.0|>=7.3.0|>=9.3.0']

Package pypy3.6 conflicts for:
ros-noetic-desktop-full -> python=3.6 -> pypy3.6[version='7.3.*|7.3.0.*|7.3.1.*|7.3.2.*']
catkin_tools -> pypy3.6[version='>=7.3.2']
catkin_tools -> psutil -> pypy3.6[version='7.3.*|7.3.0.*|7.3.1.*|7.3.2.*|>=7.3.1']

Package libgcc-ng conflicts for:
ros-noetic-desktop-full -> python=3.6 -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0|>=9.3.0']
ros-noetic-desktop-full -> libgcc-ng[version='>=7.5.0']
Note that strict channel priority may have removed packages required for satisfiability.
```

PeterMitrano, 2020-11-30:
Yes somehow In an existing environment I with `ros-noetic-desktop-full` was able to install it... so that's confusing. In that context, I _think_ this is fixed, but I can't recreate it from scratch so I'm not sure

PeterMitrano, 2020-12-02:
Turns out it's a channel priority issue -- reversing the order of the channels makes this work.

```
channels:
- robostack
- conda-forge
- defaults
dependencies:
- ros-noetic-desktop-full
- catkin_tools
name: test_catkin_tools_env
```

Tobias-Fischer, 2020-12-02:
Ahh yes, sorry, I should have caught that. This priority order should happen though following the instructions?!?

Glad to hear it's working now :)

ImanHosseini, 2021-04-04:
when I try installing ros-noetic-desktop, on a newly created (empty) environment I get: 
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: -
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError:
```

Tobias-Fischer, 2021-04-04:
Can you please try mamba instead of conda and try again?

Tobias-Fischer, 2021-04-04:
Also please include the command that you used.

ImanHosseini, 2021-04-04:
I installed ros-noetic-jackal-gzebo and it worked (conda), so I stopped trying to install ros-noetic-desktop-full. 
For the record, I was doing: conda install -c robostack ros-noetic-desktop-full