RoboStack/ros-noetic #12: roscore error on Windows

traversaro, 2020-11-26:
If I install rviz with: 
~~~
conda install ros-noetic-rviz
~~~

Then if I try `roscore`, it fails with error: 
[image]

traversaro, 2020-11-26:
Env info:

~~~
(ros-tests) C:\Users\STraversaro>conda info

     active environment : ros-tests
    active env location : C:\Users\STraversaro\Miniforge3\envs\ros-tests
            shell level : 1
       user config file : C:\Users\STraversaro\.condarc
 populated config files : C:\Users\STraversaro\Miniforge3\.condarc
                          C:\Users\STraversaro\.condarc
          conda version : 4.9.2
    conda-build version : not installed
         python version : 3.8.6.final.0
       virtual packages : __cuda=10.2=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\STraversaro\Miniforge3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://conda.anaconda.org/msys2/win-64
                          https://conda.anaconda.org/msys2/noarch
          package cache : C:\Users\STraversaro\Miniforge3\pkgs
                          C:\Users\STraversaro\.conda\pkgs
                          C:\Users\STraversaro\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\STraversaro\Miniforge3\envs
                          C:\Users\STraversaro\.conda\envs
                          C:\Users\STraversaro\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.25.0 CPython/3.8.6 Windows/10 Windows/10.0.19041
          administrator : False
             netrc file : None
           offline mode : False

(ros-tests) C:\Users\STraversaro>conda list
# packages in environment at C:\Users\STraversaro\Miniforge3\envs\ros-tests:
#
# Name                    Version                   Build  Channel
apr                       1.6.5                ha89961a_2    conda-forge
assimp                    5.0.1                h7d0832f_4    conda-forge
bcrypt                    3.2.0            py36h779f372_1    conda-forge
boost                     1.72.0           py36h9956518_1    conda-forge
boost-cpp                 1.72.0               h54f0996_4    conda-forge
bzip2                     1.0.8                he774522_3    conda-forge
ca-certificates           2020.11.8            h5b45459_0    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.11.8        py36ha15d459_0    conda-forge
cffi                      1.14.4           py36he58ceb7_0    conda-forge
cmake                     3.19.1               h39d44d4_0    conda-forge
console_bridge            1.0.1                h7ef1ec2_0    conda-forge
cryptography              3.2.1            py36he58ceb7_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.16             py36hd36e781_2    conda-forge
eigen                     3.3.8                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.2.9                h33f27b4_2    conda-forge
freeimage                 3.18.0               h33f27b4_0    conda-forge
freetype                  2.10.4               h546665d_0    conda-forge
gettext                   0.19.8.1          hfbb10ce_1004    conda-forge
gmock                     1.10.0               h57928b3_4    conda-forge
gtest                     1.10.0               h5362a0b_4    conda-forge
icu                       67.1                 h33f27b4_0    conda-forge
ilmbase                   2.5.3                h7bd577a_0    conda-forge
[... 125 lines cut ...]
rospkg                    1.2.9              pyhd3deb0d_0    conda-forge
ruby                      2.7.2                h8b1b97a_2    conda-forge
sdl2                      2.0.12               h0e60522_1    conda-forge
setuptools                49.6.0           py36hd36e781_2    conda-forge
sip                       4.19.20          py36h6538335_0    conda-forge
six                       1.15.0             pyh9f0ad1d_0    conda-forge
sqlite                    3.33.0               he774522_1    conda-forge
tinyxml                   2.6.2                h7ef1ec2_1    conda-forge
tinyxml2                  8.0.0                h33f27b4_1    conda-forge
tk                        8.6.10               he774522_1    conda-forge
urdfdom                   2.3.3                h7ef1ec2_0    conda-forge
urdfdom_headers           1.0.5                h7ef1ec2_2    conda-forge
vc                        14.1                 h869be7e_1    conda-forge
vs2015_runtime            14.16.27012          h30e32a0_2    conda-forge
wheel                     0.35.1             pyh9f0ad1d_0    conda-forge
wincertstore              0.2             py36h9f0ad1d_1005    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.5                h1f3a1b7_2    conda-forge
zziplib                   0.13.69              ha81a0f5_1    conda-forge

(ros-tests) C:\Users\STraversaro>
~~~

Tobias-Fischer, 2020-11-27:
Fixed by https://github.com/conda-forge/log4cxx-feedstock/pull/6

That should all work again after a rebuild 👍

Tobias-Fischer, 2020-11-30:
Hey @traversaro - could you please test whether this works fine now? Closing in the meantime, please reopen if it's still an issue.

traversaro, 2020-11-30:
I may be doing something wrong, but as far as I can see there are no new ros-noetic binaries in https://anaconda.org/robostack/repo , so I think the problem is still there. Perhaps it is necessary to comment out the `skip_existing` json parameter ( https://github.com/RoboStack/ros-noetic/blob/master/vinca_linux_64.yaml#L22 ) to force a mass rebuild, similar to what was done in https://github.com/RoboStack/ros-melodic by @wolfv ?

Tobias-Fischer, 2020-11-30:
Ah yes I forgot that there's a new version as well. Need to wait for a rebuild indeed, sorry for the confusion..

Tobias-Fischer, 2020-12-15:
Needs a new version of conda-build with rebuilt sip & pyqt, see conda-forge/conda-forge.github.io#1066 conda/conda-build#4126

Tobias-Fischer, 2021-01-06:
This should be fixed with the recent rebuild. Could you please check @traversaro?

closed 2021-01-06 by Tobias-Fischer