RoboStack/ros-noetic #14: Testing contributing instructions

PeterMitrano, 2020-11-30:
I'm following along the new instructions and thought we could use this to track issues:

1. `conda update conda -c conda-forge` does not work because conda is not installed in the brand new conda env. I think we can use `conda install ....` here instead?
2. After running vinca, I don't see a recipies folder anywhere. My ros-noetic folder looks like this:

```
├── activate.bat
├── activate.sh
├── bld_ament_cmake.bat
├── bld_ament_python.bat
├── bld_catkin.bat
├── bld_catkin_merge.bat
├── bld_colcon_merge.bat
├── build_ament_cmake.sh
├── build_ament_python.sh
├── build_catkin.sh
├── conda_build_config.yaml
├── conda-forge.yaml
├── CONTRIBUTING.md
├── deactivate.bat
├── deactivate.sh
├── packages-ignore.yaml
├── patch
│   ├── patch_sipconfig.py
│   ├── ros-noetic-actionlib-tutorials.patch
│   ├── ros-noetic-catkin.patch
... lots more ...
│   ├── ros-noetic-turtle-actionlib.patch
│   └── ros-noetic-urg-c.osx.patch
├── README.md
├── vinca_linux_64.yaml
├── vinca_linux_aarch64.yaml
├── vinca_osx.yaml
├── vinca_win.yaml
└── vinca.yaml -> vinca_linux_64.yaml
```

Tobias-Fischer, 2020-11-30:
Great, thanks for the feedback! Can you please check the updated instructions (which also mention why there is probably no recipes folder)?

PeterMitrano, 2020-12-01:
The first issue is fixed, and I see your changes and let me see if I understa.d Because pybind11-catkin just got added to conda (https://anaconda.org/RoboStack/ros-noetic-pybind11-catkin), so there are no "new" packages and therefore no recipes generated? If so, shouldn't there be a way to "regenerate" a package? I'm looking for a way to test that my PR will work on your build farm, so normally there won't be an existing version. Or what if I need to update a package? Thanks!

PeterMitrano, 2020-12-01:
btw I tested with another package, got a little further, but some then hit this:
```
Adding files for ros-noetic-cnpy

- bin/mat2npz
- bin/npy2mat
- bin/npz2mat
- include/cnpy/cnpy.h
- lib/cmake/cnpy/cnpy-config.cmake
- lib/cmake/cnpy/cnpy-targets-release.cmake
- lib/cmake/cnpy/cnpy-targets.cmake
- lib/libcnpy.so

INFO :: Time taken to mark (prefix)
        0 replacements in 0 files was 0.06 seconds
Files containing CONDA_PREFIX
-----------------------------
lib/cmake/cnpy/cnpy-targets.cmake (text): Patching
Traceback (most recent call last):
  File "/home/peter/miniconda3/envs/robostackenv/bin/boa", line 11, in <module>
    sys.exit(main())
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/cli/boa.py", line 94, in main
    run_build(args)
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/core/run_build.py", line 429, in run_build
    build_recipe(args, recipe["recipe_file"], cbc, config)
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/core/run_build.py", line 393, in build_recipe
    final_outputs = build(meta, None, allow_interactive=args.interactive)
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/core/build.py", line 616, in build
    final_outputs = bundle_conda(
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/core/build.py", line 260, in bundle_conda
    output["checksums"] = create_info_files(
  File "/home/peter/miniconda3/envs/robostackenv/lib/python3.8/site-packages/boa/core/build.py", line 127, in create_info_files
    source.git_info(
TypeError: git_info() missing 1 required positional argument: 'build_prefix'
```

Tobias-Fischer, 2020-12-01:
Ahhh that's a bug in boa which has not yet been released (/cc @wolfv - it would be great if you could release a new version).

For now, could you manually apply the changes in https://github.com/mamba-org/boa/commit/87e6749497f2be5b83ae13d59acb48c901a6b045?

Tobias-Fischer, 2020-12-01:
Re your other question - please see https://github.com/RoboStack/ros-noetic/commit/0d36feebd60e0fb602f004e3d8d05df3728be14c

PeterMitrano, 2020-12-01:
I just tried using https://github.com/mamba-org/boa/commit/87e6749497f2be5b83ae13d59acb48c901a6b045 and it seems to work. This was for a new package, `cnpy` which depends on zlib. I guess it succeeded -- output is attached
[boa_build_log.txt](https://github.com/RoboStack/ros-noetic/files/5625922/boa_build_log.txt)

What would it look like if it "failed" ?

wolfv, 2020-12-01:
Yes, that looks like a success! You can install that package from your local channel (mamba or conda install ... -c local)

wolfv, 2020-12-01:
The package file is stored in miniconda/conda-bld/linux-64

closed 2020-12-01 by Tobias-Fischer