RoboStack/ros-noetic #19: Patch pybind11_catkin

Tobias-Fischer, 2020-12-15:
Rather than downloading pybind11 as an external dependency as done upstream (https://github.com/ipab-slmc/pybind11_catkin/blob/master/CMakeLists.txt), we should use pybind11 shipped in conda.

@PeterMitrano would you be happy to create a patch for this?

However, looking at the project, I am not even sure whether we need this package in conda. All it does is copying the pybind stuff into the devel space. But in conda pybind11 is readily available and this step should not be required. Any thoughts @traversaro @wolfv?

PeterMitrano, 2020-12-16:
FWIW I started with that package because it was the one I was familiar with, if it doesn't add anything over pybind11 we can just drop it. What would the patch be for?

One advantage of pybind11-catkin can be specified in the package.xml, and therefore installed via rosdep. However that doesn't even work for conda, so that point is moot.

Tobias-Fischer, 2020-12-16:
Agreed, I think we can drop it. We can map pybind11-catkin to pybind11 in the conda-forge.yaml. Any thoughts @wolfv?

traversaro, 2020-12-16:
> However, looking at the project, I am not even sure whether we need this package in conda. All it does is copying the pybind stuff into the devel space. But in conda pybind11 is readily available and this step should not be required. Any thoughts @traversaro @wolfv?

I think that is case that will repeat again. ROS/ROS2 .deb packages are basically a distribution of software on the top of another set of distributions (Ubuntu/Debian) and they need sometimes to vendor packages that otherwise are packaged in the "base" distro to have a recent version of a given dependency. With conda-forge, the latest dependency should be available (or can be made available) directly in conda-forge, so I totally agree that for these vendored packages it make sense to just map them to the corresponding conda-forge package. 

A similar issue is there for packages that do not have any ROS-specific dependency, and that the authors package in the build farm just because it is an easy and familiar way to easily get .deb packages, such as https://github.com/ethz-adrl/ifopt . For those we can package them in RoboStack, but I imagine that in the future someone could be interested in package them in conda-forge directly.

PeterMitrano, 2020-12-16:
I like the idea of just mapping it -- if it truly provides the same thing, then it would be nice if `conda install ros-noetic-pybind11-catkin` worked, the user doesn't need to know what's happening

wolfv, 2020-12-16:
One other issue is that the catkin package installs some automatically generated cmake files to find pybind11_catkin etc.

We have the same problem with the libyaml vendor and other deps in ROS2 and I'm not yet sure what the right solution is since it might involve quite a bit of patching. Or we find a good solution to fix the vendor packages in "shim" packages...

Tobias-Fischer, 2021-04-26:
See https://github.com/RoboStack/ros-noetic/issues/20 for the "meta-issue"

closed 2021-04-26 by Tobias-Fischer