RoboStack/ros-noetic #20: Track packages that are released in rosdistro but are/should be in conda-forge

traversaro, 2020-12-20:
The ROS Debian binary packages that are released via [Bloom](http://wiki.ros.org/bloom) are essentially a distribution of software that builds on top of another distribution of software (the base Debian/Ubuntu repositories). This relation is similar to the one that in the conda world exists between conda-forge and bioconda channels, and in a sense is the same relation that we would like to create between conda-forge and robostack. 

Over the time some packages that we not ROS-specific (i.e. they did not depend on the ROS communication middleware) were released with Bloom for several reasons. The major reason is that it was much faster and easier to get a package published via Bloom rather then by going through the actual Debian/Ubuntu packaging process, while in some other cases a package was indeed available in Debian/Ubuntu repos, but the version available in the repos was not recent enough, and it was not possible to update it due to Debian policies for updates in already released distros. 

In the case of robostack, given that getting new packages in conda-forge is relatively easy and fast, and the same holds for updating the version of existing packages, I think it make sense to avoid to re-build packages that are or should be in conda-forge even if they are release in bloom/rosdistro, but rather we should have some systematic way to skip them and install the conda-forge version instead. I am not sure what is the proper way of doing so, but for the time being I opened this issue to track the occurrences of this pattern that I spot. 

| Package repo       |  It is already in conda-forge? | rosdistro package name | Does the ROS wrapper also adds some files? |   Related issue(s) | 
|:-------------------:|:--------------------------------:|:---------------------------:|:----------------------:|:------------:|
| https://github.com/pybind/pybind11 | https://github.com/conda-forge/pybind11-feedstock | `pybind11_catkin` in ROS1,  `pybind11_vendor` in ROS2 | Yes | https://github.com/RoboStack/ros-noetic/issues/19 | 
| https://github.com/OctoMap/octomap | https://github.com/conda-forge/octomap-feedstock | `octomap` | No | https://github.com/ros/rosdistro/issues/26527  | 
| https://github.com/flexible-collision-library/fcl |  https://github.com/conda-forge/fcl-feedstock | `fcl` | No | https://github.com/ros/rosdistro/pull/27789  | 
| https://github.com/flexible-collision-library/fcl |  https://github.com/conda-forge/fcl-feedstock | `fcl` | No | https://github.com/ros/rosdistro/pull/27789  | 
| https://github.com/IntelRealSense/librealsense | No | `librealsense` | No | https://github.com/robotology/robotology-superbuild/issues/564 |
| https://github.com/borglab/gtsam | No | `gtsam` | No | https://github.com/ros/rosdistro/pull/23198 |
| https://github.com/ros/console_bridge | https://github.com/conda-forge/console_bridge-feedstock | `console_bridge_vendor` | |

traversaro, 2020-12-27:
Related ROS Discourse discussion: https://discourse.ros.org/t/release-package-requiring-custom-version-of-system-dependency/18200 .

traversaro, 2021-01-02:
An interesting case of this packages are the one that do not depend on anything ROS-specific and so can go in conda-forge, but that use the ROS-like strategy of a single repo with multiple directories that contain different interdependent CMake or Python packages (see for example https://github.com/ethz-adrl/ifopt or tesseract, https://github.com/ros-industrial-consortium/tesseract  https://github.com/ros-industrial-consortium/tesseract/issues/419). 

An example of this kind that is already in conda-forge is orocos-kdl (see https://github.com/conda-forge/orocos-kdl-feedstock/blob/master/recipe/meta.yaml) that indeed is using a multi-output recipe structure for which each "package" results in a different output. I don't know if vinca can help in this case, even if the recipe then goes in a usual conda-forge feedstock.

traversaro, 2021-03-29:
I added eigenpy to the table @wolfv .

Tobias-Fischer, 2021-04-07:
There's also `fcl_catkin` which is used by exotica, and should pull in conda-forge's fcl instead of building it.

/cc @wxmerkt

wxmerkt, 2021-04-07:
Regarding fcl_catkin, once upstream makes a new FCL release (https://github.com/flexible-collision-library/fcl/issues/532) and updates it as ros-noetic-fcl, I am considering unreleasing fcl_catkin from Noetic altogether (https://github.com/ipab-slmc/exotica/pull/735 makes its use redundant). The primary motivation for it was to provide bleeding edge FCL when there was no ros-*-fcl and now that there is, it has outlived its intention.

Tobias-Fischer, 2021-08-01:
In ROS2, we now also have `tvm-vendor` which should be unvendored.

traversaro, 2022-07-03:
apriltag (https://github.com/AprilRobotics/apriltag) is another package that is released as part of ROS but could also be a standalone package in conda-forge, see discussion in https://github.com/stack-of-tasks/pinocchio/issues/1700 .

traversaro, 2022-08-02:
A strange package of this kind is `urdf_parser_py`/`urdfdom-py`. A rather old version is availably on PyPI: https://pypi.org/project/urdf-parser-py/ , however new versions are regularly released in ROS: under the `urdfdom_py` name: https://index.ros.org/r/urdfdom_py/ . The fact that is not on conda-forge is blocking the release of downstream packages that depend on it: https://github.com/ami-iit/ADAM/issues/26 .