RoboStack/ros-noetic #21: ROS tools not found in PowerShell on Windows

binitshah, 2020-12-28:
On Windows 10 x86 64bit, ROS tools like `roscore`, `rostopic`, etc. all work fine with Command Prompt. But when I run `roscore` with PowerShell, I see the following:

```
(ros) PS C:\Users\Binit Shah> roscore
Traceback (most recent call last):
  File "C:\ProgramData\Miniconda3\envs\ros\Library\bin\roscore", line 36, in <module>
    from rosmaster.master_api import NUM_WORKERS
ModuleNotFoundError: No module named 'rosmaster'
```

`rostopic`, `rosservice`, and the others show similar ModuleNotFoundErrors.

Tobias-Fischer, 2020-12-28:
See https://github.com/conda-forge/gazebo-feedstock/issues/42 for a related issue ..

/Cc @traversaro @wolfv

binitshah, 2020-12-28:
Ah okay, sounds like there isn't a `.ps1` activation script for PowerShell yet. If there's already a tracking issue, I can close this one.

Tobias-Fischer, 2020-12-28:
I don't really know what's needed to get this working. If you have an idea, please feel free to open a pull request - this would be a great addition 👍

binitshah, 2020-12-28:
Thanks for the quick replies, I'll try to help with this.

traversaro, 2020-12-28:
> If there's already a tracking issue, I can close this one.

Thanks for reporting the issue! The issue https://github.com/conda-forge/gazebo-feedstock/issues/42 is related but it is not tracking the same problem as gazebo is a standalone conda-forge recipe that it is not related to the ROS packages generated on the `robostack` repo. If you want to look into the issue, I guess you should look in https://github.com/RoboStack/vinca/blob/master/vinca/template.py#L89 that is the code that generates the activation scripts for ROS packages, using the templates in https://github.com/RoboStack/vinca/tree/master/vinca/templates .

binitshah, 2020-12-29:
Thanks for the info @traversaro. On Gitter, I've seen mention of [multisheller](https://github.com/wolfv/multisheller). Is it worth looking into that and if so, are there any examples of it being used here yet?

traversaro, 2020-12-29:
> Thanks for the info @traversaro. On Gitter, I've seen mention of [multisheller](https://github.com/wolfv/multisheller). Is it worth looking into that and if so, are there any examples of it being used here yet?

I guess that in the ROS use case, `multisheller` may not be useful, as in ROS packages you already have the `setup.bat` for each package, and you need to call those one also in powershell, or understand if catkin can generate setup scripts for powershell.

binitshah, 2021-01-03:
Ah okay. Thanks for you help.

traversaro, 2021-02-15:
I looked a bit into this, and from the vinca point of view, it should be sufficient to add the `activate.ps1.in` and `deactivate.ps1.in` in https://github.com/RoboStack/vinca/tree/master/vinca/templates . 

However, the problem for ROS1 is that catkin does not support at the moment generating PowerShell environment scripts (see https://github.com/ros/catkin/blob/203a57ad70430ae165ebe014738a37c3af3e9166/cmake/catkin_generate_environment.cmake#L102 for the supported shells), so for solving that we should either add this to catkin, or add it as a catkin patch. I also looked if there was some way of "sourcing" .bat scripts in powershell, but I could not find any. 

The nice thing is that instead on ROS2 ament_cmake/colcon already supports generating .ps1 environment scripts (see https://colcon.readthedocs.io/en/released/developer/environment.html#entry-points), so supporing PowerShell on ROS2 would be easier, as it would only require to modify vinca to generate the correct activation scripts.