RoboStack/ros-noetic #22: Catkin_make on Windows needs Visual Studio Command Prompt

binitshah, 2020-12-28:
On Windows 10 x86 64bit on Command Prompt, trying to build a workspace with `catkin_make` gives the following:

```
(ros) C:\Users\Binit Shah\temp2>catkin_make
Base path: C:\Users\Binit Shah\temp2
Source space: C:\Users\Binit Shah\temp2\src
Build space: C:\Users\Binit Shah\temp2\build
Devel space: C:\Users\Binit Shah\temp2\devel
Install space: C:\Users\Binit Shah\temp2\install
Copying file from "C:\ProgramData\Miniconda3\envs\ros\Library\share\catkin\cmake\toplevel.cmake" to "C:\Users\Binit Shah\temp2\src\CMakeLists.txt"
####
#### Running command: "cmake C:\Users\Binit Shah\temp2\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\Binit Shah\temp2\devel -DCMAKE_INSTALL_PREFIX=C:\Users\Binit Shah\temp2\install -G NMake Makefiles" in "C:\Users\Binit Shah\temp2\build"
####
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_C_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_CXX_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/Users/Binit Shah/temp2/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Binit Shah/temp2/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

It's possible to resolve this as suggested in the printout by Installing Visual Studio 2019 Community and invoking `C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build>vcvarsall.bat`. But, I'm wondering if there is some conda package for `cl` compiler, so that Windows users don't need to install Visual Studio 2019 Community first. Or if it is the norm on Windows to have VS installed, perhaps the conda env activation scripts can automatically setup the path variables as needed for using catkin.

traversaro, 2020-12-28:
>  But, I'm wondering if there is some conda package for cl compiler, so that Windows users don't need to install Visual Studio 2019 Community first.

As far as I know, at the moment there is no such package, so Visual Studio needs to be installed on the system. 

> Or if it is the norm on Windows to have VS installed, perhaps the conda env activation scripts can automatically setup the path variables as needed for using catkin.

There is a package for this, it is called [`vs2017_win-64`](https://anaconda.org/conda-forge/vs2017_win-64) for VS 2017 and [`vs2019_win-64`](https://anaconda.org/conda-forge/vs2019_win-64) for Visual Studio 2019, however the `vs2019_win-64` package was added only lately in https://github.com/conda-forge/vc-feedstock/pull/19, and I never used it successfully, but perhaps it was only a local problem.

Note that however sourcing the Visual Studio Developer Prompt is only necessary if you use the `NMake Makefiles` or `Ninja` Generator, but if you use the `Visual Studio 16 2019` Generator, all the location of the compiler will be automatically detected without using the Visual Studio Developer Prompt . I do not know if when using `catkin_make` there is the possibility of specifying the CMake Generator, I know that with colcon you can specify it via `--cmake-args -G "Visual Studio 16 2019"`.

danzimmerman, 2021-02-22:
> _however the vs2019_win-64 package was added only lately in conda-forge/vc-feedstock#19, and I never used it successfully, but perhaps it was only a local problem._

`conda install vs2019_win-64` seemed to work fine here on my new conda-Noetic installation.

Tobias-Fischer, 2021-02-22:
Great - should we add this to the readme?

commit 4740984d (2021-02-24): Merge pull request #48 from RoboStack/Tobias-Fischer-patch-1

Fix #22 (add vs2019 to readme)
commit c8c541ff (2021-02-24): Fix #22 (add vs2019 to readme)
closed 2021-02-24 by wolfv