RoboStack/ros-noetic #24: RVIZ font issue with TF

capaulson, 2021-01-03:
Using on Mac OS X. RVIZ runs fine, but when you attempt to add the TF info, RVIZ crashes with: 

```libc++abi.dylib: terminating with uncaught exception of type Ogre::Exception: : Could not find font Liberation Sans in MovableText::setFontName```

Not sure where to start, any pointers?

Tobias-Fischer, 2021-01-03:
Hiya,

Could you try applying the following patch? I'm not sure where this file ends up, let me know if you cannot find it.

```
diff --git a/ogre_media/fonts/liberation_sans.fontdef b/ogre_media/fonts/liberation_sans.fontdef
index e714db0f..311a7ac5 100644
--- a/ogre_media/fonts/liberation_sans.fontdef
+++ b/ogre_media/fonts/liberation_sans.fontdef
@@ -1,12 +1,12 @@
-Liberation Sans
-{
-  type truetype
-  source liberation-sans/LiberationSans-Regular.ttf
-  size 18
-  resolution 96
-}
+// Liberation Sans
+// {
+//   type truetype
+//   source liberation-sans/LiberationSans-Regular.ttf
+//   size 18
+//   resolution 96
+// }
 ```

(from https://github.com/conda-forge/ogre-feedstock/issues/15)

Tobias-Fischer, 2021-01-06:
Actually this patch I mentioned before is not needed anymore - it must be something else. I am not sure where to start either, then, sorry .. I would build `rviz` with the patches applied (https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-rviz.patch & https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-rviz.osx.patch) in debug configuration within a catkin_ws, and then try to see whether you can get any more specific pointers where things fail.

Tobias-Fischer, 2021-01-07:
@capaulson: Could you please post the output of `rosrun rviz rviz -l`?

Tobias-Fischer, 2021-01-07:
Found the culprit - rviz is rebulding now. Please uninstall & reinstall and it should work (wait an hour at least for the rebuild)

closed 2021-01-07 by Tobias-Fischer

capaulson, 2021-01-15:
Fantastic! Thank you for the very quick response!

commit 42b79538 (2021-01-07): Fix #24