RoboStack/ros-noetic #41: rosdep alternative?

chisarie, 2021-02-16:
Hi, 
thank you very much for this project! I am trying it out to see if I can finally use ROS in a conda environment :).
I had a question: I am trying to run a moveit example from their [documentation](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html), and I am asked to run rosdep in order to install the missing dependencies:
`rosdep install -y --from-paths . --ignore-src --rosdistro noetic`

Unfortunately it seems to me that it is not able to find the missing packages, since I get errors similar to 
`
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:  
moveit_tutorials: Cannot locate rosdep definition for [pcl_conversions]
`

Installing all the packages individually with `conda install ros-noetic-my-package` works, so this is what I am doing now, but I am wondering if there is a way to make rosdep work with your system.

Thank you in advance for your help!
Eugenio

Tobias-Fischer, 2021-02-16:
Hi @chisarie, many thanks for opening this issue! Unfortunately we don't have a solution for this yet, but we have been talking about this previously as it would be a great feature. Stay tuned, it's on our to-do list.

chisarie, 2021-02-17:
Thank you very much for letting me know. Keep up with the great work!

wolfv, 2021-02-17:
I think we can leave this open until we have some solution (either in rosdep, or as a seperate tool).

How did everything work for you @chisarie ?

chisarie, 2021-02-17:
Ok let's leave it open then. 
I was unfortunately not able to get the moveit tutorial running, but it might be that the problem has to do with my system.
I am using Ubuntu 18.04 and I already have a system wide installation of ROS melodic. I then installed in a conda environment ros-noetic from RoboStack, I was able to complete the installation and get the basic ros command working (roscore, rostopic list, etc.). But then I tried building the [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) from moveit (noetic version), and I was getting build errors. I didn't investigate much, since I didn't have much time.
It is worth noticing that the problem could be from different reason:
1) My system wide ros-melodic installation was interfering with the RoboStack ros-noetic
2) The noetic branch of moveit is their developement branch, so it's not stable. Maybe the errors had nothing to do with Robostack then.

I might try again in the next weeks if I have more time. If you have the chance to try using Moveit in ros-noetic while on Ubuntu 18.04, I would be interested to hear how it goes for you :).
Thank you again!

Tobias-Fischer, 2021-02-17:
Hi @chisarie, many thanks for your feedback! I tried running the moveit tutorials (or at least the very basic one) and it was working fine with me, with an almost identical setup. The only thing to do with a system wide ROS installation is to not source the setup.sh, as otherwise the Python path of the system installation is conflicting with the conda installation.

When you have more time it would be great if you could report the precise issue you were running into, so we can fix potential issues.

traversaro, 2021-02-18:
Another possible source of difference is whether you were using the system compilers and build tools, or the conda-forge provided ones  (i.e. `conda install compilers cmake pkg-config make ninja` ). This can be easy checked by running `conda list` and reporting the result.

Tobias-Fischer, 2021-02-18:
That's a good point! I think we should add installing these packages in the readme, at least optionally.

chisarie, 2021-02-18:
@Tobias-Fischer Hi, thank you for letting me know that it works for you! I will try again next week and let you know how it goes. @traversaro Yes you are right, I was probably using the system compilers and build tools. I could also try with conda installed ones next time.

traversaro, 2021-05-01:
An interesting related PR is https://github.com/ros-infrastructure/rosdep/pull/697, that added support in rosdep for nix,  that similarly to robostack is used in https://github.com/lopsided98/nix-ros-overlay to mantain a user-level distribution mechanism for ROS.

Tobias-Fischer, 2021-05-03:
I think I got it working :). I think there are some improvements that can be made, but overall it should be a good start. It would be great if you could have a look @wolfv @traversaro.

- vinca: https://github.com/RoboStack/vinca/pull/2
- rospkg: https://github.com/RoboStack/rospkg/pull/1
- rosdep: https://github.com/RoboStack/rosdep/pull/1
- ros-noetic: https://github.com/RoboStack/ros-noetic/pull/119

Tobias-Fischer, 2021-05-03:
Nice, thanks for the reviews @wolfv! Now I think we need to rebuild the rospkg, rosdep & rosdistro packages with the patches from the respective PRs applied. Some might not apply cleanly .. will have to check.

wolfv, 2021-05-03:
I think it would be good to also make upstream PRs, and use them as reference in the patches.

Tobias-Fischer, 2021-05-03:
Ok. So I opened an upstream PR in https://github.com/ros-infrastructure/rospkg/pull/224

As soon as that's merged, we can tackle rosdep. 

I don't think that they'll merge the rosdistro changes and it might be inconvenient for us to have the mapping robostack.yaml file upstream, so I suggest to patch that locally instead.

I also opened a conda-forge PR in the rospkg-feedstock: https://github.com/ros-infrastructure/rospkg/pull/224
Similarly, as soon as that's merged, we can tackle rosdep-feedstock.

wolfv, 2021-05-04:
rosdep and rospkg in the feedstocks are merged :)

Tobias-Fischer, 2021-05-04:
Nice! It would be great if someone could test whether this works okay on Windows. I can give it a try on OSX tomorrow.

Tobias-Fischer, 2021-05-04:
If that works, we can modify the moveit CI PR to make use of this functionality - that would be amazing!

wolfv, 2021-05-04:
good idea re. the moveit PR.

Tobias-Fischer, 2021-05-11:
Done - let's close this :)

closed 2021-05-11 by Tobias-Fischer