RoboStack/ros-noetic #47: Build against Python 3.9

mstoelzle, 2021-02-22:
As Python 3.9.* has been release for several months already, could you please also build against Python 3.9?

traversaro, 2021-02-23:
Hi @mstoelzle ! It may be sufficient to just add a line in https://github.com/RoboStack/ros-noetic/blob/ce93a99bf435d1f116988a8aa14bebdb3a96f64f/conda_build_config.yaml#L586 with `  - 3.9.* *_cpython` (see https://conda.io/projects/conda-build/en/latest/resources/variants.html for the underlying conda mechanism) but i guess that then this will require a full rebuilt of the packages, at least the one that depend on Python.

Probably @wolfv or @Tobias-Fischer can provide a more informed opinion.

Tobias-Fischer, 2021-02-23:
Yes that's correct. At this time the only thing stopping us is that it doubles the build time ..

traversaro, 2021-04-11:
Probably we will automatically align with what conda-forge does once we start using also the conda-forge-pinnings of conda-forge (from https://github.com/conda-forge/conda-forge-pinning-feedstock)?

Tobias-Fischer, 2021-04-11:
We recently started using the conda-forge-pinnings (they are included in https://github.com/RoboStack/ros-noetic/blob/master/.ci_support/conda_forge_pinnings.yaml), but then add and modify some bits and pieces in https://github.com/RoboStack/ros-noetic/blob/master/conda_build_config.yaml - including the Python version: https://github.com/RoboStack/ros-noetic/blob/817172343a2784672c13916916fa5e3741cb8745/conda_build_config.yaml#L44-L53

The main problem is that the build time is - at least at the moment - multiplied by the number of Python versions that we want to build. Especially for some of the packages we would probably hit the maximum build time in the current setup. We would need to modify vinca to generate separate build pipelines ideally. What do you think @wolfv?

Tobias-Fischer, 2021-05-04:
This is something we should chat about with Microsoft @wolfv - if we get the 4 hour time limit instead of one hour, we can build for multiple versions of Python etc.

resident12, 2022-01-20:
Actual situation is that Robostack supports Python3.6|3.7|3.8. The "full support" status of Python3.6 has ended per since 2018-12-24 while Python3.10 has become "lastest stable" release per 2021-10-04. Please consider that typical development projects that are based on RoboStack will be stretched over ~3yrs.  So starting a project today (latest RoboStack is Python3.8) will lack novel features from the very beginning and security updates at the end. 

May I suggest to reconsider Robostacks support for the various Python releases and preferably stay in sync with the official Python roadmap please.

Tobias-Fischer, 2022-01-20:
Hi @resident12,
Thanks for your feedback! We restricted RoboStack to just Python 3.8 a long time ago, the builds for 3.6 and 3.7 are ancient. 

I fully agree that the current situation is not ideal, but we are bound to use a single python version because of the build timeouts. We can consider using e.g. Python 3.9 for the next rebuild - for Python 3.10 there are still too many packages missing on conda-forge (see https://conda-forge.org/status/).

If you want to use Python 3.9 or newer now, you have two options: You can build the packages locally, or you can create a Pull Request here that bumps the Python version etc. and rebuilds the packages in CI. RoboStack is a community project, it would be great to contribute back :).

Tobias-Fischer, 2022-01-30:
Have just triggered this in https://github.com/RoboStack/ros-noetic/commit/5ae5d075358cf2f4cbc7c11dbed81c6c0efecf52 but it will take quite some time to get the packages to build.

closed 2022-02-01 by Tobias-Fischer

adammacudzinski, 2022-02-01:
@Tobias-Fischer My team is dependent on the Python3.6 packages. Our CI pipelines actually just started breaking within the last few days because of this change. IMO this change should entail incrementing the ros-noetic package version because you're making a public interface (python) not backwards compatible.  

Would it be possible to send me the old python3.6 tarball files? So we can upload these to our own artifactory.

Tobias-Fischer, 2022-02-01:
See #216