RoboStack/ros-noetic #49: rosdep fails on Windows 10

danzimmerman, 2021-02-22:
Seems identical to this: https://github.com/ros-infrastructure/rosdep/issues/610

Suggested fix works.

traversaro, 2021-02-22:
Note that when using the RoboStack channel the logical equivalent of using rosdep is installing dependencies via conda-forge, see https://github.com/RoboStack/ros-noetic/issues/41 for a related issue.

danzimmerman, 2021-02-23:
> Note that when using the RoboStack channel the logical equivalent of using rosdep is installing dependencies via conda-forge, see #41 for a related issue.

Right, and I guess generally conda will ensure that anything installed will come with its system dependencies, so something like `rosdep check` is redundant...

Tobias-Fischer, 2021-02-25:
I suggest to continue discussion in https://github.com/RoboStack/ros-noetic/issues/41 and will close here.

closed 2021-02-25 by Tobias-Fischer