RoboStack/ros-noetic #53: mavros package fow Windows 10

Y-Grigoriy, 2021-02-25:
Hello, is it planning to add mavros package compatible with Windows? If yes, when can it be possible?

Tobias-Fischer, 2021-02-26:
Do you want it to give a go and open a pull request? In this way you can assist us and accelerate getting the package built.

Y-Grigoriy, 2021-02-26:
Unfortunately, I don't have enough skills to help in the package built. I just start to work with ROS for project aimed to development UAV. I plan to create new code based on the old version written by my colleague in Ubuntu using mavros.

Tobias-Fischer, 2021-03-04:
Hi @Y-Grigoriy, I patched the mavros package to compile on Windows: https://github.com/RoboStack/ros-noetic/pull/57
I will merge this pull request soon. However, this is totally untested, and I will not have time to fix the package in the very likely case that things don't work. Feel free to test the package, and to open pull requests with changes if they are needed - it is easy to do so, and https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md describes how to do it.

closed 2021-03-04 by Tobias-Fischer