RoboStack/ros-noetic #55: Gazebo fails to start on MacOS Catalina with "Symbol not found: _alcOpenDevice"

sonelu, 2021-03-02:
Environment:

MacOS Catalina 10.15.7
Conda: 4.9.2
Python: 3.8.8

Installed ros-noetic-desktop-full in it's own environment.

Installation run without any error.

When starting Gazebo there is the following runtime error:

```
dyld: lazy symbol binding failed: Symbol not found: _alcOpenDevice
  Referenced from: /Users/Alex/anaconda3/envs/robostack/bin/../lib/libgazebo_util.11.dylib
  Expected in: flat namespace

dyld: Symbol not found: _alcOpenDevice
  Referenced from: /Users/Alex/anaconda3/envs/robostack/bin/../lib/libgazebo_util.11.dylib
  Expected in: flat namespace
```

Not sure if this is relevant for the ROS build but I also have:

```
> xcodebuild -version
Xcode 12.4
Build version 12D4e
```

Thanks,

Alex.

Tobias-Fischer, 2021-03-02:
Hiya, could you please post the output of `conda list`?

sonelu, 2021-03-02:
Here it is:

```
 > conda list
# packages in environment at /Users/Alex/anaconda3/envs/robostack:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        3.38.0               hd0a4bf8_1    conda-forge
apr                       1.7.0                hbcf498f_4    conda-forge
assimp                    5.0.1                h3b8ecc4_5    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py38h5406a74_1    conda-forge
boost                     1.74.0           py38h692b87f_3    conda-forge
boost-cpp                 1.74.0               h43a636a_2    conda-forge
bzip2                     1.0.8                hc929b4f_4    conda-forge
c-ares                    1.17.1               hc929b4f_0    conda-forge
c-compiler                1.1.3                h35c211d_0    conda-forge
ca-certificates           2020.12.5            h033912b_0    conda-forge
cairo                     1.16.0            h4ec454a_1007    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
catkin_tools              0.6.1            py38h50d1736_2    conda-forge
cctools                   949.0.1             hd9211c8_19    conda-forge
cctools_osx-64            949.0.1             h2f0f38f_19    conda-forge
certifi                   2020.12.5        py38h50d1736_1    conda-forge
cffi                      1.14.5           py38ha97d567_0    conda-forge
clang                     11.0.1               h694c41f_1    conda-forge
clang-11                  11.0.1          default_hf8bb9ca_1    conda-forge
clang_osx-64              11.0.1               hb91bd55_2    conda-forge
clangxx                   11.0.1          default_hf8bb9ca_1    conda-forge
clangxx_osx-64            11.0.1               h7e1b574_2    conda-forge
cmake                     3.19.6               h641592c_0    conda-forge
compiler-rt               11.0.1               h654b07c_0    conda-forge
compiler-rt_osx-64        11.0.1               h8c5fa43_0    conda-forge
compilers                 1.1.3                h694c41f_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
cppzmq                    4.7.1                h6daa668_1    conda-forge
cryptography              3.4.4            py38h43df06b_0    conda-forge
curl                      7.71.1               hcb81553_8    conda-forge
cxx-compiler              1.1.3                h926bf3e_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dbus                      1.13.6               h0c50699_1    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.16             py38h50d1736_3    conda-forge
eigen                     3.3.9                h926bf3e_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.5                h35c211d_0    conda-forge
expat                     2.2.10               h1c7c35f_0    conda-forge
ffmpeg                    4.3.1                hb787a4c_2    conda-forge
flann                     1.9.1             hed988e5_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      2.001                hab24e00_0    conda-forge
font-ttf-source-code-pro  2.030                hab24e00_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            hd23ceaa_1004    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.1.3                h6818c3e_0    conda-forge
freeimage                 3.18.0               h6802f34_4    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.3.0               hc4feadb_7    conda-forge
gdbm                      1.18                 h8a0c380_2    conda-forge
gdk-pixbuf                2.42.2               h976c81d_2    conda-forge
gettext                   0.19.8.1          h7937167_1005    conda-forge
gfortran_impl_osx-64      9.3.0               h9cc0e5e_19    conda-forge
gfortran_osx-64           9.3.0               h18f7dce_13    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
glib                      2.66.7               h1c7c35f_0    conda-forge
glib-tools                2.66.7               h1c7c35f_0    conda-forge
[... 418 lines cut ...]
xorg-libxi                1.7.10               h01d97ff_0    conda-forge
xorg-libxinerama          1.1.4             h1c7c35f_1001    conda-forge
xorg-libxmu               1.1.3                h0b31af3_0    conda-forge
xorg-libxpm               3.5.13               h0b31af3_0    conda-forge
xorg-libxrandr            1.5.2                h0b31af3_1    conda-forge
xorg-libxrender           0.9.10            h01d97ff_1002    conda-forge
xorg-libxt                1.1.5             h01d97ff_1003    conda-forge
xorg-libxtst              1.2.3             h0b31af3_1002    conda-forge
xorg-randrproto           1.5.0             h0b31af3_1001    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                haf1e3a3_0    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                h1c7c35f_0    conda-forge
zlib                      1.2.11            h7795811_1010    conda-forge
zstd                      1.4.8                hf387650_1    conda-forge
zziplib                   0.13.69              hb0a8c7a_1    conda-forge
```

Thanks for this guys.

traversaro, 2021-03-02:
Another useful debug information would be to print the output of this command in a terminal in which the conda environment is activated:
~~~
otool -L /Users/Alex/anaconda3/envs/robostack/lib/libgazebo_util.11.dylib
~~~

sonelu, 2021-03-02:
Here it is:

```
(robostack) Alex3-2:~ Alex$ otool -L /Users/Alex/anaconda3/envs/robostack/lib/libgazebo_util.11.dylib
/Users/Alex/anaconda3/envs/robostack/lib/libgazebo_util.11.dylib:
	@rpath/libgazebo_util.11.dylib (compatibility version 11.0.0, current version 11.3.0)
	@rpath/libgazebo_transport.11.dylib (compatibility version 11.0.0, current version 11.3.0)
	@rpath/libgazebo_msgs.11.dylib (compatibility version 11.0.0, current version 11.3.0)
	@rpath/libtinyxml2.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	@rpath/libgazebo_common.11.dylib (compatibility version 11.0.0, current version 11.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.200.5)
	@rpath/libsdformat9.9.dylib (compatibility version 9.0.0, current version 9.3.0)
	@rpath/libprotobuf.26.dylib (compatibility version 27.0.0, current version 27.1.0)
	@rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_iostreams.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtbb.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libignition-transport8.8.dylib (compatibility version 8.0.0, current version 8.1.0)
	@rpath/libignition-msgs5.5.dylib (compatibility version 5.0.0, current version 5.3.0)
	@rpath/libignition-math6.6.dylib (compatibility version 6.0.0, current version 6.7.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

Thanks,

Alex.

traversaro, 2021-03-02:
@Tobias-Fischer did you ever tried in the past the Gazebo conda-forge binary on macOS? On which macOS version?

If we never tested Gazebo on macOS to be working correctly, then the explanation could be that there may be a bug in Gazebo upstream in handling the OpenAL dependency, in particular in https://github.com/osrf/gazebo/blob/b0dfa186838de0874e0084bf0e08d4f0a1d6f15c/cmake/SearchForStuff.cmake#L74 OpenAL is enabled without the possibility of disabling it, but only gazebo_common (https://github.com/osrf/gazebo/blob/67c43463c87685b7c9d485a9f8d2e6ac760815b9/gazebo/common/CMakeLists.txt#L274) and not gazebo_util are linked to it. 

If instead Gazebo via conda-forge on other macOS machines was tested to run, then there may be something specific going on @sonelu system, perhaps related to the OS version or something else.

traversaro, 2021-03-02:
I added a regression test to understand if this is a problem of the gazebo binary in https://github.com/conda-forge/gazebo-feedstock/pull/54 .

sonelu, 2021-03-02:
@traversaro I've seen you added `gazebo --version` in the smoke test. The problem is that is running (at least on my machine):

```
> gazebo --version
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

So this check might not be conclusive for the conda-forge build.

traversaro, 2021-03-02:
> @traversaro I've seen you added `gazebo --version` in the smoke test. The problem is that is running (at least on my machine):
>
> ```
> Gazebo multi-robot simulator, version 11.3.0
> Copyright (C) 2012 [...]

gazebo is actually a rather small process that just spawns two more process, that are the one that actually link all the libraries, that are `gzserver` and `gzclient`. Can you check if `gzserver --version` and `gzclient --version` work for you? In theory one of the two should fail as well.

traversaro, 2021-03-02:
Anyhow, nothing seems strange in https://github.com/conda-forge/gazebo-feedstock/pull/54 .

sonelu, 2021-03-02:
they both work for `--version`

```
(robostack) Alex3-2:~ Alex$ gzserver --version
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

(robostack) Alex3-2:~ Alex$ gzclient --version
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

```

If I run them with `--verbose` `gzserver` is the one to fail:

```
(robostack) Alex3-2:~ Alex$ gzclient --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
^C
```

```
(robostack) Alex3-2:~ Alex$ gzserver --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.30
[Msg] Loading world file [/Users/Alex/anaconda3/envs/robostack/share/gazebo-11/worlds/empty.world]
dyld: lazy symbol binding failed: Symbol not found: _alcOpenDevice
  Referenced from: /Users/Alex/anaconda3/envs/robostack/bin/../lib/libgazebo_util.11.dylib
  Expected in: flat namespace

dyld: Symbol not found: _alcOpenDevice
  Referenced from: /Users/Alex/anaconda3/envs/robostack/bin/../lib/libgazebo_util.11.dylib
  Expected in: flat namespace

Abort trap: 6
 ```

traversaro, 2021-03-02:
Cool thanks, then indeed probably there is some library that are loaded in a lazy way, let's see if we can transform it in something that we can run and quickly get an output.

traversaro, 2021-03-02:
Can you try to run:
~~~
gzserver --verbose -p this-file-does-not-exist.txt
~~~

?

sonelu, 2021-03-02:
```
(robostack) Alex3-2:~ Alex$ gzserver --verbose -p this-file-does-not-exist.txt
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.30
[Err] [LogPlay.cc:70] EXCEPTION: Invalid logfile [this-file-does-not-exist.txt]. Does not exist.

[Err] [LogPlay.cc:70] EXCEPTION: Invalid logfile [this-file-does-not-exist.txt]. Does not exist.
```

traversaro, 2021-03-02:
Ok, this trick does not work then.

Does the `gz` command work fine instead?

sonelu, 2021-03-02:
Yes, `gz` works:

```
(robostack) Alex3-2:~ Alex$ gz 
This tool modifies various aspects of a running Gazebo simulation.

  Usage:  gz <command>

List of commands:

  help      Print this help text.
  camera    Control a camera
  debug     Returns completion list for a command. Used for bash completion.
  help      Outputs information about a command
  joint     Modify properties of a joint
  log       Introspects and manipulates Gazebo log files.
  marker    Add, modify, or delete visual markers
  model     Modify properties of a model
  physics   Modify properties of the physics engine
  sdf       Converts between SDF versions, and provides info about SDF files
  stats     Print statistics about a running gzserver instance.
  topic     Lists information about topics on a Gazebo master
  world     Modify world properties

Use "gz help <command>" to print help for a command.
```

sonelu, 2021-03-03:
@traversaro @Tobias-Fischer I have reinstalled Gazebo after you published the patch on conda-forge and now I can start it (just simple start). I'm still gettting a seg fault in the `spawn-urdf` when launching a robot that works on a VirtualBox VM with noetic, but I will have to look more on this error. It's very possible that it comes from the `ros_control`. Unfortunatelly there are no logs generated.

If you want you can close this issue as the original problem with Gazebo is fixed.

traversaro, 2021-03-03:
Thanks @sonelu for checking this! Actually we are experiencing some segfault in our tests, so it could be also related to Gazebo itself. We will track some issue on gazebo-feedstock to track those problems, feel free to open new issues if you isolate your problem, thanks!

Tobias-Fischer, 2021-03-05:
Let's keep discussing in https://github.com/RoboStack/ros-noetic/issues/60

closed 2021-03-05 by Tobias-Fischer