RoboStack/ros-noetic #58: rqt fails with "Fatal Python error: _PyInterpreterState_Get(): no current thread state" on MacOS

sonelu, 2021-03-03:
Context:

MacOS Catalina 10.15.7 (19H524)
Conda: 4.9.2
Python: 3.8.8

Installed ros-noetic-desktop-full in it's own environment.

Installation run without any error.

When starting `rqt` there is a python crash:

```
> rqt
Fatal Python error: _PyInterpreterState_Get(): no current thread state
Python runtime state: unknown

Abort trap: 6
```

There is a long dump that MacOS produces and the top is the following:

```
Process:               python3.8 [67316]
Path:                  /Users/USER/*/python
Identifier:            python3.8
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [38679]
Responsible:           Terminal [688]
User ID:               502

Date/Time:             2021-03-03 08:38:42.847 +0000
OS Version:            Mac OS X 10.15.7 (19H524)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        D8DB1A5A-2F6F-7F4E-01CC-74BEA39189FC

Sleep/Wake UUID:       3BEDFC53-F08F-46DA-BAC4-C3666E7B7DEE

Time Awake Since Boot: 66000 seconds
Time Since Wake:       29000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6832233a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff683dee60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff682a9808 abort + 120
3   libpython3.8.dylib            	0x000000011febdebc fatal_error + 60
4   libpython3.8.dylib            	0x000000011febdfe3 Py_FatalError + 19
5   libpython3.8.dylib            	0x000000011fd75662 PyModule_Create2 + 50
6   libqt_gui_cpp_sip.so          	0x000000011f6c97d8 PyInit_libqt_gui_cpp_sip + 28
7   python                        	0x000000010f0b853c _PyImport_LoadDynamicModuleWithSpec + 572
8   python                        	0x000000010f0b7637 _imp_create_dynamic + 183
9   python                        	0x000000010ef793b4 cfunction_vectorcall_FASTCALL + 84
10  python                        	0x000000010ef17ef9 PyVectorcall_Call + 121
...
```

The issue seems to be in `libqt_gui_cpp_sip.so` that might somehow be linked to a different `libpython` although `otool` seems fine:

```
(robostack) Alex3-2:~ Alex$ otool -L /Users/Alex/anaconda3/envs/robostack/lib/python3.8/site-packages/qt_gui_cpp/libqt_gui_cpp_sip.so 
/Users/Alex/anaconda3/envs/robostack/lib/python3.8/site-packages/qt_gui_cpp/libqt_gui_cpp_sip.so:
	@rpath/libqt_gui_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libclass_loader.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libPocoFoundation.71.dylib (compatibility version 71.0.0, current version 71.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
	@rpath/librosconsole.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosconsole_log4cxx.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosconsole_backend_interface.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/liblog4cxx.11.dylib (compatibility version 11.0.0, current version 11.0.0)
	@rpath/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librostime.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_date_time.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libcpp_common.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_thread.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libconsole_bridge.1.0.dylib (compatibility version 1.0.0, current version 0.0.0)
	@rpath/libroslib.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librospack.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libpython3.8.dylib (compatibility version 3.8.0, current version 3.8.0)
	@rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_system.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtinyxml2.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	@rpath/libQt5Core.5.dylib (compatibility version 5.12.0, current version 5.12.9)
	@rpath/libQt5Gui.5.dylib (compatibility version 5.12.0, current version 5.12.9)
	@rpath/libQt5Widgets.5.dylib (compatibility version 5.12.0, current version 5.12.9)
	@rpath/libQt5PrintSupport.5.dylib (compatibility version 5.12.0, current version 5.12.9)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)

```

Thanks,

Alex.

Tobias-Fischer, 2021-03-03:
Hi Alex, thanks for reporting this! That'll be a tricky one with all the sip and Qt stuff involved. Could you please check whether rviz and rqt_rviz work? Did you ever manage to successfully run rqt on OSX using e.g. a homebrew installation?

sonelu, 2021-03-03:
`rviz` starts, although the view is skewed vertically:

[image]

`rqt_rviz` is not avaialble with the RoboStack conda environment:

```
(robostack) Alex3-2:catkin_ws Alex$ rqt_
rqt_bag           rqt_console       rqt_dep           rqt_graph         rqt_image_view    rqt_logger_level  rqt_plot          rqt_shell
```

If I start it as a plugin I get this:

```
(robostack) Alex3-2:rqt_rviz Alex$ rqt -s rqt_rviz
Fatal Python error: _PyInterpreterState_Get(): no current thread state
Python runtime state: unknown

Abort trap: 6
```

Or like this:

```
(robostack) Alex3-2:rqt_rviz Alex$ rosrun rqt_rviz rqt_rviz
Fatal Python error: _PyInterpreterState_Get(): no current thread state
Python runtime state: unknown

Abort trap: 6
```

sonelu, 2021-03-03:
I have never used `rqt` with Homebrew, but I will try to and let you know.

sonelu, 2021-03-03:
Actually it doesn't seem to be a formula / tap anymore in `brew` that installs `rqt`.  I'll check if it would be possible to install it from source.

jc-bao, 2021-03-11:
I have the same problem, I have tried the following method:
1. downgrade the package: not work, as there is no other version to choose from.
2. install by brew: in fact, there are no brew packages called rqt, I only find rqt in mac when installing a full ros desktop.
3. build from source: everything is fine until I use rosdep, which cannot identify Big Sur(`ERROR: Rosdep experienced an error: unrecognized version: 11.2`). I don't know if there are other ways to build without rosdep, but up to now, I still cannot figure out a way to build from source. (actually, there are a [patch](https://github.com/ros-infrastructure/rospkg/pull/216) for this problem, but I don't know how to reinstall rospkg in conda QAQ)
By the way, my rqt_rviz also cannot be run. Hopefully, this issue can be solved. I'd be appreciated for any valuable advice.

sonelu, 2021-03-19:
The problem seems to come from `qt_gui_cpp`, specifically from the sip generated library.

It can be replicated very easily in python like this:

```
(robostack) Alex3-2:~ Alex$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:12:38) 
[Clang 11.0.1 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import qt_gui_cpp                                                <<< this seems to be fine
>>> import qt_gui_cpp.libqt_gui_cpp_sip                              <<< this not
Fatal Python error: _PyInterpreterState_Get(): no current thread state
Python runtime state: unknown

Abort trap: 6
(robostack) Alex3-2:~ Alex$ 
```

Should we raise an issue in the `ros_visualization/qt_gui_core`?

Tobias-Fischer, 2021-03-22:
Fixed with https://github.com/RoboStack/ros-noetic/commit/70d6309529be3c82352ef42b1d31029024280438 and locally tested (new packages are uploaded). Give it a go @sonelu :)

closed 2021-03-22 by Tobias-Fischer

Tobias-Fischer, 2021-03-22:
As this is a recurring theme - @traversaro (as a CMake expert) - is there a way to use catkin to remove `PYTHON_LIBRARIES` whenever it is linked to it? And add the `LINK_FLAGS "-undefined dynamic_lookup"` instead?

wolfv, 2021-03-22:
@Tobias-Fischer is it not enough to add the undefined dynamic_lookup? Do we need to remove the Python libraries? I remember that I had to add the undefined dynamic_lookup to MoveIt as well...

Tobias-Fischer, 2021-03-22:
Oh I'm not sure. I thought the undefined lookup stuff is only needed because otherwise an error is thrown when removing the python libraries? Better be safe ;)

traversaro, 2021-03-22:
> 
> 
> As this is a recurring theme - @traversaro (as a CMake expert) - is there a way to use catkin to remove `PYTHON_LIBRARIES` whenever it is linked to it? And add the `LINK_FLAGS "-undefined dynamic_lookup"` instead?

If I remember correctly catkin redefines `add_library` on Windows (see https://github.com/ros/catkin/blob/203a57ad70430ae165ebe014738a37c3af3e9166/cmake/platform/windows.cmake#L42), so a possible strategy would be to re-define it also on macOS to ensure that `LINK_FLAGS` is added, but to be honest I am afraid that this will bring more harm then benefit. Instead for removing the `${PYTHON_LIBRARIES}` link I think it would be more hard, in theory something could be done via [`cmake_language(DEFER ..)`](https://cmake.org/cmake/help/latest/command/cmake_language.html#defer) but again this is kind of an invasive change that probably bring more problems. If we just need to add some linker option, can't we just do that by manipulating the `LDFLAGS` environment variable for the affected package?

traversaro, 2021-03-22:
Actually I took at step back, and I tried to understand why removing the linking to `${PYTHON_LIBRARIES}` is necessary at all, i.e. by looking into https://github.com/ros/geometry2/pull/502 and https://github.com/pytorch/pytorch/commit/73f6715f4725a0723d8171d3131e09ac7abf0666 . If I understood correctly, for some reason the wrong Python libraries are linked, and then the solution is to remove the linking to the libraries, and add `-undefined dynamic_lookup` because by default the linking process in macOS is quite strict on making sure that all the used symbols are defined at linking time. 

However, if I am not wrong the original problem seems to be that the deprecated `FindPythonLibs.cmake` for some reason is linking the wrong Python libraries? To fix that, perhaps it sufficient to specify directly the `PYTHON_LIBRARY` and `PYTHON_INCLUDE_DIR` during CMAke invocation (see https://github.com/Kitware/CMake/blob/f4f3fca99fd4dc473fba31f17c5d09b8b6e2d1fb/Modules/FindPythonLibs.cmake#L39)? Otherwise, another suggestion in `FindPythonLibs` is to call before `find_package(PythonInterp)`, so the active Python interpreter is found first (in conda case the one of coda) and the `PYTHON_LIBRARIES` then will be populated correctly. Yes another fix could be to use the non-deprecated `find_package(Python)`, but that probably will require more changes.

traversaro, 2021-03-22:
Another common strategy is to just set the `PYTHON_EXECUTABLE` cmake variable to `$PYTHON` , see https://github.com/search?l=Shell&q=org%3Aconda-forge+PYTHON_EXECUTABLE&type=Code , that from what I understand from `FindPythonLibs.cmake` should work as well.

sonelu, 2021-03-22:
@Tobias-Fischer is it `rospack` that was the problem? Do I have to test it from source? I haven't seen any new builds pushed to the conda channel.

Tobias-Fischer, 2021-03-22:
Thanks for your suggestions, Silvio! I guess we want to avoid changing any code in random packages, as this would require us to know the python libraries are being linked to, which isn't obvious as in the case of rospack (I'm referring to adding a call to PythonInterp). 

We already set the PYTHON_EXECUTABLE as far as I remember. Howeve, I like your suggestion of adding PYTHON_LIBRARY and PYTHON_INCLUDE_DIR. Do you happen to know where they should point to? This is easy to change in the build template.

Tobias-Fischer, 2021-03-22:
@sonelu https://anaconda.org/RoboStack/ros-noetic-rospack/files

traversaro, 2021-03-22:
> We already set the PYTHON_EXECUTABLE as far as I remember.

Interesting, how wonder why it is not working then (or perhaps the issue is another).

> Howeve, I like your suggestion of adding PYTHON_LIBRARY and PYTHON_INCLUDE_DIR. Do you happen to know where they should point to? This is easy to change in the build template.

Unfortunately I am afraid those are platform and python version dependent, for Python 3.9 with macOS I guess is:
* PYTHON_LIBRARY  ${PREFIX}/lib/libpython3.9.dylib
* PYTHON_INCLUDE_DIR ${PREFIX}/include/python3.9

sonelu, 2021-03-22:
The good news is that `rqt` now works:

[image]

I haven't yet gone through all the plug-ins, but I'll do that in time.

The bad news is that `librospack.dylib` has a bug now.

The easiest way to reproduce it is to invoke `roslaunch` or `rosrun` and use the autocomplete:

```
roslaunch [tab]dyld: Symbol not found: __Py_NoneStruct
  Referenced from: /Users/Alex/anaconda3/envs/ros1/bin/../lib/librospack.dylib
  Expected in: flat namespace
 in /Users/Alex/anaconda3/envs/ros1/bin/../lib/librospack.dylib
```

if I enter manually the package and the launch file I get later the same error when the nodes are started:

```
(ros1) Alex3-2:catkin_ws Alex$ roslaunch mh5_description mh5_rviz.launch
... logging to /Users/Alex/.ros/log/5440e8d0-8b02-11eb-849f-8c8590d4cfc8/roslaunch-Alex3-2.local-11607.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://Alex3-2.local:65532/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [11621]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5440e8d0-8b02-11eb-849f-8c8590d4cfc8
process[rosout-1]: started with pid [11623]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [11624]
process[rviz-3]: started with pid [11625]
dyld: Symbol not found: __Py_NoneStruct
  Referenced from: /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib
  Expected in: flat namespace
 in /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib
dyld: Symbol not found: __Py_NoneStruct
  Referenced from: /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib
  Expected in: flat namespace
 in /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib
[robot_state_publisher-2] process has died [pid 11624, exit code -6, cmd /Users/Alex/anaconda3/envs/ros1/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/Users/Alex/.ros/log/5440e8d0-8b02-11eb-849f-8c8590d4cfc8/robot_state_publisher-2.log].
log file: /Users/Alex/.ros/log/5440e8d0-8b02-11eb-849f-8c8590d4cfc8/robot_state_publisher-2*.log
================================================================================REQUIRED process [rviz-3] has died!
process has died [pid 11625, exit code -6, cmd /Users/Alex/anaconda3/envs/ros1/lib/rviz/rviz -d /Users/Alex/catkin_ws/src/mh5_robot/mh5_description/rviz/urdf.rviz __name:=rviz __log:=/Users/Alex/.ros/log/5440e8d0-8b02-11eb-849f-8c8590d4cfc8/rviz-3.log].
log file: /Users/Alex/.ros/log/5440e8d0-8b02-11eb-849f-8c8590d4cfc8/rviz-3*.log
Initiating shutdown!
================================================================================
[rviz-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

the `otool -L` for the `librospack.dylib`:

```
(ros1) Alex3-2:catkin_ws Alex$ otool -L /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib
/Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib:
	@rpath/librospack.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtinyxml2.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	@rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)

```

Should I setup a different issue for this and keep this one closed (as rqt) starts?

sonelu, 2021-03-22:
Oh, and I forgot to say that `rqt` menu is not accessible until until you task-switch from `rqt` and come back. It's not a big deal and I remember seen that somewhere on ROS Answers.

sonelu, 2021-03-22:
I've downgraded to the build 5 of `rospack` and otool for this is:

```
(ros1) Alex3-2:~ Alex$ otool -L /Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib 
/Users/Alex/anaconda3/envs/ros1/lib/librospack.dylib:
	@rpath/librospack.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtinyxml2.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	@rpath/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libpython3.8.dylib (compatibility version 3.8.0, current version 3.8.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

so the link to `libpython3.8.dylib` needs to be there.

wolfv, 2021-03-22:
I am wondering if the fix with removing python linking was too much? maybe just allowing undefined lookup stuff does the job?

sonelu, 2021-03-22:
It's actually even more strange:

1. now that I downgraded to the previous build of `rospack` I can... run `rqt`. Seriously, with the previous build `rqt` runs and I can start various plugins:

[image]

While upgrading earlier to `rospack` build 6 I noticed 2 other non-ROS packages that were updated, but I don't recall which. Maybe they were the ones that make the difference.  I'll see if I can find some logs.

2. BUT, now `rviz` doesn't show the main window. I can start a `.launch` file and all is reported to be OK, but there is no visualisation window.

3. on the other hand I can add now the RVIZ plugin in `rqt` and (although I have to configure it manually) that displays the robot OK:

[image]

I can also use the `joint_state_publisher' to control the joints:

[image]

Tobias-Fischer, 2021-03-22:
I pushed a new version of rospack that links against python, and it is working fine. I also pushed a new version of rqt, which seems to work fine. However, rviz still doesn't start. When was rviz last working for you, @sonelu? Could you try digging further whether it is actually related with the rqt stuff?

Tobias-Fischer, 2021-03-22:
Quickly checked - it seems to be related to this bug: https://github.com/RoboStack/ros-noetic/issues/58 - see my comment there

sonelu, 2021-03-22:
Actually the `rviz` problem is the same as in #69.

If I downgrade to 1.14.4 is works fine. Looking back I don't think I had 1.14.5 ever working (either because of short dumps or not displaying the visualisation window.

And I can still use `rqt`.

The full list of "rviz"-es and "qt"-s that seems to work now (I also have a full list of the environment that seems to work that I can post if you need):

```
(ros1) Alex3-2:~ Alex$ conda list rviz
# packages in environment at /Users/Alex/anaconda3/envs/ros1:
#
# Name                    Version                   Build  Channel
ros-noetic-librviz-tutorial 0.11.0           py38he3dbe5f_5    robostack
ros-noetic-rqt-rviz       0.6.1            py38h3b5c783_5    robostack
ros-noetic-rviz           1.14.4           py38h38af97b_3    robostack
ros-noetic-rviz-plugin-tutorials 0.11.0           py38he3dbe5f_5    robostack
ros-noetic-rviz-python-tutorial 0.11.0           py38h7e910e5_5    robostack
ros-noetic-rviz-visual-tools 3.9.1            py38he3dbe5f_5    robostack

```

```
(ros1) Alex3-2:~ Alex$ conda list qt
# packages in environment at /Users/Alex/anaconda3/envs/ros1:
#
# Name                    Version                   Build  Channel
pyqt                      5.12.3           py38h50d1736_7    conda-forge
pyqt-impl                 5.12.3           py38h721a93c_7    conda-forge
pyqt5-sip                 4.19.18          py38h5745d40_7    conda-forge
pyqtchart                 5.12             py38h721a93c_7    conda-forge
pyqtwebengine             5.12.1           py38h721a93c_7    conda-forge
qt                        5.12.9               h126340a_4    conda-forge
ros-noetic-python-qt-binding 0.4.3            py38he3dbe5f_5    robostack
ros-noetic-qt-dotgraph    0.4.2            py38h7e910e5_5    robostack
ros-noetic-qt-gui         0.4.2            py38he3dbe5f_5    robostack
ros-noetic-qt-gui-cpp     0.4.2            py38hb993ecd_6    robostack
ros-noetic-qt-gui-py-common 0.4.2            py38h7e910e5_5    robostack
ros-noetic-rqt-action     0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-bag        0.5.1            py38h7e910e5_5    robostack
ros-noetic-rqt-bag-plugins 0.5.1            py38h7e910e5_5    robostack
ros-noetic-rqt-common-plugins 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-console    0.4.11           py38h7e910e5_5    robostack
ros-noetic-rqt-dep        0.4.10           py38h7e910e5_5    robostack
ros-noetic-rqt-graph      0.4.14           py38h7e910e5_5    robostack
ros-noetic-rqt-gui        0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-gui-cpp    0.5.2            py38he3dbe5f_5    robostack
ros-noetic-rqt-gui-py     0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-image-view 0.4.16           py38he3dbe5f_5    robostack
ros-noetic-rqt-launch     0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-logger-level 0.4.11           py38h7e910e5_5    robostack
ros-noetic-rqt-moveit     0.5.9            py38h7e910e5_5    robostack
ros-noetic-rqt-msg        0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-nav-view   0.5.7            py38h7e910e5_5    robostack
ros-noetic-rqt-plot       0.4.13           py38h7e910e5_5    robostack
ros-noetic-rqt-pose-view  0.5.10           py38h7e910e5_5    robostack
ros-noetic-rqt-publisher  0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-py-common  0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-py-console 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-reconfigure 0.5.3            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-dashboard 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-monitor 0.5.13           py38h7e910e5_5    robostack
ros-noetic-rqt-robot-plugins 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-steering 0.5.12           py38h7e910e5_5    robostack
ros-noetic-rqt-runtime-monitor 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-rviz       0.6.1            py38h3b5c783_5    robostack
ros-noetic-rqt-service-caller 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-shell      0.4.10           py38h7e910e5_5    robostack
ros-noetic-rqt-srv        0.4.8            py38h7e910e5_5    robostack
ros-noetic-rqt-tf-tree    0.6.2            py38h7e910e5_5    robostack
ros-noetic-rqt-top        0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-topic      0.4.12           py38h7e910e5_5    robostack
ros-noetic-rqt-web        0.4.9            py38h7e910e5_5    robostack
```

jc-bao, 2021-03-24:
@sonelu Could you please tell me how do you run the rqt? After upgrade rospack, the error of python has gone, I can even see the logo of ROS after I run rqt. But no window pop up and the programjust stuck there. Do you have encountered the same problem? Any information will be highly appreciated, Thanks!

sonelu, 2021-03-24:
I don't see a ROS logo when I start `rqt`. There is just a window with the plugins - or empty if it is the first time running. If you already have plugins they may want to connect to `roscore` so you might need to start that in another terminal first, or better run a `.launch` file.

Have you compared the packages above with the ones in your environment? Is any different (make sure you check the build and the channel too)?

sonelu, 2021-03-24:
And one more thing, I mentioned that somewhere above: when starting `rqt` is not responding to mouse clicks. You need to task switch to another app (command-tab) and then return to `rqt` to have it working. Can you also try this?

jc-bao, 2021-03-24:
I have tried your advice, the package runs fine in other environment, but I cannot start anything related to rqt, like rqt_graph or rqt_robot_steering. Before the update, my rqt just quit with python error. But now, I can find it in dock and activity monitor, but no window pop up. Here is my rqt and rospack version.
```
ros-noetic-rospack        2.6.2            py38h38ba479_7    robostack
ros-noetic-rqt-action     0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-bag        0.5.1            py38h7e910e5_5    robostack
ros-noetic-rqt-bag-plugins 0.5.1            py38h7e910e5_5    robostack
ros-noetic-rqt-common-plugins 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-console    0.4.11           py38h7e910e5_5    robostack
ros-noetic-rqt-dep        0.4.10           py38h7e910e5_5    robostack
ros-noetic-rqt-graph      0.4.14           py38h7e910e5_5    robostack
ros-noetic-rqt-gui        0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-gui-cpp    0.5.2            py38he3dbe5f_5    robostack
ros-noetic-rqt-gui-py     0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-image-view 0.4.16           py38he3dbe5f_5    robostack
ros-noetic-rqt-launch     0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-logger-level 0.4.11           py38h7e910e5_5    robostack
ros-noetic-rqt-moveit     0.5.9            py38h7e910e5_5    robostack
ros-noetic-rqt-msg        0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-nav-view   0.5.7            py38h7e910e5_5    robostack
ros-noetic-rqt-plot       0.4.13           py38h7e910e5_5    robostack
ros-noetic-rqt-pose-view  0.5.10           py38h7e910e5_5    robostack
ros-noetic-rqt-publisher  0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-py-common  0.5.2            py38h7e910e5_5    robostack
ros-noetic-rqt-py-console 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-reconfigure 0.5.3            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-dashboard 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-monitor 0.5.13           py38h7e910e5_5    robostack
ros-noetic-rqt-robot-plugins 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-robot-steering 0.5.12           py38h7e910e5_5    robostack
ros-noetic-rqt-runtime-monitor 0.5.8            py38h7e910e5_5    robostack
ros-noetic-rqt-rviz       0.6.1            py38h3b5c783_5    robostack
ros-noetic-rqt-service-caller 0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-shell      0.4.10           py38h7e910e5_5    robostack
ros-noetic-rqt-srv        0.4.8            py38h7e910e5_5    robostack
ros-noetic-rqt-tf-tree    0.6.2            py38h7e910e5_5    robostack
ros-noetic-rqt-top        0.4.9            py38h7e910e5_5    robostack
ros-noetic-rqt-topic      0.4.12           py38h7e910e5_5    robostack
ros-noetic-rqt-web        0.4.9            py38h7e910e5_5    robostack
```
Here is what happened when I run `rqt`:
[image]
Python is running but no window pop up.
Thanks for your timely responding.

sonelu, 2021-03-25:
What I have different is the version of `rospack`. I have:

```
# Name                    Version                   Build  Channel
ros-noetic-rospack        2.6.2            py38h22c0035_5    robostack
```

Your version is the one that Tobias [pushed earlier](https://github.com/RoboStack/ros-noetic/issues/58#issuecomment-804414306). I actually haven't tested that new one on my system.

Try downgrading with:

conda install ros-noetic-rospack=2.6.2=py38h22c0035_5

(maybe use `--force-reinstall` if it doesn't let you use the above).

Can you also send the packages for Qt that you have installed to compare with my stack?

sonelu, 2021-03-25:
Also: which OS version you have? Catalina or Big Sur. And are you using an Intel or M1 Mac?

sonelu, 2021-03-25:
It could be because of Big Sur. I've seen packages that have different build rules for OS version 11. We can compare the whole conda environment if you want, but that might not be the that conclusive.

If you are up for it you could setup a separate volume where you install Catalina and check a fresh conda install there. I also do that: I have a volume with High Sierra because I'm using an external GPU (Thuderbolt 3) with an NVidia 1080 for DL work. And there are no drivers for NVidia after High Sierra...

Let me know if you would be interested in this, I could send you the procedure.

sonelu, 2021-03-25:
I've found this about Qt on Big Sur
https://bugreports.qt.io/browse/QTBUG-85546

Do you have any application that uses Qt that runs (nut just from ROS stack)?

traversaro, 2021-03-25:
Possibly related issue: https://github.com/conda-forge/qt-feedstock/issues/186 .

jc-bao, 2021-03-26:
@sonelu Yes, I have just tried a small project I previously built on Qt and it works fine. I find that the Qt version I installed by Homebrew is 5.15.2, in which the Qt issue is resloved.(https://bugreports.qt.io/browse/QTBUG-87014)
But when it comes to Conda environment, the qt version is 5.12.9 that leads to the error.
So is it possible for me to install another version in conda?

sonelu, 2021-03-26:
I see @Tobias-Fischer is on the list of maintainers for the feedstock for Qt. Maybe he can help here.

Tobias-Fischer, 2021-03-26:
There is this PR: https://github.com/conda-forge/qt-feedstock/pull/189

If someone could try building QT on OSX that'd be great. I'm already building the aarch builds and I don't have time to maintain OSX as well ..

sonelu, 2021-03-26:
Do you want it on Catalina or Big Sur?

Tobias-Fischer, 2021-03-26:
I think @hmaarrfk tried to build on Big Sur and it didn't work. So ideally building on Catalina and then test whether it works on Big Sur.

Tobias-Fischer, 2021-03-26:
If you'd be happy to help out with this that'd be great @sonelu . It seems indeed that @jc-bao's problem is because of Big Sur and could be fixed by https://github.com/conda-forge/qt-feedstock/pull/189. If you'd be happy to try building the new version of QT on OSX, let's keep chatting in https://github.com/conda-forge/qt-feedstock/pull/189 what the best way to do that would be.

wolfv, 2021-03-27:
I have been checking rqt on OS X (Big Sur) and libqt_gui_cpp_sip crashes for me:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff204ef462 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff2051d610 libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20470720 libsystem_c.dylib`abort + 120
    frame #3: 0x000000011c2e6ebc libpython3.8.dylib`fatal_error + 60
    frame #4: 0x000000011c2e6fe3 libpython3.8.dylib`Py_FatalError + 19
    frame #5: 0x000000011c19e662 libpython3.8.dylib`PyModule_Create2 + 50
    frame #6: 0x000000011bb087d8 libqt_gui_cpp_sip.so`PyInit_libqt_gui_cpp_sip + 28
    frame #7: 0x00000001001cf53c python`_PyImport_LoadDynamicModuleWithSpec + 572
    frame #8: 0x00000001001ce637 python`_imp_create_dynamic + 183
    frame #9: 0x00000001000903b4 python`cfunction_vectorcall_FASTCALL + 84
    frame #10: 0x000000010002eef9 python`PyVectorcall_Call + 121
 ....
```

with Fatal Python error: _PyInterpreterState_Get(): no current thread state
Python runtime state: unknown

I am going to check if there is something we can do that doesn't require rebuilding Qt!

wolfv, 2021-03-27:
ok that was a false alarm. I had build number 5 installed. Latest rqt doesn't have this issue, but I cannot see the window either.

wolfv, 2021-03-27:
If I run rqt apps "standalone", they start to a blank screen. (e.g. `rosrun rqt_topic rqt_topic`).

hmaarrfk, 2021-03-27:
I made a PR to qt for big sur issues. I think it might be related.

wolfv, 2021-03-27:
I am trying to build Qt right now! Thanks @hmaarrfk

hmaarrfk, 2021-03-27:
So am I 😂

Tobias-Fischer, 2021-03-27:
Thanks a lot @wolfv @hmaarrfk :)! I hope it's not too tricky.

Ghanmohan, 2021-06-15:
@sonelu sir can you pls tell how to run the the launch file through roslaunch ?
cause it is showing the package is neither a launch file, also I was using this package previously so it is a launch file.
Thank for answering

sonelu, 2021-06-15:
@Ghanmohan Which package are you referring to? And what exactly are you trying to launch with `roslaunch`?

Ghanmohan, 2021-06-15:
@sonelu I am having a rviz launch file
The command goes like this roslaunch foldername filename.launch
I was using the above command In ubuntu. what should I do to make it run in ROS on Mac?

sonelu, 2021-06-15:
Assuming you have installed the full ROS stack (ex `conda install ros-noetic-desktop-full`) and you have `source`'d your catkin workspace where you have your package with the launch file, you should be able to run it the same way.

Can you run `rviz` directly from the command line? If not you either have not installed all the ROS packages or the installation is not sourced (ex. the conda environment is not activated).

If you can run `rviz` but you cannot `roslaunch` your launch file, then most likely you have not `source` your catkin workspace where you have your package with the launch file.

Ghanmohan, 2021-06-16:
@sonelu when I am sourcing my file it is showing that the file or directory is not present.
I am writing the code ...devel/setup.bash
Please tell what I have to do to correct it.

binnurr, 2022-05-13:
Hello, this issue persists on my side with qt version 5.12.9. how can I upgrade that package?

traversaro, 2022-05-13:
Hi @binnurr, can you open a new issue including:
* the exact error that you are experiencing 
* the output of `mamba list` 
* the output of `mamba info`

This would be helpful for people to help you, thanks!