RoboStack/ros-noetic #59: mavros geographiclib datasets are not being installed

Tobias-Fischer, 2021-03-03:
We need to patch the `mavros/scripts/install_geographiclib_datasets.sh` file I think. I am also not sure whether the conda-forge version of geographiclib is being shipped with the tools to download datasets.

```
2021-02-17T10:38:25.4743676Z -- No geoid model datasets found. This will result on a SIGINT! Please execute the script install_geographiclib_dataset.sh in /mavros/scripts
2021-02-17T10:38:25.4745543Z -- No gravity field model datasets found. Please execute the script install_geographiclib_dataset.sh in /mavros/scripts
2021-02-17T10:38:25.4755320Z -- No magnetic field model datasets found. Please execute the script install_geographiclib_dataset.sh in /mavros/scripts
```

Tobias-Fischer, 2021-03-26:
Unless someone needs it, let's mark as wontfix

closed 2021-04-26 by Tobias-Fischer

GeoRos, 2024-04-04:
Three years later, this is actually needed. Has anyone figured this out?

traversaro, 2024-04-04:
@GeoRos can you open a new issue following the template? Thanks!