RoboStack/ros-noetic #63: rviz rendering issue on MacOS Catalina

sonelu, 2021-03-05:
Hi guys, another bizarre bug with the MacOS installation.

Context:

MacOS Catalina 10.15.7
Conda: 4.9.2
Python: 3.8.8

Installed ros-noetic-desktop-full in it's own environment.

I'm starting a visualization with rviz using a launch file with a URDF robot description. When rviz starts the robot is shown normally on the vertical, but the horizontal axis seems to be squeezed:

[image]

If I reselect the view in the Views panel:

[image]

then the robot is rendered correctly:

[image]

Thanks,

Alex.

Tobias-Fischer, 2021-03-07:
Hi Alex,

Thanks again for the report! In this particular case, I'm tempted to attach a `wontfix` label. I've seen these issues before in other similar projects on "non-native" platforms (e.g. iCubGui on Windows @traversaro), and I have never found a proper solution for it. Seeing that it's just a "minor inconvenience" having to select a panel / resize the panel, that it is potentially super trick to track down (is it a bug in rviz? OGRE? QT? etc.) and that there are lots of other bugs and issues to be addressed, I think other issues are higher in the priority list for now.
That is not to stop you from having a go at it though, Alex! Feel free to play around - but honestly I wouldn't even know where to start.

What do you guys think @traversaro @wolfv?

Tobi

sonelu, 2021-03-08:
@Tobias-Fischer I'm perfectly fine with this.

wolfv, 2021-03-08:
I think we can keep this issue around for other users to see (in the future). It's likely we won't have the time to fix this ourselves, but it would be great to report this upstream. Maybe it's an easy fix for upstream?!

Also, I have noticed a similar problem on Linux where a resize after starting RViz fixes the size of the 3D viewport.

@sonelu do you want to file an issue with RViz? Maybe pointing to this one?

sonelu, 2021-03-10:
Guys, after the last build rviz is also broken. After I did a:

`conda update --all`

rviz is also broken now. It simply doesn't show any window when launched with a `launch` file. The PID reported by the `launch` file seems to be running ok.

If I run rviz on it's own there is just a small empty window that is shown and the console outputs:

```
(robostack) Alex3-2:catkin_ws Alex$ rviz
[ INFO] [1615398617.967631000]: rviz version 1.14.5
[ INFO] [1615398617.967680000]: compiled against Qt version 5.12.9
[ INFO] [1615398617.967690000]: compiled against OGRE version 1.10.12 (Xalafu)
```

[image]

Running rviz with `-v` or `--verbose` doesn't seem to make any difference.

wolfv, 2021-03-10:
Argh

Tobias-Fischer, 2021-03-10:
Could be the new MESA stuff they have added upstream .. 
@sonelu would you be comfortable building rviz locally (see https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md) and moving [this patch](https://github.com/RoboStack/ros-noetic/blob/558423b6d11926a604813335be915c28f9fa5f03/patch/ros-noetic-rviz.win.patch#L49-L63) from rviz.win.patch into rviz.patch so it is applied on OSX, too? And then see whether that solves this issue?

Tobias-Fischer, 2021-03-10:
Also, before doing that, could you try `rviz --opengl 310` and `rviz --opengl 210`

sonelu, 2021-03-10:
`rviz --opengl 310` and `rviz --opengl 210` both behave the same (the blank window and program stuck with the beach ball).

I'll try the local build tomorrow and report on the progress.

jc-bao, 2021-03-11:
Hi, I slove this problem just by down grade rviz by
`conda install ros-noetic-rviz=1.14.4`
Hopefully this will help😊

wolfv, 2021-03-11:
We should ping upstream rviz to see what they think, too. I can also try locally today!

sonelu, 2021-03-11:
I confirm that downgrading to 1.14.4 works (the skewed view is still there, but at least rviz starts and I can control joints).

@jc-bao do you also see this bug with in the stretched view on your machine?

traversaro, 2021-03-11:
@wolfv @Tobias-Fischer could it make sense to delete the broken macos build from the robostack for the time being? Or this could be creating worse problems?

traversaro, 2021-03-11:
BTW, perhaps it could make sense to change the title name to be more descriptive of the problem (or actually open a new one, as indeed both the crash and rendering problems are issues on their own.

sonelu, 2021-03-11:
> Could be the new MESA stuff they have added upstream ..
> @sonelu would you be comfortable building rviz locally (see https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md) and moving [...]

@Tobias-Fischer I've tried this but I'm not sure if there are some missing steps in order to be able to use the new environment.

Here is what I did:

```
conda create -n rs-test python=3.8
conda activate rs-test
```

these are not needed as they are already done due to previous environment setup:

```
conda config --append channels defaults
conda config --add channels conda-forge
conda config --add channels robostack
conda config --set channel_priority strict 
```

Then:

```
conda install pip conda-build anaconda-client mamba conda catkin_pkg ruamel_yaml rosdistro empy networkx requests boa
pip install git+https://github.com/RoboStack/vinca.git@master --no-deps
cd
git clone https://github.com/RoboStack/ros-noetic.git
cd ros-noetic
cp vinca_osx.yaml vinca.yaml
```

I have commented in `vinca.yaml`:

```
skip_existing:
#  - https://conda.anaconda.org/robostack/
```

I have added the [patch](https://github.com/RoboStack/ros-noetic/blob/558423b6d11926a604813335be915c28f9fa5f03/patch/ros-noetic-rviz.win.patch#L49-L63) into the `patch/ros-noetic-rviz.patch` as requested.

Then I run:

```
vinca --multiple
cp recipes/ros-noetic-rviz/recipe.yaml recipe.yaml
boa build .
```

After build rviz is not available as a command. Is there an activation step that I need to do after the build?

wolfv, 2021-03-11:
after that you need to install the rviz package from the local channel `conda/mamba install ros-noetic-rviz -c local`

wolfv, 2021-03-11:
FYI the build artifacts are found in `$ROOT_PREFIX/conda-bld/osx-64/...`

sonelu, 2021-03-11:
Ok, I've done that and rviz behaves the same (hangs and shows the blank window).

But just to make sure that the correct channel was used, should in the bellow:

```
(rs-test) Alex3-2:ros-noetic Alex$ conda install ros-noetic-rviz -c local
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /Users/Alex/anaconda3/envs/rs-test

  added / updated specs:
    - ros-noetic-rviz

The following NEW packages will be INSTALLED:

  ...
  ros-noetic-roswtf  robostack/osx-64::ros-noetic-roswtf-1.15.9-py38h7e910e5_5
  ros-noetic-rviz    robostack/osx-64::ros-noetic-rviz-1.14.5-py38he3dbe5f_5
  ros-noetic-sensor~ robostack/osx-64::ros-noetic-sensor-msgs-1.13.1-py38h7e910e5_5
  ...

Proceed ([y]/n)? 
```

Is `robostack/osx-64::ros-noetic-rviz-1.14.5...` correct or should be `local/osx-64::ros-noetic-rviz-1.14.5...` ?

If I run `conda list` I still see it with the robostack channel:

```
(rs-test) Alex3-2:ros-noetic Alex$ conda list
# packages in environment at /Users/Alex/anaconda3/envs/rs-test:
#
# Name                    Version                   Build  Channel
...
ros-noetic-roswtf         1.15.9           py38h7e910e5_5    robostack
ros-noetic-rviz           1.14.5           py38he3dbe5f_5    robostack
ros-noetic-sensor-msgs    1.13.1           py38h7e910e5_5    robostack
...

```

jc-bao, 2021-03-11:
> I confirm that downgrading to 1.14.4 works (the skewed view is still there, but at least rviz starts and I can control joints).
> 
> @jc-bao do you also see this bug with in the stretched view on your machine?

Year, my view was also stretched. Besides, I encountered another issue that my `rqt`, as well as `rqt_ros`, seem not to work now. Can you run `rqt` in macOS?

Tobias-Fischer, 2021-03-11:
It would be great if we could keep the rviz&rqt issues separated - please see https://github.com/RoboStack/ros-noetic/issues/58 for rqt related issues.

sonelu, 2021-03-18:
I've put a question on ROS Answers for the time being.
https://answers.ros.org/question/374159/rviz-visualisation-skewed-on-macos/

Let's see what comes out of it.

closed 2021-04-26 by Tobias-Fischer