RoboStack/ros-noetic #65: Reccomend to add channels only on specified environment?

traversaro, 2021-03-08:
In the README we currently suggest users to add `robostack` to their global channel list ` ~/.condarc`. Could it make sense that we suggest to specify options (such as `conda config --env --add channels glotzer`, see https://stackoverflow.com/questions/40616381/can-i-add-a-channel-to-a-specific-conda-environment) to only add the channel to a specific environment?

Tobias-Fischer, 2021-03-11:
I think it's a good idea. What do you think, @wolfv?

wolfv, 2021-03-12:
I think that would make `conda create -n myrosenv ros-noetic-desktop` more annoying, wouldn't it? because when you do that command from teh base env, you need to explicitly give `-c robostack`.

wolfv, 2021-03-12:
sorry, accidental close!

tsoud, 2021-04-20:
Hi everyone! First of all, thanks for creating this awesome and much-needed channel for ROS on conda. As an ROS newbie and regular conda user, I can't  stress enough how much easier this has made my life.

I did come across what seems to be an edge-case related to channel configuration. I  decided to bring it up here instead of creating a new issue because it seemed relevant:
After uninstalling ROS and all its dependencies from my system, I created the a new environment for it and followed your installation directions to the letter. Things were going smoothly until I hit a snag installing Tensorflow-GPU from the anaconda repo using `conda install -c anaconda`. The installation would fail with compatibility errors after spending a very long time resolving the packages. I spent hours trying to fix this (`pip install` and manually add libraries/paths, etc.) before recalling that setting `channel_priority` to `strict` was the likely culprit. Adding the `anaconda` channel and changing `channel_priority` to `flexible` fixed it.

It will probably help to add some notes in the documentation like "If you encounter issues installing a specific conda package into the environment try changing channel priority to flexible..." or something similar (feel free to use this case for illustration). I do agree with @traversaro that only adding the channel to the ROS environment and setting up a specific `.condarc` for it is a good idea. If someone sets this up the configuration globally and forgets, they could be stuck for hours trying to find out why installations in their other non-ROS environments are failing.

traversaro, 2021-04-20:
Thanks for the feedback on this @tsoud (and for pinging the issue, as I had to forgot to reply to @wolfv).

> I think that would make `conda create -n myrosenv ros-noetic-desktop` more annoying, wouldn't it? because when you do that command from teh base env, you need to explicitly give `-c robostack`.

Yes, but in experience expert users are able to do customize the installation instructions in an way that it is convenient for them. My main concern here is how less experience users that do not really have a lot of experience with Conda, and that they do not realize that by executing that command they are perturbing the global state of their system, rather then just doing changes on their environment with no risk of interfering with the rest of the system. In any case, unless I am missing something we do not have the `conda create -n myrosenv ros-noetic-desktop` or similar command in the README.

Tobias-Fischer, 2021-04-21:
Seeing that someone actually encountered a problem, I think it would be good to address this.

@traversaro @tsoud - would one of you be happy to open a PR addressing this?

tsoud, 2021-04-21:
> Seeing that someone actually encountered a problem, I think it would be good to address this.
> 
> @traversaro @tsoud - would one of you be happy to open a PR addressing this?

Happy to do it but it will be a few days to get around to it.

traversaro, 2021-05-12:
PR opened at https://github.com/RoboStack/ros-noetic/pull/123 .

Tobias-Fischer, 2021-05-14:
@wolfv - happy to merge, how about you?

closed 2021-05-14 by wolfv