RoboStack/ros-noetic #67: Building the UR driver package

clemense, 2021-03-09:
Hello,
I'm trying to create packages for the UR driver: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver
What's the best way to go about it?
I only found a screenshot in a medium post that shows how to build custom ROS packages: https://miro.medium.com/max/1400/1*D-2IRRYeUzMprK5M12ayqg.png
Unfortunately, I can't reproduce these steps for the UR package without errors.
Are there more detailed instructions (apart from the contributing.md file)? Or should I rather not go down this road at this point in time?

Thanks!

Tobias-Fischer, 2021-03-09:
@wolfv: We should think about a process to make this more automated - in this way people could directly package their stuff here rather than having to package it in the main ROS repos via bloom :)

Tobias-Fischer, 2021-03-10:
I just pushed some packages for Jackal using this method: https://github.com/RoboStack/ros-noetic/commit/2cad45ee207fc782acec3f7ed56393a3e1ef232e

traversaro, 2021-03-11:
Note that that package is not release in ROS directly via bloom due to the fact that depend on a fork or an upstream dependency:
* https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/254
* https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/248
* https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/84
* https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/97

Tobias-Fischer, 2021-03-16:
Hi @clemense - did you give this a go? Would be great if you let us know how things went.

clemense, 2021-03-17:
Hi,
Sorry, we went the more conventional route. But I would like to revisit this in the future, once things are more streamlined.

AndrewJSchoen, 2021-05-13:
I am attempting to run this workflow to generate the *Universal_Robots_ROS_Driver* on macOS, however, I am encountering issues. What I first attempted to do was build the package with boa, as described above:

```
cd Universal_Robots_ROS_Driver 
vinca -p "**/package.xml" -d ../ros-noetic
boa build .
```

However, I get an error when trying to build `ros-noetic-ur-calibration`:
```
...
Finalizing build for ros-noetic-ur-calibration
Finalizing host for ros-noetic-ur-calibration
Mamba failed to solve:
 - ros-noetic-catkin
 - eigen
 - ros-noetic-roscpp
 - ros-noetic-ur-client-library
 - yaml-cpp
 - eigen
 - ros-noetic-roscpp
 - ros-noetic-ur-client-library
 - yaml-cpp
 - ros-noetic-rosunit
 - libcxx >=11.1.0

with channels:
 - robostack
 - conda-forge
 - defaults

The reported errors are:
   Encountered problems while solving:
     - nothing provides requested ros-noetic-ur-client-library
     - nothing provides requested ros-noetic-ur-client-library
```

This led me to looking into the `ros-noetic-ur-client-library`, since it seemed to be the missing piece. However, it fails to compile on macOS due to a number of issues:
 - Usage of `#include <endian.h>` instead of a portable/cross-platform approach (fixed in my fork, below)
 - Undeclared Identifier `TCP_QUICKACK` from `/src/comm/tcp_socket.cpp` (This is not available on macOS)
 - Finally, an error I am not too sure about how to fix: 
 
   ```
   Undefined symbols for architecture x86_64:
     "console_bridge::log(char const*, int, console_bridge::LogLevel, char const*, ...)", referenced from:
      urcl::comm::TCPSocket::setup(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&, int) in tcp_socket.cpp.o
      urcl::comm::TCPSocket::getIP() const in tcp_socket.cpp.o
      urcl::comm::TCPSocket::write(unsigned char const*, unsigned long, unsigned long&) in tcp_socket.cpp.o
      urcl::comm::TCPServer::init() in tcp_server.cpp.o
      urcl::comm::TCPServer::bind() in tcp_server.cpp.o
      urcl::comm::TCPServer::startListen() in tcp_server.cpp.o
      urcl::comm::TCPServer::~TCPServer() in tcp_server.cpp.o
      ...
   ld: symbol(s) not found for architecture x86_64
   clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
   ninja: build stopped: subcommand failed.
   ```

Like I mentioned, I have fixed a couple of issues in building the library in [this fork](https://github.com/AndrewJSchoen/Universal_Robots_Client_Library) (boost branch)
These changes include:
 - A generalized lookup instead of `endian.h` that _should_ be more cross-platform.
 - I removed the usage of `TCP_QUICKACK` (this one I don't know much about and this might be a terrible idea)

If there is any interest, some of this might be helpful in getting it to work on other platforms than linux (right now the only supported one).

Tobias-Fischer, 2021-05-13:
Hi @AndrewJSchoen - check out https://github.com/AndrewJSchoen/Universal_Robots_Client_Library/pull/1which should resolve your issues.

Tobias-Fischer, 2021-05-13:
Please consider opening a pull request upstream with your changes :)

Tobias-Fischer, 2021-05-13:
And also please open a pull request adding ur-client-library and the other packages into RoboStack (additional_recipes)

Tobias-Fischer, 2021-06-01:
So I just had another look at this. @traversaro is right, they have not bloomed/released a proper version of the UR packages, neither for ROS Melodic nor Noetic. They have further not even release tags in their repositories, and rely on weird forks of forks. Therefore, I do not think that it makes sense to push on releasing RoboStack packages before those issues are resolved upstream.

@AndrewJSchoen - I have triggered a build of the ur_client_library for linux+osx: https://github.com/RoboStack/ros-noetic/commit/6aee2cd0311ebe70219cd32bf85e67a313bb7e0d
You should still be able to build the packages locally in a recipe as per https://github.com/RoboStack/ros-noetic/issues/67#issuecomment-794611797

Tobias-Fischer, 2021-06-07:
Great to hear that this worked for you @AndrewJSchoen. Is it okay to close this issue? It is very hard for us to package the UR driver seeing that it is not released properly.

AndrewJSchoen, 2021-06-07:
@Tobias-Fischer I think we can close until anything changes upstream. Thanks for the help!

closed 2021-06-07 by Tobias-Fischer