RoboStack/ros-noetic #71: CUDA incompatible error

zhangxiaochuan, 2021-03-12:
Hi, I tried to install ros-noetic through `conda install -c robostack ros-noetic-desktop-full`, but failed with the following error messages:
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your CUDA driver:

  - feature:/linux-64::__cuda==11.2=0

Your installed CUDA driver is: 11.2
```

Are there any suggestions?

Tobias-Fischer, 2021-03-12:
Hi,
Could you please try installing `ros-noetic-desktop-full` in a new environment and see whether the error occurs as well (following the [README](https://github.com/RoboStack/ros-noetic))?

Also, the output given by `conda` is not very informative compared to what you get with `mamba`. Could you please `conda install mamba` and try again with `mamba install ros-noetic-desktop-full`?

Tobias-Fischer, 2021-03-16:
Closing due to inactivity - feel free to reopen @zhangxiaochuan

closed 2021-03-16 by Tobias-Fischer