RoboStack/ros-noetic #78: Add slam-toolbox

Tobias-Fischer, 2021-03-15:
Interesting error for slam_toolbox - any ideas @wolfv @traversaro?

```

2021-03-15T01:17:54.5893235Z CMake Error at C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:44 (list):
2021-03-15T01:17:54.5894383Z   Maximum recursion depth of 1000 exceeded
2021-03-15T01:17:55.3888693Z Call Stack (most recent call first):
2021-03-15T01:17:55.3890230Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
2021-03-15T01:17:55.3893465Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
....
2021-03-15T01:17:55.5097622Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
2021-03-15T01:17:55.5098505Z -- Configuring incomplete, errors occurred!
2021-03-15T01:17:55.5099504Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
2021-03-15T01:17:55.5100431Z See also "C:/Miniconda/conda-bld/ros_1615769769608/work/build/CMakeFiles/CMakeOutput.log".
2021-03-15T01:17:55.5101808Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
2021-03-15T01:17:55.5102867Z See also "C:/Miniconda/conda-bld/ros_1615769769608/work/build/CMakeFiles/CMakeError.log".
2021-03-15T01:17:55.5104013Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/platform/windows.cmake:47 (_add_library)
2021-03-15T01:17:55.5105125Z   C:/Miniconda/conda-bld/ros_1615769769608/_build_env/Library/share/cmake-3.19/Modules/FindThreads.cmake:237 (add_library)
2021-03-15T01:17:55.5106277Z   C:/Miniconda/conda-bld/ros_1615769769608/_build_env/Library/share/cmake-3.19/Modules/FindGTest.cmake:213 (find_package)
2021-03-15T01:17:55.5107740Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/test/gtest.cmake:340 (find_package)
2021-03-15T01:17:55.5109039Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/all.cmake:164 (include)
2021-03-15T01:17:55.5110155Z   C:/Miniconda/conda-bld/ros_1615769769608/_h_env/Library/share/catkin/cmake/catkinConfig.cmake:20 (include)
2021-03-15T01:17:55.5111057Z   CMakeLists.txt:13 (find_package)
```

Relevant catkin code (line 47 is the `_add_library` line):
```
if(BUILD_SHARED_LIBS)
  if(WIN32)
    function(add_library library)
      # Check if its an external, imported library (e.g. boost libs via cmake module definition)
      list(FIND ARGN "IMPORTED" FIND_IMPORTED)
      list(FIND ARGN "ALIAS" FIND_ALIAS)
      list(FIND ARGN "INTERFACE" FIND_INTERFACE)
      _add_library(${ARGV0} ${ARGN})
      if(${FIND_IMPORTED} EQUAL -1 AND ${FIND_ALIAS} EQUAL -1 AND ${FIND_INTERFACE} EQUAL -1)
        set_target_properties(${ARGV0}
          PROPERTIES
              RUNTIME_OUTPUT_DIRECTORY ${CATKIN_DEVEL_PREFIX}/${CATKIN_GLOBAL_BIN_DESTINATION}
              RUNTIME_OUTPUT_DIRECTORY_DEBUG ${CATKIN_DEVEL_PREFIX}/${CATKIN_GLOBAL_BIN_DESTINATION}
              RUNTIME_OUTPUT_DIRECTORY_RELEASE ${CATKIN_DEVEL_PREFIX}/${CATKIN_GLOBAL_BIN_DESTINATION}
              RUNTIME_OUTPUT_DIRECTORY_MINSIZEREL ${CATKIN_DEVEL_PREFIX}/${CATKIN_GLOBAL_BIN_DESTINATION}
              RUNTIME_OUTPUT_DIRECTORY_RELWITHDEBINFO ${CATKIN_DEVEL_PREFIX}/${CATKIN_GLOBAL_BIN_DESTINATION}
              LIBRARY_OUTPUT_DIRECTORY ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              LIBRARY_OUTPUT_DIRECTORY_DEBUG ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              LIBRARY_OUTPUT_DIRECTORY_RELEASE ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              LIBRARY_OUTPUT_DIRECTORY_MINSIZEREL ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              LIBRARY_OUTPUT_DIRECTORY_RELWITHDEBINFO ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              ARCHIVE_OUTPUT_DIRECTORY ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              ARCHIVE_OUTPUT_DIRECTORY_DEBUG ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              ARCHIVE_OUTPUT_DIRECTORY_RELEASE ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              ARCHIVE_OUTPUT_DIRECTORY_MINSIZEREL ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
              ARCHIVE_OUTPUT_DIRECTORY_RELWITHDEBINFO ${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_LIB_DESTINATION}
        )
      endif()
    endfunction()
  endif()
endif()
```

traversaro, 2021-03-15:
Related issues: 
* https://github.com/microsoft/vcpkg/issues/8047
* https://github.com/microsoft/vcpkg/issues/11307

It is possible that either: 
* `slam_toolbox` or some included CMake module is re-defining `add_library`?
* For some reason the `share/catkin/cmake/platform/windows.cmake` file is included two times?

Tobias-Fischer, 2021-03-16:
So this error went away when not calling `catkin_package` multiple times.

Closing this issue anyhow as wontfix as there are many unix-specific things in there, which I can't be bothered to fix (getrlimit, setrlimit etc.).

closed 2021-03-16 by Tobias-Fischer