RoboStack/ros-noetic #85: Unable to register with master node [http://127.0.0.1:11311]: master may not be running yet. Will keep trying.

jackfrisht, 2021-03-17:
I followed your instruction to build the environment and select python 3.6 on Windows 10, then installed Intel Real Sense SDK  pyrealsense2, but met the error below, 
```
NOHA :: multistream:::open :: added 1000ms sleep !
<pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 30fps Z16>
<pyrealsense2.video_stream_profile: Color(0) 640x480 @ 30fps BGR8>
NOHA :: multistream:::open :: added 1000ms sleep !
Unable to register with master node [http://127.0.0.1:11311]: master may not be running yet. Will keep trying.
```
It seems the roscore didn't launch correctly(http://wiki.ros.org/ROS/Troubleshooting)

The startup information is below 

````
Microsoft Windows [Version 10.0.19042.867]
(c) 2020 Microsoft Corporation。

C:\Users\Javier\Documents\com>conda activate robostack

C:\Users\Javier\Documents\com>SET DISTUTILS_USE_SDK=1

C:\Users\Javier\Documents\com>SET MSSdk=1

C:\Users\Javier\Documents\com>SET "VS_VERSION=16.0"

C:\Users\Javier\Documents\com>SET "VS_MAJOR=16"

C:\Users\Javier\Documents\com>SET "VS_YEAR=2019"

C:\Users\Javier\Documents\com>set "MSYS2_ARG_CONV_EXCL=/AI;/AL;/OUT;/out"

C:\Users\Javier\Documents\com>set "MSYS2_ENV_CONV_EXCL=CL"

C:\Users\Javier\Documents\com>set "PY_VCRUNTIME_REDIST=\bin\vcruntime140.dll"

C:\Users\Javier\Documents\com>set "CXX=cl.exe"

C:\Users\Javier\Documents\com>set "CC=cl.exe"

C:\Users\Javier\Documents\com>set "VSINSTALLDIR="

C:\Users\Javier\Documents\com>for /F "usebackq tokens=*" %i in (`vswhere.exe -nologo -products * -version [16.0,17.0) -property installationPath`) do (set "VSINSTALLDIR=%i\" )

C:\Users\Javier\Documents\com>(set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" )

C:\Users\Javier\Documents\com>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" (for /F "usebackq tokens=*" %i in (`vswhere.exe -nologo -products * -requires Microsoft.VisualStudio.Component.VC.v142.x86.x64 -property installationPath`) do (set "VSINSTALLDIR=%i\" ) )

C:\Users\Javier\Documents\com>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\" )

C:\Users\Javier\Documents\com>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\" )

C:\Users\Javier\Documents\com>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\" )

C:\Users\Javier\Documents\com>if not exist "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" (set "VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\" )

C:\Users\Javier\Documents\com>IF NOT "" == "" (
set "INCLUDE=;"
 set "LIB=;"
 set "CMAKE_PREFIX_PATH=;C:/Users/Javier/.conda/envs/robostack/Library"
)

C:\Users\Javier\Documents\com>call :GetWin10SdkDir

C:\Users\Javier\Documents\com>call :GetWin10SdkDirHelper HKLM\SOFTWARE\Wow6432Node  1>nul 2>&1

C:\Users\Javier\Documents\com>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE\Wow6432Node  1>nul 2>&1

C:\Users\Javier\Documents\com>if errorlevel 1 call :GetWin10SdkDirHelper HKLM\SOFTWARE  1>nul 2>&1

C:\Users\Javier\Documents\com>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE  1>nul 2>&1

C:\Users\Javier\Documents\com>if errorlevel 1 exit /B 1

C:\Users\Javier\Documents\com>exit /B 0

C:\Users\Javier\Documents\com>for /F %i in ('dir /ON /B "C:\Program Files (x86)\Windows Kits\10\\include\10.*"') DO (SET WindowsSDKVer=%~i )

C:\Users\Javier\Documents\com>(SET WindowsSDKVer=10.0.18362.0 )

C:\Users\Javier\Documents\com>if errorlevel 1 (echo "Didn't find any windows 10 SDK. I'm not sure if things will work, but let's try..." )  else (echo Windows SDK version found as: "10.0.18362.0" )
Windows SDK version found as: "10.0.18362.0"

C:\Users\Javier\Documents\com>IF 2019 GEQ 2019 (
set "CMAKE_GEN=Visual Studio 16 2019"
 IF "win-64" == "win-64" (
set "BITS=64"
 set "CMAKE_PLAT=x64"
)  ELSE (
set "BITS=32"
 set "CMAKE_PLAT=Win32"
)
)  ELSE (IF "win-64" == "win-64" (
set "CMAKE_GEN=Visual Studio 16 2019 Win64"
 set "BITS=64"
)  else (
set "CMAKE_GEN=Visual Studio 16 2019"
 set "BITS=32"
) )

C:\Users\Javier\Documents\com>pushd C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\

C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise>CALL "VC\Auxiliary\Build\vcvars64.bat" -vcvars_ver=14.28 10.0.18362.0
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.6
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

(robostack) C:\Users\Javier\Documents\com>python RunRos_yolo.py
NOHA :: multistream:::open :: added 1000ms sleep !
<pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 30fps Z16>
<pyrealsense2.video_stream_profile: Color(0) 640x480 @ 30fps BGR8>
NOHA :: multistream:::open :: added 1000ms sleep !
Unable to register with master node [http://127.0.0.1:11311]: master may not be running yet. Will keep trying.

Can you please tell me how to manually start the roscore in conda.

Tobias-Fischer, 2021-03-17:
Hi there,

Many thanks for your interest in our work! I recommend using Python 3.8, as all the latest builds are based on Python 3.8. You can simply run `roscore` in a command terminal.

jackfrisht, 2021-03-17:
@Tobias-Fischer Thank you very much for your assist. Because the pyrealsense2 only supports up to Python 3.7, I can only use Python 3.6.

When I trying to call roscore, there is an alert comes.
[image]
The chinese means cannot locate entry point
```
(robostack) C:\Users\Javier\Documents\com>roscore
... logging to C:\Users\Javier\.ros\log\b5fcea2e-86c9-11eb-96ca-ac1203569b44\roslaunch-DESKTOP-QKLO6UV-14520.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://127.0.0.1:51597/
ros_comm version 1.15.8

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES

auto-starting new master
process[master]: started with pid [17912]
ROS_MASTER_URI=http://127.0.0.1:11311/
setting /run_id to b5fcea2e-86c9-11eb-96ca-ac1203569b44
process[rosout-1]: started with pid [4200]
started core service [/rosout]
```

Tobias-Fischer, 2021-03-17:
Yes - this is a known problem for our old builds which have been fixed for the Python 3.8 builds. I would recommend asking the Intel to provide pyrealsense2 for Python 3.8.

@wolfv @traversaro - do you know of any way to roll back to an old version of log4cxx so that @jackfrisht can run our old builds?

traversaro, 2021-03-17:
>  installed Intel Real Sense SDK pyrealsense2

How did you installed `pyrealsense2` ? If the 3.7 limitation comes from wheels updated to PyPI (see https://github.com/conda-forge/librealsense-feedstock/issues/4) an alternative option would be to provide pyrealsense2 python package via conda-forge. I opened a issue to track this in https://github.com/conda-forge/librealsense-feedstock/issues/4, even if at the moment I do not plan to work on it, but anyone that wants to want on it is welcome! 

> @wolfv @traversaro - do you know of any way to roll back to an old version of log4cxx so that @jackfrisht can run our old builds?

I am not sure, but checking the log4cxx releases (https://anaconda.org/conda-forge/log4cxx/files) I guess the old binaries were build against log4cxx 0.10, so perhaps we can try with:
~~~
conda install log4cxx=0.10
~~~
but it is possible that more problem will emerge.

jackfrisht, 2021-03-17:
@traversaro Thank you for your suggestion
I just tried to build the pyrealsense2 from source with Python 3.8 exe, then moved the generated files to conda folder, it works perfectly. 
Thank you guys support.

closed 2021-03-17 by jackfrisht