RoboStack/ros-noetic #86: Segmentation fault when importing tf2_py on MacOS

sonelu, 2021-03-17:
Not sure if this is an issue with the build on RoboStack or an upstream issue, but importing tf2_py issues a "Segmentation fault: 11" message.

 MacOS Catalina 10.15.7
Conda: 4.9.2
Python: 3.8.8

steps to reproduce:

```
(rs-test) Alex3-2:~ Alex$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:12:38) 
[Clang 11.0.1 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import tf2_py
Segmentation fault: 11
```

packages installed:

```
(rs-test) Alex3-2:~ Alex$ conda list tf
# packages in environment at /Users/Alex/anaconda3/envs/rs-test:
#
# Name                    Version                   Build  Channel
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      2.001                hab24e00_0    conda-forge
font-ttf-source-code-pro  2.030                hab24e00_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
ros-noetic-roswtf         1.15.9           py38h7e910e5_5    robostack
ros-noetic-rqt-tf-tree    0.6.2            py38h7e910e5_5    robostack
ros-noetic-tf             1.13.2           py38heb8910a_5    robostack
ros-noetic-tf-conversions 1.13.2           py38h7e910e5_5    robostack
ros-noetic-tf2            0.7.5            py38h7b88c24_5    robostack
ros-noetic-tf2-eigen      0.7.5            py38h7e910e5_5    robostack
ros-noetic-tf2-geometry-msgs 0.7.5            py38h7e910e5_5    robostack
ros-noetic-tf2-kdl        0.7.5            py38h7e910e5_5    robostack
ros-noetic-tf2-msgs       0.7.5            py38h7e910e5_5    robostack
ros-noetic-tf2-py         0.7.5            py38h7e910e5_5    robostack
ros-noetic-tf2-ros        0.7.5            py38h7e910e5_5    robostack
ros-noetic-turtle-tf      0.2.3            py38h7e910e5_5    robostack
ros-noetic-turtle-tf2     0.2.3            py38h7e910e5_5    robostack
```

traversaro, 2021-03-17:
Can you also post all the packages installed in the environment?

sonelu, 2021-03-17:
I've setup a new environment with python 3.8 and installed `ros-noetic-desktop-full` (after setting up the channels and priorities).

```
(rs-test) Alex3-2:~ Alex$ conda list

# packages in environment at /Users/Alex/anaconda3/envs/rs-test:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        3.38.0               hd0a4bf8_1    conda-forge
apr                       1.7.0                hbcf498f_4    conda-forge
assimp                    5.0.1                h3b8ecc4_5    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.0            py38h5406a74_1    conda-forge
boost                     1.74.0           py38h692b87f_3    conda-forge
boost-cpp                 1.74.0               h43a636a_2    conda-forge
bzip2                     1.0.8                hc929b4f_4    conda-forge
c-ares                    1.17.1               h0d85af4_1    conda-forge
ca-certificates           2020.12.5            h033912b_0    conda-forge
cairo                     1.16.0            h4ec454a_1007    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.12.5        py38h50d1736_1    conda-forge
cffi                      1.14.5           py38ha97d567_0    conda-forge
cmake                     3.19.6               h641592c_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
cppzmq                    4.7.1                hf09c4ce_2    conda-forge
cryptography              3.4.6            py38h1fa4640_0    conda-forge
curl                      7.75.0               h06286d4_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dbus                      1.13.6               h0c50699_1    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.16             py38h50d1736_3    conda-forge
eigen                     3.3.9                h926bf3e_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.5                h35c211d_0    conda-forge
expat                     2.2.10               h1c7c35f_0    conda-forge
ffmpeg                    4.3.1                hb787a4c_2    conda-forge
flann                     1.9.1             hed988e5_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      2.001                hab24e00_0    conda-forge
font-ttf-source-code-pro  2.030                hab24e00_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            hd23ceaa_1004    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hcc97f4c_6    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.3.0              hc02bdab_11    conda-forge
gdbm                      1.18                 h8a0c380_2    conda-forge
gdk-pixbuf                2.42.2               h976c81d_2    conda-forge
gettext                   0.19.8.1          h7937167_1005    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
glib                      2.66.7               he49afe7_1    conda-forge
glib-tools                2.66.7               he49afe7_1    conda-forge
gmock                     1.10.0               h926bf3e_7    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.6.13               h756fd2b_1    conda-forge
gpgme                     1.15.1               h046ec9c_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  2.47.0               ha2e93d9_0    conda-forge
gtest                     1.10.0               h926bf3e_7    conda-forge
gtk2                      2.24.33              h55c0dba_0    conda-forge
gtk3                      3.24.27              h0de7d9a_0    conda-forge
gts                       0.7.6                hccb3bdf_2    conda-forge
harfbuzz                  2.8.0                h159f659_0    conda-forge
hdf5                      1.10.6          nompi_hc5d9132_1114    conda-forge
hicolor-icon-theme        0.17                 h694c41f_2    conda-forge
icu                       68.1                 h74dc148_0    conda-forge
ilmbase                   2.5.5                hfab91a5_0    conda-forge
[... 388 lines cut ...]
xorg-libxext              1.3.4                h01d97ff_0    conda-forge
xorg-libxfixes            5.0.3             h01d97ff_1004    conda-forge
xorg-libxi                1.7.10               h01d97ff_0    conda-forge
xorg-libxinerama          1.1.4             h1c7c35f_1001    conda-forge
xorg-libxmu               1.1.3                h0b31af3_0    conda-forge
xorg-libxpm               3.5.13               h0b31af3_0    conda-forge
xorg-libxrandr            1.5.2                h0b31af3_1    conda-forge
xorg-libxrender           0.9.10            h01d97ff_1002    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                haf1e3a3_0    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                h1c7c35f_0    conda-forge
zlib                      1.2.11            h7795811_1010    conda-forge
zstd                      1.4.9                h582d3a0_0    conda-forge
zziplib                   0.13.69              hb0a8c7a_1    conda-forge
```

traversaro, 2021-03-17:
I suspect what could be happening is somekind of ABI incompatibility or ABI conflict, can you please run this command and report the result?
~~~
otool -L ${CONDA_PREFIX}/lib/python3.8/site-packages/tf2_py/_tf2.so
~~~

sonelu, 2021-03-17:
Please see below:

```
(robostack) Alex3-2:catkin_ws Alex$ otool -L ${CONDA_PREFIX}/lib/python3.8/site-packages/tf2_py/_tf2.so
/Users/Alex/anaconda3/envs/robostack/lib/python3.8/site-packages/tf2_py/_tf2.so:
	@rpath/_tf2.so (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libpython3.8.dylib (compatibility version 3.8.0, current version 3.8.0)
	@rpath/libtf2.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librostime.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

traversaro, 2021-03-17:
The Python bindings do not link anything strange, can you do the same with `libtf2.dylib` (that I guess is in `${CONDA_PREFIX}/lib`)?

sonelu, 2021-03-17:
```
(robostack) Alex3-2:catkin_ws Alex$ otool -L ${CONDA_PREFIX}/lib/libtf2.dylib 
/Users/Alex/anaconda3/envs/robostack/lib/libtf2.dylib:
	@rpath/libtf2.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librostime.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libconsole_bridge.1.0.dylib (compatibility version 1.0.0, current version 0.0.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

traversaro, 2021-03-17:
Ok, then probably I was wrong, nothing strange appears from that.

sonelu, 2021-03-17:
I'm trying to make a fresh install on a linux VM to check if it happens there too. Will post shortly when I'm done.

sonelu, 2021-03-17:
Seems to be working fine on Linux.
Installed on a brand new Ubuntu 20.4 with conda 4.9.2.
Installed ros-noetic-desktop-full.

```
(base) ros@ros-test:~$ conda activate rs
(rs) ros@ros-test:~$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:22:27) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tf2_py
>>> 
```

Tobias-Fischer, 2021-03-17:
See https://github.com/ros/geometry2/pull/502 for the likely source of the segfault, and https://github.com/RoboStack/ros-noetic/pull/87 for a potential patch. I'll merge this PR once it builds successfully, it would be great if once that's done you could upgrade ros-noetic-tf2-py and try again @sonelu.

Tobias-Fischer, 2021-03-17:
Hi @sonelu - please try whether it works, and feel free to reopen if it doesn't.

closed 2021-03-17 by Tobias-Fischer

sonelu, 2021-03-18:
Just as a hint, might need to apply the patch also to the ROS2 stack too. I haven't been running that yet, but I assume the same would happen there.

Tobias-Fischer, 2021-03-18:
Great - would you be happy to create the patch?

sonelu, 2021-03-18:
I could try. Two questions:

1. sorry for probably a basic question but I haven't been doing patches before: do you create the patch content manually or do you have another method (ex. branch the repo, change the file, `git diff` between the versions and dump to `.patch`, then commit only the `.patch` without the changes to the original file)?

2. in this case, because the changes are OS conditioned (`if (NOT APPLE) ... ` etc.) couldn't we save it into a `ros-foxy-tf2_py.patch` instead of `ros-foxy-tf2_py.osx.patch`? I know that it will create new builds for Linux and Windows too, but I guess it would make it easier to propagate upstream changes when necessary.

Tobias-Fischer, 2021-03-18:
Hi @sonelu 
1. Yes, the `git diff` way that you describe would work exactly like that. Alternatively, you can create a PR upstream (for tf2_py in ROS1, it would be https://github.com/ros/geometry2/pull/502) and append `.diff` at the end of the URL (i.e. https://github.com/ros/geometry2/pull/502.diff) to get the diff from the PR. Either method works fine.
2. Yes, you are correct. It could be saved without the `.osx`.

sonelu, 2021-03-19:
Actually, on second inspection the `CMakeList.txt` on `ros2/geometry2/tf2_py` was completely re-written and I don't see any `link_libraries(${PYTHON_LIBRARIES})` that would require OS conditioning. The only question is if:

```
if(APPLE)
  set_target_properties(tf2_py PROPERTIES
                        LINK_FLAGS "-undefined dynamic_lookup")
endif()
```

is still needed.

I'll try to setup a ROS2 environment and see if that segmentation fault is thrown.

wolfv, 2021-03-19:
@sonelu you might be aware that we have some initial packages for ors-foxy over at the robostack/ros-foxy repo? They are also on the robostack channel. here is the list of available packages per platform: https://robostack.github.io/foxy.html

So you can just run `mamba / conda create ros2 ros-foxy-tf2-py -c robostack -c conda-forge` and it should work. curious to hear about it.

wolfv, 2021-03-19:
PS if you have packages missing for ROS Foxy, we can try to add them. It's the same process as noetic.

sonelu, 2021-03-19:
Yes, I was aware of that and that is how I was planning to setup the environment.
I'll let you know how it's going.

PR #87 (2021-03-17): Rebuild tf2_py with patch
Fix #86