RoboStack/ros-noetic #88: Error building catkin ws

gonzalocasas, 2021-03-24:
First of all, thanks for the extraordinary effort of moving ROS to conda, it's epic!

I was able to install ros-noetic + moveit and the panda config in my Windows 10 machine without any issues. Afterwards, I tried to create a catkin ws and build a super trivial package, but got some errors I could not resolve.

The package I tried to build is https://github.com/gramaziokohler/ros_file_server

The error I get is this when I run `catkin_make` on the root of my workspace (containing only the source package mentioned above):
```
[ 90%] Building CXX object ros_file_server/CMakeFiles/file_server.dir/src/file_server.cpp.obj
file_server.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include\xlocale(319): warning C4530: C++ exception handler used, but unwind semantics are not enabled. Specify /EHsc
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(126): error C2039: 'shared_ptr': is not a member of 'boost'
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(71): note: see declaration of 'boost'
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(126): error C2143: syntax error: missing ';' before '<'
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(126): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(127): error C2065: 'TokenPtr': undeclared identifier
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(127): error C2923: 'std::vector': 'TokenPtr' is not a valid template type argument for parameter '_Ty'
C:\Users\gcasas\anaconda3\envs\ros\Library\include\ros/console.h(127): error C3203: 'allocator': unspecialized class template can't be used as a template argument for template parameter '_Alloc', expected a real type
C:\Users\gcasas\anaconda3\envs\ros\Library\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
NMAKE : fatal error U1077: 'C:\PROGRA~2\MIB055~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```

I installed compilers and all other tooling as described in your docs (e.g. `conda install compilers cmake pkg-config make ninja`), but one thing that I could not install is `catkin_tools` because it's only available for mac and linux, not windows.

Tobias-Fischer, 2021-03-24:
It seems like that https://github.com/ros/rosconsole/blob/noetic-devel/include/ros/console.h should include boost.h but doesn't ..

Tobias-Fischer, 2021-03-24:
Can you try `#include <boost/shared_ptr.hpp>` before including any ROS headers?

gonzalocasas, 2021-03-24:
Including boost's header worked! The first time around, it failed with a `throw_exception` missing linker error, but then I added this and it built:
```
#include <boost/shared_ptr.hpp>

namespace boost {
#ifdef BOOST_NO_EXCEPTIONS
void throw_exception( std::exception const & e ) { throw 11; };
#endif
}
```

Tobias-Fischer, 2021-03-24:
Great. It would be nice if you open an issue (or, even better, a pull request) upstream in rosconsole to get this fixed. Closing here as it's not a robostack specific problem.

closed 2021-03-24 by Tobias-Fischer