RoboStack/ros-noetic #89: add package: kdl-parser-py in Noetic stack

sonelu, 2021-03-24:
Could we add `ros-noetic-kdl-parser-py` in the list? I see it was initially in the melodic stack:

```
(ros1) Alex3-2:catkin_ws Alex$ conda search kdl-parser
Loading channels: done
No match found for: kdl-parser. Search: *kdl-parser*
# Name                       Version           Build  Channel             
ros-foxy-kdl-parser            2.4.1  py38h46e7d52_0  robostack           
ros-melodic-kdl-parser          1.13.1  py36h0130604_1  robostack           
ros-melodic-kdl-parser-py          1.13.1  py36h0130604_1  robostack           
ros-noetic-kdl-parser          1.14.1  py36h0130604_2  robostack           
ros-noetic-kdl-parser          1.14.1  py38h143cb38_3  robostack           
ros-noetic-kdl-parser          1.14.1  py38h7e910e5_5  robostack
```

Thanks.

wolfv, 2021-03-24:
added!

wolfv, 2021-03-24:
@sonelu it's here: https://anaconda.org/robostack/ros-noetic-kdl-parser-py 

Let us know how that one goes :)

sonelu, 2021-03-24:
Well, I have a seg fault...:

```
(ros1) Alex3-2:catkin_ws Alex$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:12:38) 
[Clang 11.0.1 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import kdl_parser_py.urdf
>>> rob = kdl_parser_py.urdf.treeFromParam('robot_description')
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_hip_r_tran']/actuator[@name='l_hip_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_hip_p_tran']/actuator[@name='l_hip_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_kne_p_tran']/actuator[@name='l_kne_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_kne_y_tran']/actuator[@name='l_kne_y_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_ank_p_tran']/actuator[@name='l_ank_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_ank_r_tran']/actuator[@name='l_ank_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_hip_r_tran']/actuator[@name='r_hip_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_hip_p_tran']/actuator[@name='r_hip_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_kne_p_tran']/actuator[@name='r_kne_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_kne_y_tran']/actuator[@name='r_kne_y_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_ank_p_tran']/actuator[@name='r_ank_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_ank_r_tran']/actuator[@name='r_ank_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_sho_p_tran']/actuator[@name='l_sho_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_sho_r_tran']/actuator[@name='l_sho_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_elb_y_tran']/actuator[@name='l_elb_y_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='l_elb_p_tran']/actuator[@name='l_elb_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_sho_p_tran']/actuator[@name='r_sho_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_sho_r_tran']/actuator[@name='r_sho_r_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_elb_y_tran']/actuator[@name='r_elb_y_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='r_elb_p_tran']/actuator[@name='r_elb_p_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='head_y_tran']/actuator[@name='head_y_motor']
Unknown tag "hardwareInterface" in /robot[@name='mh5_robot']/transmission[@name='head_p_tran']/actuator[@name='head_p_motor']
Segmentation fault: 11
(ros1) Alex3-2:catkin_ws Alex$
```

The short dump looks like this:

```
Process:               python3.8 [17451]
Path:                  /Users/USER/*/python3.8
Identifier:            python3.8
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        bash [1895]
Responsible:           Terminal [1893]
User ID:               502

Date/Time:             2021-03-24 18:49:04.873 +0000
OS Version:            Mac OS X 10.15.7 (19H524)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        D8DB1A5A-2F6F-7F4E-01CC-74BEA39189FC

Time Awake Since Boot: 37000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000646570
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [17451]

VM Regions Near 0x646570:
--> 
    __TEXT                 000000010b604000-000000010b940000 [ 3312K] r-x/r-x SM=COW  /Users/USER/*/*.8
[... 30 lines cut ...]
28  python                        	0x000000010b7823a2 call_function + 162
29  python                        	0x000000010b77ec56 _PyEval_EvalFrameDefault + 42646
30  python                        	0x000000010b7726fe _PyEval_EvalCodeWithName + 558
31  python                        	0x000000010b800585 PyRun_InteractiveOneObjectEx + 1141
32  python                        	0x000000010b7fcb56 PyRun_InteractiveLoopFlags + 246
33  python                        	0x000000010b7fca2c PyRun_AnyFileExFlags + 60
34  python                        	0x000000010b8285f1 pymain_run_stdin + 321
35  python                        	0x000000010b82784f pymain_run_python + 735
36  python                        	0x000000010b827515 Py_RunMain + 37
37  python                        	0x000000010b828d11 pymain_main + 49
38  python                        	0x000000010b605198 main + 56
39  libdyld.dylib                 	0x00007fff6a21dcc9 start + 1
```

And the `otool` for `PyKDL.so`:

```
(ros1) Alex3-2:catkin_ws Alex$ otool -L /Users/Alex/anaconda3/envs/ros1/lib/python3.8/site-packages/PyKDL.so 
/Users/Alex/anaconda3/envs/ros1/lib/python3.8/site-packages/PyKDL.so:
	@rpath/liborocos-kdl.1.4.dylib (compatibility version 1.4.0, current version 1.4.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1197.1.1)
```

I'll try also with a less complex example (I'll take one from their test) and see if that does the same.

wolfv, 2021-03-24:
Do you know how sip is involved (Qt somewhere?)?

sonelu, 2021-03-24:
I have to look, but it's definitely not from Qt. I think the KDL library is in cpp and probably they used sip to wrap it for python use.

sonelu, 2021-03-24:
I've tried running it with the [simple URDF](https://github.com/ros/kdl_parser/blob/noetic-devel/kdl_parser_py/test/test.urdf) they have in the test and it also seg faults:

```
(ros1) Alex3-2:Downloads Alex$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:12:38) 
[Clang 11.0.1 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import kdl_parser_py.urdf
>>> rob = kdl_parser_py.urdf.treeFromFile('test.urdf')
Segmentation fault: 11
(ros1) Alex3-2:Downloads Alex$ 
```

the shortdump is a little different though:

```
Process:               python3.8 [19031]
Path:                  /Users/USER/*/python3.8
Identifier:            python3.8
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        bash [1895]
Responsible:           Terminal [1893]
User ID:               502

Date/Time:             2021-03-24 19:41:05.455 +0000
OS Version:            Mac OS X 10.15.7 (19H524)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        D8DB1A5A-2F6F-7F4E-01CC-74BEA39189FC

Time Awake Since Boot: 40000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000060
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [19031]

VM Regions Near 0x60:
--> 
    __TEXT                 00000001017e3000-0000000101b1f000 [ 3312K] r-x/r-x SM=COW  /Users/USER/*/*.8

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fff6a413e52 _platform_strlen + 18
1   PyKDL.so                      	0x0000000101e4a663 convertTo_std_string(_object*, void**, int*, _object*) + 131
2   sip.so                        	0x0000000101ff1792 sip_api_convert_to_type + 99
3   sip.so                        	0x0000000101ffd2b7 parsePass2 + 2811
4   sip.so                        	0x0000000101ffab13 parseKwdArgs + 364
5   sip.so                        	0x0000000101ff6c29 sip_api_parse_kwd_args + 141
6   PyKDL.so                      	0x0000000101e3d476 init_type_Tree(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 150
7   sip.so                        	0x0000000101ff1ef6 sipSimpleWrapper_init + 173
8   python                        	0x00000001018a7c46 type_call + 246
9   python                        	0x0000000101811527 _PyObject_MakeTpCall + 167
10  python                        	0x0000000101961447 call_function + 327
11  python                        	0x000000010195dc56 _PyEval_EvalFrameDefault + 42646
12  python                        	0x00000001019516fe _PyEval_EvalCodeWithName + 558
13  python                        	0x0000000101812ef4 _PyFunction_Vectorcall + 420
14  python                        	0x00000001019613a2 call_function + 162
15  python                        	0x000000010195f021 _PyEval_EvalFrameDefault + 47713
16  python                        	0x0000000101812e42 _PyFunction_Vectorcall + 242
17  python                        	0x00000001019613a2 call_function + 162
18  python                        	0x000000010195dc56 _PyEval_EvalFrameDefault + 42646
19  python                        	0x00000001019516fe _PyEval_EvalCodeWithName + 558
20  python                        	0x00000001019df585 PyRun_InteractiveOneObjectEx + 1141
21  python                        	0x00000001019dbb56 PyRun_InteractiveLoopFlags + 246
22  python                        	0x00000001019dba2c PyRun_AnyFileExFlags + 60
23  python                        	0x0000000101a075f1 pymain_run_stdin + 321
24  python                        	0x0000000101a0684f pymain_run_python + 735
25  python                        	0x0000000101a06515 Py_RunMain + 37
26  python                        	0x0000000101a07d11 pymain_main + 49
27  python                        	0x00000001017e4198 main + 56
28  libdyld.dylib                 	0x00007fff6a21dcc9 start + 1

```

sonelu, 2021-03-24:
Actually the problem seems to be from the `PyKDL.so` that is installed by the [python_orocos_kdl](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/CMakeLists.txt)

wolfv, 2021-03-24:
Would you mind filing an issue upstream? Maybe they’ve got an idea

Tobias-Fischer, 2021-03-24:
Note a few things:
- There are quite a few patches in the orocos-kdl feedstock: https://github.com/conda-forge/orocos-kdl-feedstock
- The last version of python_orocos_kdl was released years ago, with many changes since that release

Could we
1) Try to pinpoint it down to PyKDL.so
2) Check whether it works on Linux?

sonelu, 2021-03-25:
For number 1 I can confirm that the seg fault is from PyKDL:

```
(ros1) Alex3-2:~ Alex$ python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:12:38) 
[Clang 11.0.1 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyKDL import Joint
>>> j = Joint("name")
Segmentation fault: 11
(ros1) Alex3-2:~ Alex$ 
```

The short-dump looks the same as the ones above (the error start with the handling of the string when the Joint object is created):

```
Process:               python3.8 [4147]
Path:                  /Users/USER/*/python3.8
Identifier:            python3.8
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        bash [3417]
Responsible:           Terminal [3415]
User ID:               502

Date/Time:             2021-03-25 08:56:11.745 +0000
OS Version:            Mac OS X 10.15.7 (19H524)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        D8DB1A5A-2F6F-7F4E-01CC-74BEA39189FC

Time Awake Since Boot: 2800 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000100000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [4147]

VM Regions Near 0x100000000:
--> 
    __TEXT                 000000010bdaa000-000000010c0e6000 [ 3312K] r-x/r-x SM=COW  /Users/USER/*/*.8

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fff6d6d6e52 _platform_strlen + 18
1   PyKDL.so                      	0x000000010c305663 convertTo_std_string(_object*, void**, int*, _object*) + 131
2   sip.so                        	0x000000010c4ac792 sip_api_convert_to_type + 99
3   sip.so                        	0x000000010c4b82b7 parsePass2 + 2811
4   sip.so                        	0x000000010c4b5b13 parseKwdArgs + 364
5   sip.so                        	0x000000010c4b1c29 sip_api_parse_kwd_args + 141
6   PyKDL.so                      	0x000000010c2fd3c2 init_type_Joint(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 258
7   sip.so                        	0x000000010c4acef6 sipSimpleWrapper_init + 173
8   python                        	0x000000010be6ec46 type_call + 246
9   python                        	0x000000010bdd8527 _PyObject_MakeTpCall + 167
10  python                        	0x000000010bf28447 call_function + 327
11  python                        	0x000000010bf26021 _PyEval_EvalFrameDefault + 47713
12  python                        	0x000000010bf186fe _PyEval_EvalCodeWithName + 558
13  python                        	0x000000010bfa6585 PyRun_InteractiveOneObjectEx + 1141
14  python                        	0x000000010bfa2b56 PyRun_InteractiveLoopFlags + 246
15  python                        	0x000000010bfa2a2c PyRun_AnyFileExFlags + 60
16  python                        	0x000000010bfce5f1 pymain_run_stdin + 321
17  python                        	0x000000010bfcd84f pymain_run_python + 735
18  python                        	0x000000010bfcd515 Py_RunMain + 37
19  python                        	0x000000010bfced11 pymain_main + 49
20  python                        	0x000000010bdab198 main + 56
21  libdyld.dylib                 	0x00007fff6d4e0cc9 start + 1
```

For number 2 if someone has a Linux machine at hand where they can run the same code as above it would be great. Otherwise I can try to spun a VM and try it.

Tobias-Fischer, 2021-06-07:
This will be fixed once https://github.com/conda-forge/orocos-kdl-feedstock/pull/11 is merged in. That PR also contains a test to avoid regression.

Tobias-Fischer, 2021-06-15:
@sonelu can you try whether this works now? We have a new build of the orocos-kdl, I'm not sure whether the ROS packages need rebuilding.

sonelu, 2021-06-15:
@Tobias-Fischer I've tried a new install in a new environment and I still get the same segmentation fault as [here](https://github.com/RoboStack/ros-noetic/issues/89#issuecomment-806480384). But the version I got from conda when installing was 1.4.0 and I noticed that you already released 1.5.0 in the [orocos](https://github.com/orocos/orocos_kinematics_dynamics) repo. I assume the conda builds need to be re-run to have the updated code.

Tobias-Fischer, 2021-06-16:
That works now - I created a new env with `mamba create -n kdltest ros-noetic-kdl-parser-py` and was able to run above commands just fine. Note that it doesn't work with `ros-noetic-desktop` yet because of https://github.com/RoboStack/ros-noetic/issues/138 but let's keep track of that in this issue.

closed 2021-06-16 by Tobias-Fischer