RoboStack/ros-noetic #92: Problem with cmake and compilers

Y-Grigoriy, 2021-03-25:
Hello,

I try to use command **catkin_make** to build the package, but I got error with mention of compilers:

`Base path: c:\catkin_ws
Source space: c:\catkin_ws\src
Build space: c:\catkin_ws\build
Devel space: c:\catkin_ws\devel
Install space: c:\catkin_ws\install
####
#### Running command: "cmake c:\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=c:\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=c:\catkin_ws\install -G NMake Makefiles" in "c:\catkin_ws\build"
####
CMake Error at C:/Users/GrigoriyYashin/Anaconda3/envs/robostackenv/Library/share/cmake-3.19/Modules/CMakeDetermineCCompiler.cmake:49 (message):
  Could not find compiler set in environment variable CC:

  cl.exe.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
Invoking "cmake" failed`

I use anaconda, ros-noetic installed on Windows 10.

wolfv, 2021-03-25:
Hi, here are some tips:

- using anaconda is discouraged, it's better to start from Miniconda or Miniforge :)
- have you installed Visual Studio?
- For compilation, I usually use the "x64 Native Tools Command Prompt" from the start menu
- Here might be more info: https://www.google.com/search?q=windows+terminal+x64+native+tools&oq=windows+x64+tols&aqs=chrome.1.69i57j0i8i13i30l3j0i8i10i13i30j0i8i13i30l4.7795j0j7&sourceid=chrome&ie=UTF-8

Let us know how it goes!

Y-Grigoriy, 2021-03-25:
@wolfv, yes I installed Visual Studio 2019. As terminal, I use Windows terminal.

wolfv, 2021-03-25:
can you __try__ using x64 Native Tools Command Prompt ??

Y-Grigoriy, 2021-03-25:
Now, I tried. But how can I use x64 Native Tools Command Prompt properly, if I install ros via conda and create individual environment? I tried to use **catkin_make** in directory where I want to build the package and I got:

`'catkin_make' is not recognized as an internal or external command,
operable program or batch file.`

Y-Grigoriy, 2021-03-25:
I tried to install some packages and returned to the initial error, which was before the highlighted error in this issue:

```
Base path: C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws
Source space: C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\src
Build space: C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\build
Devel space: C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\devel
Install space: C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\install
Creating symlink "C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\src\CMakeLists.txt" pointing to "C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\Library\share\catkin\cmake\toplevel.cmake"
####
#### Running command: "cmake C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\install -G NMake Makefiles" in "C:\Users\GrigoriyYashin\Anaconda3\envs\robostackenv\catkin_ws\build"
####
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_C_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_CXX_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/Users/GrigoriyYashin/Anaconda3/envs/robostackenv/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GrigoriyYashin/Anaconda3/envs/robostackenv/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

P.S. I also tried to move package' folder to the directory of environment with ros.

wolfv, 2021-03-25:
This issue has little to do with ros-noetic actually. The problem is that cmake does not find your `cl.exe` compiler. This is usually done through some magic with a file called vcvarsall.bat but I don't really know how to fix that in your case. 
There is also a utility called `vswhere` that might help you.

I always use the `x64 Native Tools prompt` and I initialized it with `conda init` and it works fine.

traversaro, 2021-03-25:
If you want that the Dev Tools script is automatically loaded when you activate the conda env in command prompt, you can install the vs2019_win-64 package.

Y-Grigoriy, 2021-03-26:
@traversaro thank you, vs2019_win-64 conda-forge package helped me, but I did not see this information early (perhaps inattention).