RoboStack/ros-noetic #93: Problem with usage MAVROS on Windows

Y-Grigoriy, 2021-03-26:
Hello,

I highly appreciate the help of Robostack community, but I need help again. I try to build specific [package](https://github.com/RuslanAgishev/px4_control) (point 2 in Simulation) for development of UAV simulation. The existed solution was developed for Ubuntu, but I'm Windows user. I successfully assembled this package in WSL, but the WSL usage of course reduces the speed of the computer. Thus, I try to perform step by step using conda on Windows. I found solutions for almost all step of this package, but I couldn't launch simulation. My installation is following:

> 1.	Install  Anaconda from website: https://www.anaconda.com/products/individual and the m2-base utilities (for convenient use of linux commands, e.g.,  mkdir): _**conda install m2-base**_
> 2.	Install Visual Studio 2019 (with following packages: Desktop development with C++, С++ Cmake tools for Windows, C++ Clang tools for Windows, Windows 10 SDK) and Git.
> 3.	Install “make” package using Windows Power Shell with administrator rules (just in case): _**choco install make**_
> 4.	Install ROS following community instructions: https://github.com/RoboStack/ros-noetic and continue work in created environment.
> 5.	(For future work) Gazebo: _**conda install -c conda-forge gazebo**_
> 6.	(For future work) Gazebo-ROS package: _**conda install -c robostack ros-noetic-gazebo-ros-pkgs**_
> 7.	Mavros: _**conda install -c robostack ros-noetic-mavros**_
> 8. Install Geographiclib for Windows from website: https://geographiclib.sourceforge.io/html/install.html Just in case, I assembled this package in individual folder using **_git clone_** and **_catkin_make_**
> 9.	PX4 Software and jMAVSim from website: https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html 
> 10. Create ROS catkin workspace (in folder of created environment) and build the package:
> **_mkdir -p ~/catkin_ws/src_**
> **_cd ~/catkin_ws/src_**
> **_git clone https://github.com/RuslanAgishev/px4_control/_**
> **_cd ~/catkin_ws_**
> **_catkin_make_**
> **_~/catkin_ws/devel/setup.bat_** (instead of **_source ~/catkin_ws/devel/setup.bash_** for Ubuntu)
> 12. Launch a simulated environment (Jmavsim) in run-console.bat (installed in step 8): **_make px4_sitl_default jmavsim_**

After that I launch **_roslaunch px4_control px4_sim.launch_** in anaconda promt, but I got error that mavconn.dll file is not found:

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
[... 112 lines cut ...]

auto-starting new master
process[master]: started with pid [10632]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to 4c02f464-8e6a-11eb-b146-002b67ca0075
process[rosout-1]: started with pid [15440]
started core service [/rosout]
process[mavros-2]: started with pid [12704]
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 12704, exit code 3221225781, cmd C:\Users\User\Anaconda3\envs\gz-postinstall\Library\lib\mavros\mavros_node.exe __name:=mavros __log:=C:\Users\User\.ros\log\4c02f464-8e6a-11eb-b146-002b67ca0075\mavros-2.log].
log file: C:\Users\User\.ros\log\4c02f464-8e6a-11eb-b146-002b67ca0075\mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
Success: Process 15440, a child of process 15084, has terminated.
[master] killing on exit
Success: Process 10892, a child of process 10632, has terminated.
Success: Process 10632, a child of process 15084, has terminated.
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Could someone help me or give ideas how to repair it?

Tobias-Fischer, 2021-03-26:
My first suggestion is to use conda to install all dependencies that you want to use (including geographiclib etc).

I cannot see the actual error - you say a DLL is not found but there is no actual error message as far as I can see.

If you find a minimal example that would be much easier, instead of roslaunch which launches many nodes but it's not clear which one fails.

traversaro, 2021-03-26:
Hi @Y-Grigoriy! 

Just as a preliminary warning, note that mixing software from many different source is always risky, because you do not know if all the different software are compatible in term of toolchains or if they instead install mutually incompatible dll libraries (so-called DLL hell) so you should try to install as much software as possible from a compatible source (for example in the conda environment the `conda-forge` and `robostack` channels). 

> Install Anaconda from website: https://www.anaconda.com/products/individual and the m2-base utilities (for convenient use of linux commands, e.g., mkdir): conda install m2-base

If you want to use `conda-forge` and `robostack`, I would suggest to stick with the packages provided by this two channels, and not mix them with packages provided by the `defaults` channel. If you really need `m2-base`, you can install it from the conda-forge channel.

> Install “make” package using Windows Power Shell with administrator rules (just in case): choco install make

What benefit does this bring you? Unless strictly necessary, I suggest to avoid mixing choco and conda packages, for the reason outline before. 

> Install Geographiclib for Windows from website: https://geographiclib.sourceforge.io/html/install.html Just in case, I assembled this package in individual folder using git clone and catkin_make

Geographiclib  is available from conda-forge, I suggest to simply install it via `conda install -c conda-forge geographiclib-cpp` .

> PX4 Software and jMAVSim from website: https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html

In general, C++ libraries built with Microsoft Visual Studio (MSVC) such as (almost) all the Windows libraries  distributed by conda-forge is not compatible with C++ libraries with CygWin. However, that may work if via that installation you just install executables that you are not trying to link with conda-provided libraries.

> After that I launch roslaunch px4_control px4_sim.launch in anaconda promt, but I got error that mavconn.dll file is not found:

How do you see that the error is related to `mavconn.dll`? There is no message related to that in the log you attached.

Tobias-Fischer, 2021-03-28:
Also note that https://github.com/RoboStack/ros-noetic/issues/59 will need attention if you try to get mavros running on Windows.

Y-Grigoriy, 2021-03-29:
@Tobias-Fischer and @traversaro, following your recommendations I installed Geographiclib using conda-forge, but the error is the same. I mentioned `mavconn.dll` due to windows message, I attached it.

[image]

Tobias-Fischer, 2021-03-29:
Hi @Y-Grigoriy, thanks for this! Unfortunately I think it will be a quite long road to get all of mavros working on Windows. Would you be happy to contribute some patches and test things?

This particular problem occurs because https://github.com/mavlink/mavros/blob/ceb53a42102eb3f6f63b561a5c0e82cb38a5c5f6/libmavconn/CMakeLists.txt#L85 should read `RUNTIME DESTINATION ${CATKIN_GLOBAL_BIN_DESTINATION})` instead (the dll is installed in the wrong path currently). 

You can either rebuild the package locally after modifying https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-libmavconn.win.patch to change this, or create a PR so that the package gets rebuild in our CI. See https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md for more information.

However, I fear that once this is fixed there will be more things that don't work out of the box. I would be happy to guide you along the way, but I won't have time to fix all the problems myself.

Y-Grigoriy, 2021-03-30:
@Tobias-Fischer thank you for finding error. I agree to try fix it following your guide. But I really not clearly understand how PR works. I checked [tutorial](https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md). But for me it looks like adding name of the patch or package in PR and then it should rebuilds itself somehow without my participation :) I assume it is wrong speculation)

wolfv, 2021-03-30:
yes, it will magically rebuild the package when you make a PR and we merge it!

Tobias-Fischer, 2021-03-30:
In your case, simply modify the https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-libmavconn.win.patch to include another patch as per above, and create a pull request. We can take care of the rest.

Y-Grigoriy, 2021-03-30:
@Tobias-Fischer Sorry for my simple questions, I reread your message twicely, but I don't have enough experience to understand what and where I should change in this file  https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-libmavconn.win.patch ? I did not find where is dll mentioned in this file. Also I lost where I can find anaconda command to check the installation folder of ros.

Y-Grigoriy, 2021-03-31:
@Tobias-Fischer, I cloned [mavros repository](https://github.com/mavlink/mavros/) and changed line "RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}" on "RUNTIME DESTINATION ${CATKIN_GLOBAL_BIN_DESTINATION}" in CMakeLists.txt But for me it is not clear how this file will influence on installed mavros package via command `conda install -c robostack ros-noetic-mavros`. Nevertheless, which indecies should I use for command `git diff` and in which terminal? Anaconda Promt and Git Bash terminals don't know command `git diff`. And in which folder I should apply command `git diff`?

Tobias-Fischer, 2022-02-02:
wontfix - closing

closed 2022-02-02 by Tobias-Fischer