RoboStack/ros-noetic #94: The road to _all_ packages on Linux

wolfv, 2021-03-28:
I have tried to build _all_ ROS packages on Linux and it's going quite smoothly for a lot of them.

Some do need some "fixing" though (a patch here and there) and this is the list of those (some are also dependent on a "lower" package being available).

- [ ] ros-noetic-apriltag
- [ ] ros-noetic-apriltag-ros
- [ ] ros-noetic-bayesian-belief-networks
- [ ] ros-noetic-behaviortree-cpp-v3
- [ ] ros-noetic-cob-gazebo-objects
- [ ] ros-noetic-cob-gazebo-plugins
- [ ] ros-noetic-cob-gazebo-ros-control
- [ ] ros-noetic-cob-gazebo-tools
- [ ] ros-noetic-cob-gazebo-worlds
- [ ] ros-noetic-codec-image-transport
- [ ] ros-noetic-downward
- [ ] ros-noetic-dynamic-graph
- [ ] ros-noetic-dynamic-graph-python
- [ ] ros-noetic-dynamic-graph-tutorial
- [ ] ros-noetic-eml
- [ ] ros-noetic-ethercat-grant
- [ ] ros-noetic-ethercat-hardware
- [ ] ros-noetic-exotica
- [ ] ros-noetic-exotica-aico-solver
- [ ] ros-noetic-exotica-cartpole-dynamics-solver
- [ ] ros-noetic-exotica-collision-scene-fcl-latest
- [ ] ros-noetic-exotica-core
- [ ] ros-noetic-exotica-core-task-maps
- [ ] ros-noetic-exotica-ddp-solver
- [ ] ros-noetic-exotica-double-integrator-dynamics-solver
- [ ] ros-noetic-exotica-dynamics-solvers
- [ ] ros-noetic-exotica-examples
- [ ] ros-noetic-exotica-ik-solver
- [ ] ros-noetic-exotica-ilqg-solver
- [ ] ros-noetic-exotica-ilqr-solver
- [ ] ros-noetic-exotica-levenberg-marquardt-solver
- [ ] ros-noetic-exotica-ompl-control-solver
- [ ] ros-noetic-exotica-ompl-solver
- [ ] ros-noetic-exotica-pendulum-dynamics-solver
- [ ] ros-noetic-exotica-pinocchio-dynamics-solver
- [ ] ros-noetic-exotica-python
- [ ] ros-noetic-exotica-quadrotor-dynamics-solver
- [ ] ros-noetic-exotica-scipy-solver
- [ ] ros-noetic-exotica-time-indexed-rrt-connect-solver
- [ ] ros-noetic-exotica-val-description
- [ ] ros-noetic-explore-lite
- [ ] ros-noetic-fetch-auto-dock-msgs
- [ ] ros-noetic-fetch-bringup
- [ ] ros-noetic-fetch-calibration
- [ ] ros-noetic-fetch-depth-layer
- [ ] ros-noetic-fetch-description
- [ ] ros-noetic-fetch-driver-msgs
- [ ] ros-noetic-fetch-drivers
- [ ] ros-noetic-fetch-ikfast-plugin
- [ ] ros-noetic-fetch-maps
- [ ] ros-noetic-fetch-moveit-config
- [ ] ros-noetic-fetch-navigation
- [ ] ros-noetic-fetch-open-auto-dock
- [ ] ros-noetic-fetch-ros
- [ ] ros-noetic-fetch-teleop
- [ ] ros-noetic-fetch-tools
- [ ] ros-noetic-fkie-master-discovery
- [ ] ros-noetic-fkie-master-sync
- [ ] ros-noetic-fkie-message-filters
- [ ] ros-noetic-fkie-multimaster
- [ ] ros-noetic-fkie-multimaster-msgs
- [ ] ros-noetic-fkie-node-manager
- [ ] ros-noetic-fkie-node-manager-daemon
- [ ] ros-noetic-fkie-potree-rviz-plugin
- [ ] ros-noetic-fmi-adapter
- [ ] ros-noetic-fmi-adapter-examples
- [ ] ros-noetic-hebi-cpp-api
- [ ] ros-noetic-ifm3d
- [ ] ros-noetic-ifm3d-core
- [ ] ros-noetic-ifopt
- [ ] ros-noetic-imagesift
- [ ] ros-noetic-image-view2
- [ ] ros-noetic-joystick-drivers
- [ ] ros-noetic-jsk-common-msgs
- [ ] ros-noetic-jsk-pcl-ros
- [ ] ros-noetic-jsk-pcl-ros-utils
- [ ] ros-noetic-jsk-perception
- [ ] ros-noetic-jsk-rqt-plugins
- [ ] ros-noetic-jsk-rviz-plugins
- [ ] ros-noetic-julius
- [ ] ros-noetic-julius-ros
- [ ] ros-noetic-knowledge-representation
- [ ] ros-noetic-kobuki-core
- [ ] ros-noetic-kobuki-dock-drive
- [ ] ros-noetic-kobuki-driver
- [ ] ros-noetic-kobuki-ftdi
- [ ] ros-noetic-kobuki-msgs
- [ ] ros-noetic-laser-ortho-projector
- [ ] ros-noetic-laser-scan-matcher
- [ ] ros-noetic-libcmt
- [ ] ros-noetic-libdlib
- [ ] ros-noetic-libntcan
- [ ] ros-noetic-libphidget22
- [ ] ros-noetic-libphidgets
- [ ] ros-noetic-libsiftfast
- [ ] ros-noetic-libuvc-camera
- [ ] ros-noetic-libuvc-ros
- [ ] ros-noetic-lusb
- [ ] ros-noetic-lvr2
- [ ] ros-noetic-mapviz
- [ ] ros-noetic-mapviz-plugins
- [ ] ros-noetic-marvelmind-nav
- [ ] ros-noetic-mesh-client
- [ ] ros-noetic-mqtt-bridge
- [ ] ros-noetic-mrpt2
- [ ] ros-noetic-multires-image
- [ ] ros-noetic-multirobot-map-merge
- [ ] ros-noetic-navigation-experimental
- [ ] ros-noetic-octovis
- [ ] ros-noetic-opencv-apps
- [ ] ros-noetic-openni2-camera
- [ ] ros-noetic-openni2-launch
- [ ] ros-noetic-opt-camera
- [ ] ros-noetic-osqp-vendor
- [ ] ros-noetic-phidgets-accelerometer
- [ ] ros-noetic-phidgets-analog-inputs
- [ ] ros-noetic-phidgets-api
- [ ] ros-noetic-phidgets-digital-inputs
- [ ] ros-noetic-phidgets-digital-outputs
- [ ] ros-noetic-phidgets-drivers
- [ ] ros-noetic-phidgets-gyroscope
- [ ] ros-noetic-phidgets-high-speed-encoder
- [ ] ros-noetic-phidgets-ik
- [ ] ros-noetic-phidgets-magnetometer
- [ ] ros-noetic-phidgets-motors
- [ ] ros-noetic-phidgets-msgs
- [ ] ros-noetic-phidgets-spatial
- [ ] ros-noetic-phidgets-temperature
- [ ] ros-noetic-points-preprocessor
- [ ] ros-noetic-pose-base-controller
- [ ] ros-noetic-posedetection-msgs
- [ ] ros-noetic-pr2-ethercat-drivers
- [ ] ros-noetic-prosilica-camera
- [ ] ros-noetic-prosilica-gige-sdk
- [ ] ros-noetic-ps3joy
- [ ] ros-noetic-rc-dynamics-api
- [ ] ros-noetic-robot-pose-ekf
- [ ] ros-noetic-robot-self-filter
- [ ] ros-noetic-ros-ign
- [ ] ros-noetic-ros-ign-bridge
- [ ] ros-noetic-ros-ign-gazebo
- [ ] ros-noetic-ros-ign-gazebo-demos
- [ ] ros-noetic-ros-ign-image
- [ ] ros-noetic-rospilot
- [ ] ros-noetic-rtabmap
- [ ] ros-noetic-rtabmap-ros
- [ ] ros-noetic-rviz-satellite
- [ ] ros-noetic-sainsmart-relay-usb
- [ ] ros-noetic-scan-to-cloud-converter
- [ ] ros-noetic-scan-tools
- [ ] ros-noetic-scenario-test-tools
- [ ] ros-noetic-sesame-ros
- [ ] ros-noetic-sick-tim
- [ ] ros-noetic-sob-layer
- [ ] ros-noetic-soem
- [ ] ros-noetic-spacenav-node
- [ ] ros-noetic-swri-geometry-util
- [ ] ros-noetic-swri-image-util
- [ ] ros-noetic-swri-route-util
- [ ] ros-noetic-swri-transform-util
- [ ] ros-noetic-swri-yaml-util
- [ ] ros-noetic-tile-map
- [ ] ros-noetic-trac-ik
- [ ] ros-noetic-trac-ik-examples
- [ ] ros-noetic-trac-ik-kinematics-plugin
- [ ] ros-noetic-trac-ik-lib
- [ ] ros-noetic-trac-ik-python
- [ ] ros-noetic-ur-client-library
- [ ] ros-noetic-visp
- [ ] ros-noetic-warehouse-ros-mongo
- [ ] ros-noetic-wiimote
- [ ] ros-noetic-willow-maps
- [ ] ros-noetic-ypspur
- [ ] ros-noetic-ypspur-ros
- [ ] ros-noetic-zbar-ros

wolfv, 2021-03-28:
These ones have all worked out of the box:

- [x] ros-noetic-ackermann-msgs-1.0.2-py38he9ab703_6.tar.bz2
- [x] ros-noetic-actionlib-tools-1.13.2-py38he9ab703_6.tar.bz2
- [x] ros-noetic-agni-tf-tools-0.1.5-py38h7d895da_6.tar.bz2
- [x] ros-noetic-app-manager-1.2.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-arbotix-firmware-0.11.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-arbotix-msgs-0.11.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-assimp-devel-2.1.21-py38h5860396_6.tar.bz2
- [x] ros-noetic-assisted-teleop-0.3.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-async-comm-0.2.1-py38h266acf0_6.tar.bz2
- [x] ros-noetic-async-web-server-cpp-1.0.3-py38h3e55461_6.tar.bz2
- [x] ros-noetic-audibot-description-0.2.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-audio-common-msgs-0.3.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-automotive-navigation-msgs-3.0.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-automotive-platform-msgs-3.0.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-autoware-can-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-autoware-config-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-autoware-lanelet2-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-autoware-map-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-autoware-system-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-auv-msgs-0.1.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-avt-vimba-camera-0.0.12-py38he9ab703_6.tar.bz2
- [x] ros-noetic-backward-ros-0.1.7-py38he9ab703_6.tar.bz2
- [x] ros-noetic-bagger-0.1.3-py38he9ab703_6.tar.bz2
- [x] ros-noetic-baldor-0.1.3-py38he9ab703_6.tar.bz2
- [x] ros-noetic-boost-sml-0.1.1-py38h266acf0_6.tar.bz2
- [x] ros-noetic-bota-signal-handler-0.5.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-bota-worker-0.5.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-can-msgs-0.8.5-py38he9ab703_6.tar.bz2
- [x] ros-noetic-capabilities-0.3.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-carla-msgs-1.3.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cartesian-msgs-0.0.3-py38he9ab703_6.tar.bz2
- [x] ros-noetic-catch-ros-0.4.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-catkin-virtualenv-0.6.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-actions-0.7.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-android-msgs-0.1.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-android-resource-server-0.1.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-android-settings-0.1.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-base-controller-utils-0.8.12-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-base-velocity-smoother-0.8.12-py38h266acf0_6.tar.bz2
- [x] ros-noetic-cob-cam3d-throttle-0.6.17-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-control-mode-adapter-0.8.12-py38h266acf0_6.tar.bz2
- [x] ros-noetic-cob-control-msgs-0.8.12-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-default-env-config-0.6.12-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-description-0.7.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-docker-control-0.6.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-cob-footprint-observer-0.8.12-py38h266acf0_6.tar.bz2
- [x] ros-noetic-collada-parser-1.12.13-py38he9ab703_6.tar.bz2
- [x] ros-noetic-color-util-0.3.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-computer-status-msgs-2.0.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-control-box-rst-0.0.7-py38he9ab703_6.tar.bz2
- [x] ros-noetic-controller-manager-tests-0.19.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-convex-decomposition-0.1.12-py38he9ab703_6.tar.bz2
- [x] ros-noetic-costmap-cspace-msgs-0.8.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-criutils-0.1.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-csm-1.0.2-py38h0a51b22_6.tar.bz2
- [x] ros-noetic-cv-camera-0.5.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dataspeed-can-msg-filters-1.0.16-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dataspeed-can-tools-1.0.16-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dataspeed-pds-msgs-1.0.6-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dataspeed-ulc-msgs-0.0.5-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-fca-description-1.0.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-fca-msgs-1.0.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-mkz-description-1.2.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-mkz-msgs-1.2.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-polaris-description-0.0.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-dbw-polaris-msgs-0.0.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-ddynamic-reconfigure-0.3.2-py38he9ab703_6.tar.bz2
- [x] ros-noetic-ddynamic-reconfigure-python-0.0.1-py38he9ab703_6.tar.bz2
[... 265 lines cut ...]
- [x] ros-noetic-toposens-msgs-2.2.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-trajectory-tracker-msgs-0.8.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-turtlebot3-gazebo-1.3.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-twist-recovery-0.3.4-py38he9ab703_6.tar.bz2
- [x] ros-noetic-ublox-serialization-1.4.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-udp-com-1.1.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-ueye-cam-1.0.19-py38he9ab703_6.tar.bz2
- [x] ros-noetic-unique-id-1.0.6-py38he9ab703_6.tar.bz2
- [x] ros-noetic-urdf-geometry-parser-0.1.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-urg-stamped-0.0.9-py38he9ab703_6.tar.bz2
- [x] ros-noetic-usb-cam-0.3.6-py38h2d2d522_6.tar.bz2
- [x] ros-noetic-usb-cam-hardware-interface-0.2.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-variant-msgs-0.1.6-py38he9ab703_6.tar.bz2
- [x] ros-noetic-vector-map-msgs-1.14.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-video-stream-opencv-1.1.6-py38he9ab703_6.tar.bz2
- [x] ros-noetic-view-controller-msgs-0.1.3-py38he9ab703_6.tar.bz2
- [x] ros-noetic-virtual-force-publisher-2.2.11-py38he9ab703_6.tar.bz2
- [x] ros-noetic-vision-msgs-0.0.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-vl53l1x-1.0.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-vrpn-7.34.0-py38he9ab703_6.tar.bz2
- [x] ros-noetic-wge100-camera-firmware-1.8.5-py38he9ab703_6.tar.bz2
- [x] ros-noetic-wireless-msgs-0.1.1-py38he9ab703_6.tar.bz2
- [x] ros-noetic-xpp-msgs-1.0.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-xpp-states-1.0.10-py38he9ab703_6.tar.bz2
- [x] ros-noetic-xv-11-laser-driver-0.3.0-py38h266acf0_6.tar.bz2

Tobias-Fischer, 2021-03-29:
I think all the ros-noetic-lib* packages are vendor packages and need special treatment.

Tobias-Fischer, 2021-04-12:
So I think we're down to a few distinct groups of packages:
- vendor packages
- jsk related packages which are missing julius in conda-forge
- kobuki packages that need ftdi
- packages that fail because of PCL build issues (there used to be a time when ROS vendored PCL - is this still the case? that might explain the issues)
- a few packages that miss dependencies in conda-forge: {'libqglviewer2-qt5', 'libqglviewer-dev-qt5', 'openni', 'bluez', 'dnsmasq', 'python3-serial', 'hostapd', 'libgphoto-dev', 'libftdi-dev', 'libvlc', 'libv4l-dev', 'libois-dev', 'libdmtx-dev', 'libcoin80-dev', 'libmongoclient-dev', 'cwiid'}
- the odd ones out with some build issues, e.g. sick-tim that does not link to libusb or swri-image-util that does not link to geos properly

I left remarks for most of the packages in the above list.

Tobias-Fischer, 2022-04-03:
I think we've given up on this because of too high maintainer load

closed 2022-04-03 by Tobias-Fischer