RoboStack/ros-noetic #97: ros-noetic-geometric-shapes on Mac OS is missing libqhull_r

wxmerkt, 2021-04-07:
Trying to debug building with your great CI scripts and saw the following runtime failure (I believe the CI reports 'green' on build success not test success since this build was green):
```
dyld: Library not loaded: lib/libqhull_r.8.0.dylib
  Referenced from: /Users/runner/micromamba/envs/test/lib/libgeometric_shapes.0.7.2.dylib
  Reason: image not found
```
Do I need to manually specify libqhull to install? It seems that this should be a dependency of geometric_shapes in this case.

Thanks for any insights :-)

Tobias-Fischer, 2021-04-07:
Hi @wxmerkt - many thanks for your interest in RoboStack and reporting some issues!

This one is an interesting one; I'm not quite sure what's going on there. qhull is a dependency of ros-noetic-geometric-shapes, so that's not the problem (you can also see in your CI that qhull is being installed). It might be that qhull needs to be added to the `DEPENDS` in https://github.com/ros-planning/geometric_shapes/blob/noetic-devel/CMakeLists.txt#L66?

Indeed we haven't enabled tests yet when building .. we should do that soon to catch these cases.

traversaro, 2021-04-07:
My guess is that qhull feedstock is missing the run_exports directive: https://github.com/conda-forge/qhull-feedstock/blob/master/recipe/meta.yaml#L14 .

Tobias-Fischer, 2021-04-08:
I added a PR: https://github.com/conda-forge/qhull-feedstock/pull/14

However, I am not convinced. `qhull` is a run dependency of ros-noetic-geometric-shapes, and `qhull` is being installed. Also there shouldn't be a version conflict; we last built ros-noetic-geometric-shapes on 7 March, and qhull 2020.2 (the latest version) was released ~3 months ago.

traversaro, 2021-04-08:
Indeed, the qhull_r library files seems to be there and be fine, by inspecting https://github.com/ipab-slmc/exotica/runs/2288535427 my guess is that is something is wrong at the RPATH level (classical case of blaming something as it is the part that you least understand : ) ) or in the activation of the environment (given that there we are using micromamba that could have still some different behaviour w.r.t. to conda or mamba). Can we add a call to `env` in https://github.com/ipab-slmc/exotica/pull/740/files#diff-a136dab2e5a2addf108c52bac524fd8aa0f969cbc1ad0d95efc043a8eb69d349L43 to check the environment variables at that point?

wxmerkt, 2021-04-08:
Great pointer, thanks @traversaro for your insights. I just added `env` so in a few minutes we should be able to see the result

traversaro, 2021-04-08:
The environment seems fine, so it is indeed some other issue.

Tobias-Fischer, 2021-04-12:
I can confirm that this happens on a physical Mac, too. It currently prevents MoveIt from running.

Tobias-Fischer, 2021-04-12:
Indeed, it might be an RPATH issue:
```
(jupyterlab-ros) tobias@Tobiass-MacBook-Pro launch % otool -L /Users/tobias/opt/anaconda3/envs/jupyterlab-ros/lib/libgeometric_shapes.0.7.2.dylib
/Users/tobias/opt/anaconda3/envs/jupyterlab-ros/lib/libgeometric_shapes.0.7.2.dylib:
	@rpath/libgeometric_shapes.0.7.2.dylib (compatibility version 0.0.0, current version 0.7.2)
	@rpath/libassimp.5.dylib (compatibility version 5.0.0, current version 5.0.0)
	lib/libqhull_r.8.0.dylib (compatibility version 8.0.0, current version 8.0.2)
	@rpath/libresource_retriever.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libconsole_bridge.1.0.dylib (compatibility version 1.0.0, current version 0.0.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

Tobias-Fischer, 2021-04-13:
Attempt to fix: https://github.com/conda-forge/qhull-feedstock/pull/15

Tobias-Fischer, 2021-04-16:
Hi @wxmerkt - this should be fixed now. Please let us know if not.

closed 2021-04-16 by Tobias-Fischer