RoboStack/ros-noetic #98: ros-noetic-catkin and install-layout

wxmerkt, 2021-04-07:
I am locally using RoboStack to debug a catkin package which contains a catkin-type `setup.py` and uses `catkin_python_setup()`. Alas, it fails to install due to the `--install-layout=deb` option used in catkin (upstream issue describing the same problem: https://github.com/ros/catkin/issues/863). 

Have you tried compiling/installing catkin packages which include a Python package? 

The `SETUPTOOLS_DEB_LAYOUT` option could potentially be a workaround unless it affects something else (dist-packages vs site-packages for instance). I saw the [patch file](https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-catkin.patch) for ros-noetic-catkin but no change that would potentially address this issue?

Tobias-Fischer, 2021-04-07:
Hi @wxmerkt - indeed make use of the `SETUPTOOLS_DEB_LAYOUT` option - it is in the build template which is in a different repository (we should clean this up / make more clear how the repos interact):
https://github.com/RoboStack/vinca/blob/1f147e39aa7903bf3953ba22cef139980b2609de/vinca/templates/build_catkin.sh.in#L52

I created a fix in #98 which should be cleaner for our purposes and avoid problems like the ones you just encountered.

wxmerkt, 2021-04-07:
Many thanks for the insights and the swift resolution :-) I'll amend locally using the option.

PR #99 (2021-04-14): Fix #98 (do not ever use deb install-layout)
Fixes #98 

Thanks @wxmerkt for reporting this issue.
commit 25f0c624 (2021-04-14): Fix #98 (do not ever use deb install-layout) (#99)

* Fix #98
closed 2021-04-14 by Tobias-Fischer