RoboStack/ros-noetic #102: RViz on Ubuntu on High-DPI screens

wxmerkt, 2021-04-09:
I'm using a high-DPI screen with scaling (4K, 200% scale). Regular RViz works just fine while RoboStack-RViz does not use the full viewport.

This is how RViz starts:
[image]

After some window resizing I get this:
[image]

Sometimes it also renders garbage/background windows in the black space.

Have you observed something similar? Could this be related to a different OGRE version? I use rviz-1.14.5/the latest.

lyh458, 2021-04-10:
I have the same problem as you:
- everything is ok if I drag the edge of RVIZ windows to resize
[image]
- however, if I press the **maximum icon**, problem like you.
[image]
[image]

SteveQUT, 2021-04-12:
Can confirm that I have this issue across multiple machines and on windows. Resizing does appear to fix most case.

wolfv, 2021-04-12:
Thanks all for this bug report! Indeed, it's something I noticed on Fedora, too (and usually fixed by resizing the window).

We have similar issues on OS X where the view is skewed (but the skew factor seems to change based on how wide the RViz Window is).

I wonder if there is a place in RViz where the window viewport width is calculated? cc @rhaschke if you have a pointer where I should look I'd be happy to debug this a bit locally.

SteveQUT, 2021-04-12:
I did a little bit of digging and I am guessing the file is render_widget.cpp in the ogre_helpers folder might be relevant. Seems to have been an issue on High DPI screens here: https://github.com/ros-visualization/rviz/issues/1306#issue-373672105

Tobias-Fischer, 2021-04-12:
How about just calling RenderWidget::resizeEvent() at the end of the RenderWidget constructor? Unfortunately I don't have a high dpi screen to test it.

rhaschke, 2021-04-12:
This issue seems to be a duplicate of https://github.com/ros-visualization/rviz/issues/1262 which was fixed via https://github.com/ros-visualization/rviz/issues/1263. The corresponding commit should be part of the RoboStack-released version as well. @wolfv, have a look at [`render_widget.cpp`](https://github.com/rlober/rviz/blob/bdc4a1848ea5e0cd7bb518977264aaf9cce3c3dc/src/rviz/ogre_helpers/render_widget.cpp#L120-L133)

ameysutavani, 2021-04-25:
I can confirm the same issue on Ubuntu 18.04

rhaschke, 2021-04-25:
@ameysutavani, do you observe this issue with the standard ROS Debian packages as well or with the RoboStack release only?
In the former case, please file a bug report at https://github.com/ros-visualization/rviz.

lyh458, 2021-04-26:
> @ameysutavani, do you observe this issue with the standard ROS Debian packages as well or with the RoboStack release only?
> In the former case, please file a bug report at https://github.com/ros-visualization/rviz.

@rhaschke my system is ubuntu 18.04, and everything of RVIZ (version 1.13.16) with the standard ROS Debian packages (melodic) is ok, but has the same issue after installing this devel with conda.

ameysutavani, 2021-04-26:
> @ameysutavani, do you observe this issue with the standard ROS Debian packages as well or with the RoboStack release only?
> In the former case, please file a bug report at https://github.com/ros-visualization/rviz.

@rhaschke The above issue does not happen with deb install rviz. I can confirm this for both ros melodic and ros noetic

Tobias-Fischer, 2021-06-07:
Hiya, I just had a look at this - it seems like the ogre selector was wrong in https://github.com/RoboStack/ros-noetic/blob/057e616d4d03c4dc23567dd1b82d2add4b7de2d0/patch/ros-noetic-rviz.patch#L101
We use Ogre 1.10 which needs the call to `render_window_->windowMovedOrResized();` which didn't happen without the patch that I just uploaded (@rhaschke).

@ameysutavani @lyh458 @SteveQUT @wxmerkt - let me know if this did the trick for you (the package rebuild will take an hour or so, after that simply `mamba upgrade ros-noetic-rviz`)

SteveQUT, 2021-06-07:
Can confirm this is now fixed on Linux64.

SteveQUT, 2021-11-10:
This appears to have been reintroduced with latest update. 
ros-noetic-rviz           1.14.4           py38h7d895da_3    robostack
[image]

Tobias-Fischer, 2021-11-10:
Seems like the package was accidentally downgraded rather than upgraded.

closed 2021-11-10 by Tobias-Fischer

simonbogh, 2023-02-10:
I experience the view still being skewed using Robostack on macOS Ventura 13.2, Noetic, RViz 1.14.19 (Compiled against Qt version 5.15.6, Compiled against OGRE version 1.10.12 (Xalafu))
[image]
[image]

commit 057e616d (2021-06-07): Fix rviz on high-dpi screens (fix #102)