RoboStack/ros-noetic #103: Boost exact version

wxmerkt, 2021-04-12:
I just ran into an issue where a target dependent on `roscpp` failed to link since the boost version that was automatically installed following the instructions was 1.75 but the binaries are built against 1.74. For most binaries that is fine, but not so for `roscpp` which specifies the exact version. Would it be possible to either resolve dependencies fixed to the required version or not to hardcode the minor version in the exported dependencies/link targets? Is there a better solution (since rebuilding on upstream conda package updates might be quite involved/frequent?)?

Many thanks for any insights :-).

Tobias-Fischer, 2021-04-12:
What do you mean by "automatically installed"? How did you create the environment?

wxmerkt, 2021-04-12:
Per the README:
```
conda create -n robostackenv python=3.8
conda activate robostackenv
# this adds the conda-forge channel to your persistent configuration in ~/.condarc
conda config --add channels conda-forge
# and the robostack channel
conda config --add channels robostack
# it's very much advised to use strict channel priority
conda config --set channel_priority strict
```

and then `conda install ros-noetic-catkin ros-noetic-std-msgs ros-noetic-message-generation`. The boost installed was 1.75 and thus a C++ project I tried to build ran into the issue that the boost install did not match the 1.74 it was looking from roscpp:
```
ros-noetic-roscpp         1.15.9           py38h14b2acc_3    robostack
ros-noetic-roscpp-serialization 0.7.2            py38he9ab703_6    robostack
ros-noetic-roscpp-traits  0.7.2            py38he9ab703_6    robostack
```

This is the output after running `conda install ros-noetic-sensor-msgs` which shows that it resolved to boost-1.75:

```
conda install ros-noetic-sensor-msgs
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/wxm/miniconda3/envs/testboost

  added / updated specs:
    - ros-noetic-sensor-msgs

The following NEW packages will be INSTALLED:

  **boost-cpp          conda-forge/linux-64::boost-cpp-1.75.0-hc6e9bd1_0**
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h7f98852_4
  c-ares             conda-forge/linux-64::c-ares-1.17.1-h7f98852_1
  catkin_pkg         conda-forge/noarch::catkin_pkg-0.4.23-pyh9f0ad1d_0
  cmake              conda-forge/linux-64::cmake-3.20.1-h541d2ed_0
  console_bridge     conda-forge/linux-64::console_bridge-1.0.1-h4bd325d_0
  docutils           conda-forge/linux-64::docutils-0.17-py38h578d9bd_0
  empy               conda-forge/noarch::empy-3.3.4-pyh9f0ad1d_1
  expat              conda-forge/linux-64::expat-2.3.0-h9c3ff4c_0
  gmock              conda-forge/linux-64::gmock-1.10.0-h4bd325d_7
  gtest              conda-forge/linux-64::gtest-1.10.0-h4bd325d_7
  icu                conda-forge/linux-64::icu-68.1-h58526e2_0
  krb5               conda-forge/linux-64::krb5-1.17.2-h926e7f8_0
  libcurl            conda-forge/linux-64::libcurl-7.76.0-hc4aaa36_0
  libedit            conda-forge/linux-64::libedit-3.1.20191231-he28a2e2_2
  libev              conda-forge/linux-64::libev-4.33-h516909a_1
  libnghttp2         conda-forge/linux-64::libnghttp2-1.43.0-h812cca2_0
  libssh2            conda-forge/linux-64::libssh2-1.9.0-ha56f1ee_6
  libuv              conda-forge/linux-64::libuv-1.41.0-h7f98852_0
  lz4-c              conda-forge/linux-64::lz4-c-1.9.3-h9c3ff4c_0
  nose               conda-forge/noarch::nose-1.3.7-py_1006
  pyparsing          conda-forge/noarch::pyparsing-2.4.7-pyh9f0ad1d_0
  python-dateutil    conda-forge/noarch::python-dateutil-2.8.1-py_0
  pyyaml             conda-forge/linux-64::pyyaml-5.4.1-py38h497a2fe_0
  rhash              conda-forge/linux-64::rhash-1.4.1-h7f98852_0
  ros-distro-mutex   robostack/linux-64::ros-distro-mutex-0.1.0-noetic
  ros-noetic-catkin  robostack/linux-64::ros-noetic-catkin-0.8.9-py38he9ab703_6
  ros-noetic-cpp-co~ robostack/linux-64::ros-noetic-cpp-common-0.7.2-py38haa43186_3
  ros-noetic-genmsg  robostack/linux-64::ros-noetic-genmsg-0.5.16-py38he9ab703_6
  ros-noetic-genpy   robostack/linux-64::ros-noetic-genpy-0.6.14-py38he9ab703_6
  ros-noetic-geomet~ robostack/linux-64::ros-noetic-geometry-msgs-1.13.1-py38he9ab703_6
  ros-noetic-messag~ robostack/linux-64::ros-noetic-message-runtime-0.4.13-py38he9ab703_6
  ros-noetic-roscpp~ robostack/linux-64::ros-noetic-roscpp-serialization-0.7.2-py38he9ab703_6
  ros-noetic-roscpp~ robostack/linux-64::ros-noetic-roscpp-traits-0.7.2-py38he9ab703_6
  ros-noetic-rostime robostack/linux-64::ros-noetic-rostime-0.7.2-py38h14b2acc_3
  ros-noetic-sensor~ robostack/linux-64::ros-noetic-sensor-msgs-1.13.1-py38he9ab703_6
  ros-noetic-std-ms~ robostack/linux-64::ros-noetic-std-msgs-0.5.13-py38he9ab703_6
  six                conda-forge/noarch::six-1.15.0-pyh9f0ad1d_0
  yaml               conda-forge/linux-64::yaml-0.2.5-h516909a_0
  zstd               conda-forge/linux-64::zstd-1.4.9-ha95c52a_0

Proceed ([y]/n)? 
```

wolfv, 2021-04-12:
Indeed, I don't think boost versions should ever be "mixed".
We have a ros-distro-mutex package that we can use to attach other versions (boost etc.), so that we can "globally pin" them.

traversaro, 2021-04-12:
I think we need to add run dependency manually for boost-cpp, as it does not have a run_exports directive ( see https://github.com/conda-forge/boost-cpp-feedstock/pull/82 ).

Tobias-Fischer, 2021-04-12:
Hmm, it seems like it's already done:
https://github.com/RoboStack/ros-noetic/blob/a29087b73f3211c60ca0d55084dcc2b4442939d4/additional_recipes/ros-distro-mutex/recipe.yaml#L32-L33

Should it just be `run` instead of `run_constrained`? Checking the index.json of the generated package, it contains `"depends": [],` which is not what we want. It could also be a bug in boa?

Tobias-Fischer, 2021-04-15:
See https://github.com/RoboStack/ros-noetic/pull/107 for the fix

Tobias-Fischer, 2021-04-16:
Fixed by #111

closed 2021-04-16 by Tobias-Fischer