RoboStack/ros-noetic #108: libfranka for Linux arm64

mlaves, 2021-04-15:
I just came across this awesome project. However for my work, `ros-noetic-libfranka` for linux on arm64 is still missing. I tried to create a custom package using `vinca` but was unable to do so. Could you create this package or hint me on how to create it myself?

mlaves, 2021-04-15:
Done! Thanks for the very quick response

wolfv, 2021-04-15:
here is the build: https://dev.azure.com/roboforge/ros_pipelines/_build/results?buildId=1841&view=logs&j=1af1dc03-1af3-5fb1-94be-7682c40b8e14

(kicked off from the `buildbranch_linux_aarch64` in this repo :)

wolfv, 2021-04-15:
Package is uploaded to the robostack channel https://anaconda.org/RoboStack/ros-noetic-libfranka/files

Let us know how it goes :)

closed 2021-04-15 by wolfv