RoboStack/ros-noetic #110: Support for arm 32 bit

arturmiller, 2021-04-15:
Hello,

this is a cool project. I like that it's possible to install the packages as user (no sudo) and that there is only one source for ROS and Python packages.
I have a question. Are you planning to support 32 bit arm packages in the future as well?

Best Regards

wolfv, 2021-04-15:
Hi, thanks for the kind words! 

There is the [berryconda](https://github.com/jjhelmus/berryconda) project that ships arm32 packages. However, it doesn't cover by far the packages that we get from `conda-forge`, and conda-forge doesn't support ARM32 at the moment. 
I am not sure that interest in conda-forge is high enough to add ARM32 as an additional architecture currently, and without conda-forge we'd have to maintain quite a lot of packages ourselves...

So I think unless you can convince conda-forge to add support for ARM32 we can't really help here.

PS: is it really not possible to use the ARM64 packages? Are there any ARM32 processors still in widespread use?

arturmiller, 2021-04-16:
Hello wolfv, thank you for the clarification. Yes, unfortunately ARM32 is still used :D

closed 2021-04-16 by arturmiller

wolfv, 2021-04-16:
afaik on Raspberry Pi's you can use ARM64 right? Just need to install a different distro or something like that?

husnoo, 2022-12-13:
Raspberry pi 2B v1.2 onwards can use 64 bits, but the ones before are 32bits only.