RoboStack/ros-noetic #117: robot_localization/navsat_transform_node missing a dependency on geographiclib-cpp

ameysutavani, 2021-04-25:
**Error description:**

When launching robot_localization/navsat_transform_node, it cannot find libGeographic.so.19:
[image]

Manually installing geographiclib-cpp solves the issue: `mamba install geographiclib-cpp`

Looks like the ros-noetic-robot-localization package is missing geographiclib-cpp as one of its dependencies.

Tobias-Fischer, 2021-04-25:
That's because upstream specifies geographiclib-cpp as a build_dependency only: https://github.com/cra-ros-pkg/robot_localization/blob/0d910c2da5150b6844c9a9aad957c53b2a002201/package.xml#L36

I think it's worth opening an issue there - we could patch it here but it's better fixed upstream.

Tobias-Fischer, 2021-04-26:
Actually never mind opening an issue upstream, this will be solved after merging https://github.com/conda-forge/geographiclib-cpp-feedstock/pull/2 and rebuilding robot_localization.

ameysutavani, 2021-04-26:
Oh, I was just about to create the issue upstream. Should we still try to fix this upstream?

Tobias-Fischer, 2021-04-26:
I'm not sure whether it should be a build_depend only, or a depend. @wolfv @traversaro - any thoughts?

ameysutavani, 2021-04-26:
The Ubuntu deb installed robot localization navsat_transform_node does not error-out like this somehow; my hunch is  geographiclib-cpp is installed by some other component in ros-noetic there

traversaro, 2021-04-26:
> 
> 
> I'm not sure whether it should be a build_depend only, or a depend. @wolfv @traversaro - any thoughts?

Checking the docs in http://wiki.ros.org/catkin/package.xml#Dependencies and considering that the package is a shared library (see https://packages.ubuntu.com/focal/amd64/libgeographic19/filelist), I guess indeed the problem is there and   `<build_depend>geographiclib</build_depend>`  should actually be `<depend>geographiclib</depend>` .

Tobias-Fischer, 2021-04-27:
I'll close this issue - it'll be resolved once we do the next rebuild.

closed 2021-04-27 by Tobias-Fischer