RoboStack/ros-noetic #122: Error in ros-noetic activation script when sourcing on start

raghavauppuluri13, 2021-05-10:
Platform: MacOS (NOT apple silicon)
Installer: conda-forge/mamba

### Error message
`/Users/raghava13/miniforge3/envs/robostackenv/etc/conda/activate.d/ros-noetic-catkin_activate.sh:10: = not found`

### Activation Script
```
# Generated by vinca http://github.com/seanyen/vinca.
# DO NOT EDIT!
# if [ -z "${CONDA_PREFIX}" ]; then
#     exit 0;
# fi
# Not sure if this is necessary on UNIX?
# export QT_PLUGIN_PATH=$CONDA_PREFIX\plugins

if [ "$CONDA_BUILD" == "1" -a "$target_platform" != "$build_platform" ]; then
     # ignore sourcing
     echo "Not activating ROS when cross-compiling";
else
     source $CONDA_PREFIX/setup.sh
fi

case "$OSTYPE" in
darwin*)  export ROS_OS_OVERRIDE="robostack:osx";; 
linux*)   export ROS_OS_OVERRIDE="robostack:linux";;
 esac
  
# Looks unnecessary for UNIX
# unset PYTHONHOME=
```

Tobias-Fischer, 2021-05-10:
Hi @raghavauppuluri13 - thanks for reporting this! This should be resolved after catkin will rebuild, please wait for an hour or so and then update `ros-noetic-catkin`. If you don't want to wait, please edit `/Users/raghava13/miniforge3/envs/robostackenv/etc/conda/activate.d/ros-noetic-catkin_activate.sh` and change 
```
if [ "$CONDA_BUILD" == "1" -a "$target_platform" != "$build_platform" ]; then
```
into
```
if [ "$CONDA_BUILD" = "1" -a "$target_platform" != "$build_platform" ]; then
```

raghavauppuluri13, 2021-05-10:
Awesome. Thanks for the quick reply.

Tobias-Fischer, 2021-05-24:
I just triggered another rebuild for Linux (previously we only rebuild osx) which should solve this issue (as reported on Gitter).

commit 203b58a6 (2021-05-24): Rebuild catkin on Linux to fix #122
closed 2021-05-24 by Tobias-Fischer