RoboStack/ros-noetic #127: it is fool to must link opencv 4.2 on ros

graphic-zhang, 2021-05-17:
it will cause confict when developer use different opencv version or different modules

wolfv, 2021-05-17:
Hi @graphic-zhang I think we're using opencv 4.5 currently. What version are you interested in? We need to decide on some version at some point, otherwise we need to constantly update the code etc.

Tobias-Fischer, 2021-05-17:
Indeed, we're using opencv 4.5: https://github.com/RoboStack/ros-noetic/blob/cca5378e0a0c09974119e39964cf14a7143baccf/additional_recipes/ros-distro-mutex/recipe.yaml#L36

And also agreed that we need to pick a certain version; we can't possibly build against all different versions of opencv that are out there.

wolfv, 2021-05-17:
@graphic-zhang do you have a specific use-case where you ran into issues?

graphic-zhang, 2021-05-17:
> Hi @graphic-zhang I think we're using opencv 4.5 currently. What version are you interested in? We need to decide on some version at some point, otherwise we need to constantly update the code etc.

first , I apologize for my talk way，and i install ros by  ros-wiki website . 
 I have to point a situation ： even  I set OpenDIR or anyother ways to  set cmake  find and link myself opencv_lib (3.4.10 which has  i have built ) ,   cv_bridge will always link ros opencv lib 4.x.x ,  and then program will dead in opencv function which i have never encountered.
At last , i have to set Opencv_DIR to ros opencv in /usr/x86_64*/cmake/opencv4 , it worked , but it's too  Exclusive, i have to use ros opencv if i wanna use ros cv_bridge . is there any other way to make them more Compatible?

wolfv, 2021-05-17:
Hi, mixing the ROS installation from ROS wiki and the conda installation will not work. 

We have a version of `ros-noetic-cv-bridge` that should use a compatible OpenCV version (4.5).

Can you try wiht `conda install ros-noetic-cv-bridge` again?

graphic-zhang, 2021-05-18:
> Hi, mixing the ROS installation from ROS wiki and the conda installation will not work.
> 
> We have a version of `ros-noetic-cv-bridge` that should use a compatible OpenCV version (4.5).
> 
> Can you try wiht `conda install ros-noetic-cv-bridge` again?

I am very confused , maybe I should try more.

closed 2021-05-18 by graphic-zhang

Tobias-Fischer, 2021-05-18:
We may be able to help if you provide us with a detailed report on what you are trying to achieve, which exact commands you use, the output of `conda list`, whether you follow the instructions in the readme precisely, etc.