RoboStack/ros-noetic #132: MoveIt setup assistant not working

yureklimert, 2021-05-31:
[image]

After I downloaded moveIt into my workspace when I tried to open moveit_setup_assistant on my terminal with roslaunch moveit_setup_assistant setup_assistant.launch, It is sticking like image above and in the dock this image appears 

[image]
 
After that nothing happens and this is the screen when It is shuted down with ctrl + C

[image]

Tobias-Fischer, 2021-06-03:
Hiya @yureklimert,
Many thanks for reporting this bug. I've just uploaded a new version - please `mamba upgrade ros-noetic-moveit-setup-assistant` and it should work. Please re-open the issue if not.

@traversaro @wolfv - it's the same issue as with rviz - it needs building with macos version 10.9 instead of 10.14. We might need to dig again and/or only use 10.14 in the packages that actually need it. I'll update #69

closed 2021-06-03 by Tobias-Fischer

yureklimert, 2021-06-03:
Thank you for helping. I am using macOS 11.4 and everything works with no problem.

yureklimert, 2021-06-03:
Thank you for helping problem solved. Also I am using macOS 11.4 and everything works perfectly fine.

Tobias-Fischer, 2021-06-03:
Great, thanks for letting us know!