RoboStack/ros-noetic #134: rosserial on robostack in windows10

AtharvaBhorpe, 2021-06-07:
Hi! I tried to find any documentation for using rosserial on windows with robostack conda. I found a package called rosserial-server on Anaconda.org [https://anaconda.org/robostack/ros-noetic-rosserial-server](URL). Could someone please tell if rosserial is supported, and if yes, then how can I use it with rosserial-Arduino?

wolfv, 2021-06-07:
Hi @AtharvaBhorpe thanks for opening this issue. We have the following table of packages available through the RoboStack channel: https://robostack.github.io/noetic.html
The only builds missing are for rosserial and rosserial-client on win-64.

Could you make a pull request to this repository, adding `rosserial` to `vinca_win.yml`? once we merge it, `rosserial` and `rosserial-client` should become available on Windows, too.

However, just because we have builds for a specific package doesn't necessarily mean that the package works flawlessly on that platform since most ROS packages are almost only used on Linux. You'll have to try and see if there any issues...

wolfv, 2021-06-07:
@Tobias-Fischer @AtharvaBhorpe I just checked, there are a bunch of other rosserial packages not yet included:

```
│ ros-noetic-rosserial-arduino                               │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-chibios                               │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-embeddedlinux                         │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-mbed                                  │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-tivac                                 │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-vex-cortex                            │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-vex-v5                                │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-windows                               │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
│ ros-noetic-rosserial-xbee                                  │ ✖        │ ✖      │ ✖      │ ✖             │                                        │
```

Especially the rosserial-windows could be relevant?

AtharvaBhorpe, 2021-06-07:
@wolfv Thanks for a very prompt reply. From the robostack channel URL you mentioned, there's a support for rosserial-python. I will be using rosserial in arduino, which can be connected to ROScore by "rosrun rosserial_python serial_node.py". So when using this with arduino on Windows10, I think using the USB COM port of Windows10 (e.g. COM4) should work like the Linux (e.g. dev/tty/ACM0). If this works, then I think rosserial-windows support would not be necessary, at least immediately. I'll try using rosserial-python on conda with Windows10's arduino and get back to you.

AtharvaBhorpe, 2021-06-07:
I tried installing `conda install ros-noetic-rosserial-python`, but it gave some kind of errors.
I'm very new to coding, let alone understand the CMD Prompt. Could you please tell me if this is the right path, or I messed up somewhere..?

>  (robostackenv) C:\Users\athar\Documents\ros_ws>conda install ros-noetic-rosserial-python
> Collecting package metadata (current_repodata.json): done
> Solving environment: done
> 
> 
> ==> WARNING: A newer version of conda exists. <==
>   current version: 4.9.2
>   latest version: 4.10.1
> 
> Please update conda by running
> 
>     $ conda update -n base -c defaults conda
> 
> 
> 
> ## Package Plan ##
> 
>   environment location: C:\Users\athar\miniconda3\envs\robostackenv
> 
>   added / updated specs:
>     - ros-noetic-rosserial-python
> 
> 
> The following packages will be downloaded:
> 
>     package                    |            build
>     ---------------------------|-----------------
>     ros-noetic-rosserial-msgs-0.9.1|   py38h4b9bc1a_5          29 KB  robostack
>     ros-noetic-rosserial-python-0.9.1|   py38h4b9bc1a_5          48 KB  robostack
>     ------------------------------------------------------------
>                                            Total:          77 KB
> 
> The following NEW packages will be INSTALLED:
> 
>   ros-noetic-rosser~ robostack/win-64::ros-noetic-rosserial-msgs-0.9.1-py38h4b9bc1a_5
>   ros-noetic-rosser~ robostack/win-64::ros-noetic-rosserial-python-0.9.1-py38h4b9bc1a_5
> 
> 
> Proceed ([y]/n)? y
> 
> 
> Downloading and Extracting Packages
> ros-noetic-rosserial | 29 KB     | ############################################################################ | 100%
> ros-noetic-rosserial | 48 KB     | ############################################################################ | 100%
> Preparing transaction: done
> Verifying transaction: done
> Executing transaction: done
> 
> C:\Users\athar\Documents\ros_ws>SET DISTUTILS_USE_SDK=1
> 
> C:\Users\athar\Documents\ros_ws>SET MSSdk=1
> 
> C:\Users\athar\Documents\ros_ws>SET "VS_VERSION=15.0"
> 
> C:\Users\athar\Documents\ros_ws>SET "VS_MAJOR=15"
> 
> C:\Users\athar\Documents\ros_ws>SET "VS_YEAR=2017"
> 
> C:\Users\athar\Documents\ros_ws>set "MSYS2_ARG_CONV_EXCL=/AI;/AL;/OUT;/out"
> 
> C:\Users\athar\Documents\ros_ws>set "MSYS2_ENV_CONV_EXCL=CL"
> 
> C:\Users\athar\Documents\ros_ws>set "PY_VCRUNTIME_REDIST=\bin\vcruntime140.dll"
> 
> C:\Users\athar\Documents\ros_ws>set "CXX=cl.exe"
> 
> C:\Users\athar\Documents\ros_ws>set "CC=cl.exe"
[... 151 lines cut ...]
> 
> C:\Users\athar\Documents\ros_ws>CALL "VC\Auxiliary\Build\vcvars64.bat" -vcvars_ver=14.28
> The system cannot find the path specified.
> 
> C:\Users\athar\Documents\ros_ws>popd
> 
> C:\Users\athar\Documents\ros_ws>IF "Visual Studio 15 2017 Win64" == "" SET "CMAKE_GENERATOR=Visual Studio 16 2019"
> 
> C:\Users\athar\Documents\ros_ws>IF 2019 GEQ 2019 (IF "x64" == "" SET "CMAKE_GENERATOR_PLATFORM=x64" )
> 
> C:\Users\athar\Documents\ros_ws>call :GetWin10SdkDirHelper HKLM\SOFTWARE\Wow6432Node  1>nul 2>&1
> 
> C:\Users\athar\Documents\ros_ws>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE\Wow6432Node  1>nul 2>&1
> 
> C:\Users\athar\Documents\ros_ws>if errorlevel 1 call :GetWin10SdkDirHelper HKLM\SOFTWARE  1>nul 2>&1
> 
> C:\Users\athar\Documents\ros_ws>if errorlevel 1 call :GetWin10SdkDirHelper HKCU\SOFTWARE  1>nul 2>&1
> 
> C:\Users\athar\Documents\ros_ws>if errorlevel 1 exit /B 1
> 
> C:\Users\athar\Documents\ros_ws>exit /B 0
> 
> (robostackenv) C:\Users\athar\Documents\ros_ws>

```

Tobias-Fischer, 2021-06-07:
As far as I can see, `rosserial-python` was installed successfully. Can you try using it?

AtharvaBhorpe, 2021-06-08:
> (robostackenv) C:\WINDOWS\system32>rosrun rosserial_python serial_node.py /COM5
> '"\python.exe"' is not recognized as an internal or external command,
> operable program or batch file.
> 
> (robostackenv) C:\WINDOWS\system32>

Tobias-Fischer, 2021-06-08:
I think sth is wrong with your setup, also judging from #136. You should not need to play with the PATH variable when using conda. I am not a Windows expert though - maybe others can help?

AtharvaBhorpe, 2021-06-08:
I'll uninstall miniconda3 and robostackenv and install everything again and report back

traversaro, 2021-06-08:
I replicate @AtharvaBhorpe problem on my system:
~~~
conda create -n robostack134
conda activate robostack134
conda install -c conda-forge -c robostack ros-noetic-rosserial-python ros-noetic-rosbash
(robostack134) C:\Users\STraversaro>rosrun rosserial_python serial_node.py /COM5
'"\python.exe"' is not recognized as an internal or external command,
operable program or batch file.
~~~

AtharvaBhorpe, 2021-06-08:
I'll try using the above command. I'm reinstalling the robostack package by reinstalling the Miniconda3 itself. After this, as @traversaro replied in #136, I'll install VS2019 with the Development Support for C/C++. Thank you guys for being active and replying promptly!!

AtharvaBhorpe, 2021-06-08:
IT'S WORKING!!!!
First, the dependency "serial" was missing, so I installed "PySerial" using:

> `conda install pyserial`

Then, run roscore:

> `roscore`

Open new terminal and activate the robostackenv, then add the "PYTHONHOME" in Conda:

> `set PYTHONHOME=%CONDA_PREFIX%`

Then run the rosserial_python serial_node, and give the specific COM Port and Baud Rate:

> `rosrun rosserial_python serial_node.py COM5`

and Voila!
[image]
[image]

Thank you guys so much for the trouble. This is really cool that I can run rosserial on Windows10 directly, without using VMWare or WSL.
Really Great Work Guys!!!🙌🤍

traversaro, 2021-06-08:
> Open new terminal and activate the robostackenv, then add the "PYTHONHOME" in Conda:
>
> >    set PYTHONHOME=%CONDA_PREFIX%

FYI I noticed that this make conda misbehave. I will open an issue to fix it properly.

Tobias-Fischer, 2021-06-08:
Should be fixed by #139

AtharvaBhorpe, 2021-06-08:
Great! I think I'll close this issue then. Thanks for the support!

closed 2021-06-08 by AtharvaBhorpe

Tobias-Fischer, 2021-06-08:
Both issues re missing dependency of pyserial and not existing python home should be fixed after upgrading to the latest version.