RoboStack/ros-noetic #136: error when runnning 'catkin_make'

AtharvaBhorpe, 2021-06-07:
Hi! I just installed robostackenv (ros-noetic-desktop-full) using miniconda3. I made a "ros_ws" in the "Documents" directory in C:. I did `catkin_init_workspace`, which worked. But when I go 1 directory back (out of the "src" folder into the ros_ws folder) and run `catkin_make`, I get some "cmake error"

> (robostackenv) C:\Users\athar\Documents\ros_ws>catkin_make
> Base path: C:\Users\athar\Documents\ros_ws
> Source space: C:\Users\athar\Documents\ros_ws\src
> Build space: C:\Users\athar\Documents\ros_ws\build
> Devel space: C:\Users\athar\Documents\ros_ws\devel
> Install space: C:\Users\athar\Documents\ros_ws\install
> ####
> #### Running command: "cmake C:\Users\athar\Documents\ros_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\athar\Documents\ros_ws\devel -DCMAKE_INSTALL_PREFIX=C:\Users\athar\Documents\ros_ws\install -G NMake Makefiles" in "C:\Users\athar\Documents\ros_ws\build"
> ####
> CMake Error at C:/Users/athar/miniconda3/envs/robostackenv/Library/share/cmake-3.20/Modules/CMakeDetermineCCompiler.cmake:49 (message):
>   Could not find compiler set in environment variable CC:
> 
>   cl.exe.
> Call Stack (most recent call first):
>   CMakeLists.txt:6 (project)
> 
> 
> CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
> CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
> -- Configuring incomplete, errors occurred!
> See also "C:/Users/athar/Documents/ros_ws/build/CMakeFiles/CMakeOutput.log".
> Invoking "cmake" failed
> 
> (robostackenv) C:\Users\athar\Documents\ros_ws>

Did I installed the robostack package wrong? Any suggestion would be helpful.

Tobias-Fischer, 2021-06-07:
Hi @AtharvaBhorpe,
Please follow the instructions in the readme - in particular, these parts:
```
# optionally, install some compiler packages if you want to e.g. build packages in a catkin_ws - with conda:
conda install compilers cmake pkg-config make ninja catkin_tools
# or with mamba:
mamba install compilers cmake pkg-config make ninja catkin_tools

# on Windows, install the Visual Studio command prompt via Conda:
conda install vs2019_win-64
```

If you still have issues after running the recommended installation steps from the readme, feel free to re-open this issue.

closed 2021-06-07 by Tobias-Fischer

AtharvaBhorpe, 2021-06-08:
When I use the command of 'catkin_tools', it says there's no package available.

> (robostackenv) C:\WINDOWS\system32>conda install compilers cmake pkg-config make ninja catkin_tools
> Collecting package metadata (current_repodata.json): done
> Solving environment: failed with initial frozen solve. Retrying with flexible solve.
> Collecting package metadata (repodata.json): done
> Solving environment: failed with initial frozen solve. Retrying with flexible solve.
> 
> PackagesNotFoundError: The following packages are not available from current channels:
> 
>   - catkin_tools
> 
> Current channels:
> 
>   - https://conda.anaconda.org/robostack/win-64
>   - https://conda.anaconda.org/robostack/noarch
>   - https://conda.anaconda.org/conda-forge/win-64
>   - https://conda.anaconda.org/conda-forge/noarch
>   - https://repo.anaconda.com/pkgs/main/win-64
>   - https://repo.anaconda.com/pkgs/main/noarch
>   - https://repo.anaconda.com/pkgs/r/win-64
>   - https://repo.anaconda.com/pkgs/r/noarch
>   - https://repo.anaconda.com/pkgs/msys2/win-64
>   - https://repo.anaconda.com/pkgs/msys2/noarch
> 
> To search for alternate channels that may provide the conda package you're
> looking for, navigate to
> 
>     https://anaconda.org
> 
> and use the search bar at the top of the page.
> 
> 
> 
> (robostackenv) C:\WINDOWS\system32>

Tobias-Fischer, 2021-06-08:
Try just `conda install compilers cmake pkg-config make ninja`

AtharvaBhorpe, 2021-06-08:
The other packages are installed, but still, the ros_ws cannot perform 'catkin_make'. Should I try making a workspace outside C: drive?

Tobias-Fischer, 2021-06-08:
What is the error?

AtharvaBhorpe, 2021-06-08:
> (robostackenv) C:\Users\athar\Documents\ros_ws>catkin_make
> Base path: C:\Users\athar\Documents\ros_ws
> Source space: C:\Users\athar\Documents\ros_ws\src
> Build space: [...]

Tobias-Fischer, 2021-06-08:
Output of `conda list`?

AtharvaBhorpe, 2021-06-08:
> (robostackenv) C:\Users\athar\Documents\ros_ws>conda list
> # packages in environment at C:\Users\athar\miniconda3\envs\robostackenv:
> #
> # Name                    Version                   Build  Channel
> adwaita-icon-theme        40.1.1               h57928b3_1    conda-forge
> apr                       1.7.0                he38c35c_5    conda-forge
> assimp                    5.0.1                hc2aa0de_5    conda-forge
> atk-1.0                   2.36.0               h7222f49_4    conda-forge
> bcrypt                    3.2.0            py38h294d835_1    conda-forge
> boost                     1.74.0           py38h1266d08_3    conda-forge
> boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
> bzip2                     1.0.8                h8ffe710_4    conda-forge
> c-compiler                1.1.3                h8ffe710_0    conda-forge
> ca-certificates           2021.5.30            h5b45459_0    conda-forge
> cairo                     1.16.0            hb19e0ff_1008    conda-forge
> catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
> certifi                   2021.5.30        py38haa244fe_0    conda-forge
> cffi                      1.14.5           py38hd8c33c5_0    conda-forge
> clangdev                  5.0.0                   flang_3  [flang]  conda-forge
> cmake                     3.20.3               h39d44d4_0    conda-forge
> compilers                 1.1.3                h57928b3_0    conda-forge
> console_bridge            1.0.1                h5362a0b_0    conda-forge
> cppzmq                    4.7.1                h4324990_2    conda-forge
> cryptography              3.4.7            py38hd7da0ea_0    conda-forge
> curl                      7.77.0               h789b8ee_0    conda-forge
> cxx-compiler              1.1.3                h2d74725_0    conda-forge
> cycler                    0.10.0                     py_2    conda-forge
> defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
> distro                    1.5.0              pyh9f0ad1d_0    conda-forge
> dlfcn-win32               1.3.0                h0e60522_0    conda-forge
> docutils                  0.17.1           py38haa244fe_0    conda-forge
> eigen                     3.3.9                h2d74725_1    conda-forge
> empy                      3.3.4              pyh9f0ad1d_1    conda-forge
> epoxy                     1.5.8                h8d14728_0    conda-forge
> expat                     2.4.1                h39d44d4_0    conda-forge
> ffmpeg                    4.3.1                ha925a31_0    conda-forge
> flang                     5.0.0           he025d50_20180525    conda-forge
> flang_win-64              5.0.0           h13ae965_20180526    conda-forge
> flann                     1.9.1             h7dfa0ef_1008    conda-forge
> font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
> font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
> font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
> font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
> fontconfig                2.13.1            h1989441_1005    conda-forge
> fonts-conda-ecosystem     1                             0    conda-forge
> fonts-conda-forge         1                             0    conda-forge
> fortran-compiler          1.1.3                h9655429_0    conda-forge
> freeglut                  3.2.1                h0e60522_2    conda-forge
> freeimage                 3.18.0               hfcb8cd7_7    conda-forge
> freetype                  2.10.4               h546665d_1    conda-forge
> fribidi                   1.0.10               h8d14728_0    conda-forge
> gazebo                    11.5.0               h83ca5bb_1    conda-forge
> gdk-pixbuf                2.42.6               h1c5aac7_0    conda-forge
> getopt-win32              0.1                  h8ffe710_0    conda-forge
> gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
> glib                      2.68.2               h0e60522_2    conda-forge
> glib-tools                2.68.2               h0e60522_2    conda-forge
> gmock                     1.10.0               h2d74725_7    conda-forge
> graphite2                 1.3.13                     1000    conda-forge
> graphviz                  2.47.2               hefbd956_0    conda-forge
> gtest                     1.10.0               h2d74725_7    conda-forge
> gtk3                      3.24.28              h9a703b0_1    conda-forge
> gts                       0.7.6                h7c369d9_2    conda-forge
> harfbuzz                  2.8.1                hc601d6f_0    conda-forge
> hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
> hicolor-icon-theme        0.17                 h57928b3_2    conda-forge
> icu                       68.1                 h0e60522_0    conda-forge
> ilmbase                   2.5.5                h12d4b20_0    conda-forge
> intel-openmp              2021.2.0           h57928b3_616    conda-forge
> jasper                    2.0.14               h77af90b_2    conda-forge
[... 379 lines cut ...]
> xorg-libxext              1.3.4                h301d43c_0    conda-forge
> xorg-libxfixes            5.0.3             h301d43c_1004    conda-forge
> xorg-libxi                1.7.10               h301d43c_0    conda-forge
> xorg-libxinerama          1.1.4             hfa435f1_1001    conda-forge
[... cut ...]
> yaml                      0.2.5                he774522_0    conda-forge
> yaml-cpp                  0.6.3                ha925a31_4    conda-forge
> zeromq                    4.3.4                h0e60522_0    conda-forge
> zlib                      1.2.11            h62dcd97_1010    conda-forge
> zstd                      1.5.0                h6255e5f_0    conda-forge
> zziplib                   0.13.69              h1d00b33_1    conda-forge
> 
> (robostackenv) C:\Users\athar\Documents\ros_ws>

Tobias-Fischer, 2021-06-08:
I guess you have done this?

```
# reload environment to activate required scripts before running anything
# on Windows, please restart the Anaconda Prompt / Command Prompt!
conda deactivate
conda activate robostackenv
```

If yes, I am out of ideas. Any clue @wolfv @traversaro?

AtharvaBhorpe, 2021-06-08:
I did do that, and also restarted my PC. :(
Should I try reinstalling the robostack in a new venv?

traversaro, 2021-06-08:
Hi @AtharvaBhorpe, did you have installed on your system Visual Studio 2017 or 2019 with C++ support ( https://docs.microsoft.com/en-us/cpp/build/vscpp-step-0-installation?view=msvc-160 )? Differently from Linux or macOS, on Windows the compiler is not automatically installed by conda.

AtharvaBhorpe, 2021-06-08:
I installed Visual Studio 19 with the "Development for C++" extension. Still, I cannot build my ros_ws.

AtharvaBhorpe, 2021-06-08:
Is this what you requested??
[image]

traversaro, 2021-06-08:
In that terminal, can you try to run the `set` command and report the output?

AtharvaBhorpe, 2021-06-08:
I'm sorry, I didn't get which `set` command to run... Could you please elaborate?

traversaro, 2021-06-08:
The command is called `set`, it is used either to set environment variables (if arguments are passed), or just to get their value (if no arguments are passed). See https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/set_1 . It should be sufficient to type `set` and press enter to execute the command.

AtharvaBhorpe, 2021-06-08:
> (rosenv) C:\Windows\System32>set
> ALLUSERSPROFILE=C:\ProgramData
> APPDATA=C:\Users\athar\AppData\Roaming
> BITS=64
> BR_TG_PATH=D:\ATHARVA D DRIVE\OS\PROGRAM FILES\Automation Studio\AsTools\Technology Guarding
> CC=cl.exe
> CMAKE_GEN=Visual Studio 16 2019
> CMAKE_GENERATOR=Visual Studio 15 2017 Win64
> CMAKE_GENERATOR_PLATFORM=x64
> CMAKE_PLAT=x64
> CMAKE_PREFIX_PATH=C:/Users/athar/miniconda3/envs/rosenv/Library
> CommonProgramFiles=C:\Program Files\Common Files
> CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
> CommonProgramW6432=C:\Program Files\Common Files
> COMPUTERNAME=DESKTOP-P4PVNM6
> ComSpec=C:\WINDOWS\system32\cmd.exe
> CONDA_DEFAULT_ENV=rosenv
> CONDA_EXE=C:\Users\athar\miniconda3\condabin\..\Scripts\conda.exe
> CONDA_PREFIX=C:\Users\athar\miniconda3\envs\rosenv
> CONDA_PREFIX_1=C:\Users\athar\miniconda3
> CONDA_PROMPT_MODIFIER=(rosenv)
> CONDA_PYTHON_EXE=C:\Users\athar\miniconda3\python.exe
> CONDA_SHLVL=2
> CPATH=;C:\Users\athar\miniconda3\envs\rosenv\Library\include
> CXX=cl.exe
> DevEnvDir=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\
> DISTUTILS_USE_SDK=1
> DriverData=C:\Windows\System32\Drivers\DriverData
> Framework40Version=v4.0
> FrameworkDir=C:\Windows\Microsoft.NET\Framework64\
> FrameworkDir64=C:\Windows\Microsoft.NET\Framework64\
> FrameworkVersion=v4.0.30319
> FrameworkVersion64=v4.0.30319
> GAZEBO_MASTER_URI=http://localhost:11345
> GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
> GAZEBO_MODEL_PATH=C:/Users/athar/miniconda3/envs/rosenv\Library\share\gazebo-11\models;
> GAZEBO_PLUGIN_PATH=C:/Users/athar/miniconda3/envs/rosenv\Library\bin\gazebo-11\plugins;
> GAZEBO_RESOURCE_PATH=C:/Users/athar/miniconda3/envs/rosenv\Library\share\gazebo-11;
> GEM_HOME=C:\Users\athar\miniconda3\envs\rosenv\Library\share\rubygems
> HOME=C:\Users\athar
> HOMEDRIVE=C:
> HOMEPATH=\Users\athar
> LD_LIBRARY_PATH=;C:/Users/athar/miniconda3/envs/rosenv\Library\bin\gazebo-11\plugins
> LIB=;C:\Users\athar\miniconda3\envs\rosenv\Library\lib
> LIBPATH=C:\Windows\Microsoft.NET\Framework64\v4.0.30319
> LOCALAPPDATA=C:\Users\athar\AppData\Local
> LOGONSERVER=\\DESKTOP-P4PVNM6
> MSSdk=1
> MSYS2_ARG_CONV_EXCL=/AI;/AL;/OUT;/out
> MSYS2_ENV_CONV_EXCL=CL
> NUMBER_OF_PROCESSORS=12
> OGRE_RESOURCE_PATH=C:/Users/athar/miniconda3/envs/rosenv\Library\bin
> OneDrive=C:\Users\athar\OneDrive
> OS=Windows_NT
> Path=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\\Extensions\Microsoft\IntelliCode\CLI;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\devinit;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\;C:\Users\athar\miniconda3\envs\rosenv\Library\share\rubygems\bin;C:/Users/athar/miniconda3/envs/rosenv/Library\bin;C:/Users/athar/miniconda3/envs/rosenv/Library\lib;;C:/Users/athar/miniconda3/envs/rosenv\Library\bin\gazebo-11\plugins;C:\Users\athar\miniconda3\envs\rosenv;C:\Users\athar\miniconda3\envs\rosenv\Library\mingw-w64\bin;C:\Users\athar\miniconda3\envs\rosenv\Library\usr\bin;C:\Users\athar\miniconda3\envs\rosenv\Library\bin;C:\Users\athar\miniconda3\envs\rosenv\Scripts;C:\Users\athar\miniconda3\envs\rosenv\bin;C:\Users\athar\miniconda3\condabin;C:\Program Files (x86)\VMware\VMware Player\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Polyspace\R2021a\runtime\win64;C:\Program Files\Polyspace\R2021a\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;C:\Users\athar\miniconda3;C:\Users\athar\miniconda3\Library\mingw-w64\bin;C:\Users\athar\miniconda3\Library\usr\bin;C:\Users\athar\miniconda3\Library\bin;C:\Users\athar\miniconda3\Scripts;C:\Users\athar\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Common Files\Hilscher GmbH\TLRDecode;C:\Users\athar\miniconda3\python.exe;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja
> PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
> PKG_CONFIG_PATH=C:/Users/athar/miniconda3/envs/rosenv/Library\lib\pkgconfig
> PROCESSOR_ARCHITECTURE=AMD64
[... cut ...] gram Files\Polyspace\R2021a\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;C:\Users\athar\miniconda3;C:\Users\athar\miniconda3\Library\mingw-w64\bin;C:\Users\athar\miniconda3\Library\usr\bin;C:\Users\athar\miniconda3\Library\bin;C:\Users\athar\miniconda3\Scripts;C:\Users\athar\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Common Files\Hilscher GmbH\TLRDecode;C:\Users\athar\miniconda3\python.exe
> __VSCMD_script_err_count=0
> 
> (rosenv) C:\Windows\System32>C

traversaro, 2021-06-08:
The `PATH` values seems ok,  In this terminal you still have the same problem you reported before? Can you also run `where cl.exe` in the same terminal?

AtharvaBhorpe, 2021-06-08:
> (rosenv) C:\Windows\System32>where cl.exe
> INFO: Could not find files for the given pattern(s).
> 
> (rosenv) C:\Windows\System32>

traversaro, 2021-06-08:
Are you sure that Visual Studio is correctly installed? Because it seems that the `cl.exe` C++ compiler can't be found. For example for an environment of mine i have:
~~~
(wbtenv) C:\Users\STraversaro>where cl.exe
C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.28.29910\bin\Hostx64\x64\cl.exe
~~~

AtharvaBhorpe, 2021-06-08:
Yes, I think it's installed properly. I'm able to find the directory that you mentioned from Files Explorer. Is there any way I can manually change the directory?