RoboStack/ros-noetic #143: The road to _more_ packages on OSX/Win/Linux-aarch

Tobias-Fischer, 2021-06-10:
Hi @wolfv,
I think it would be a good idea to follow a similar approach as in #94 and build as many packages as possible for the other platforms. I am not sure which script you used to create the list(s) in #94 (successful vs unsuccessful builds); could you please add it to the repo? Then we can build the successful packages for the other platforms as well, which will be nice.

Let me know if/how I can assist.

Best, Tobi

wolfv, 2021-06-11:
Hi, IIRC I generated all recipes with vinca using the `build_all: true` option (I added that option here: https://github.com/RoboStack/vinca/commit/e7dfa6d53746d59adb03c4d5a7c99d79fea1a5fa#diff-104bf9378a734ee0fc4cb7d8842f9cf7b422f2bf2dc6a8b5731baf830a6601ca)

I probably generated multiple recipes.

Then I run `boa build recipes/ --skip-existing=fast`. The `--skip-existing=fast` option makes a less accurate comparison of wether a package has already been built (it only checks the name + version and doesn't compute the hash).

Then I let it run until a package doesn't build. Once that happens, I manually moved them from `mv ./recipes/mybrokenpkg ./broken_recipes` and re-ran boa. It was quite the manual process, unfortunately.

wolfv, 2021-06-11:
We could also extend boa to "continue-on-failure" and automatically remove packages with unsatisfiable deps (because an earlier build didn't go through). That could be quite cool (and would be similar to how catkin does things with the `-c` flag).

Tobias-Fischer, 2021-06-11:
Oh gosh, that must have been cumbersome ;). And it would be even more painful on the other platforms as more packages will fail. A continue on failure feature would be great! It shouldn't be too bad to implement, is it?

wolfv, 2021-06-11:
No, it shouldn't be too bad ... just need to continue skipping all packages where some parent has been skipped ...

wolfv, 2021-06-29:
On Windows, I got the following to build (not tested, though)

```
06/28/2021  11:03 PM            36,328 ros-noetic-abb-egm-msgs-0.5.2-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:04 PM            58,928 ros-noetic-abb-rapid-msgs-0.5.2-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:05 PM            97,665 ros-noetic-abb-robot-msgs-0.5.2-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:06 PM            32,303 ros-noetic-ackermann-msgs-1.0.2-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:08 PM            44,549 ros-noetic-actionlib-tools-1.13.2-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:16 PM            10,852 ros-noetic-allocators-1.0.25-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:17 PM           119,412 ros-noetic-app-manager-1.2.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:18 PM             8,239 ros-noetic-arbotix-firmware-0.11.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:19 PM            43,629 ros-noetic-arbotix-msgs-0.11.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:21 PM            87,114 ros-noetic-assisted-teleop-0.3.4-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:22 PM            11,741 ros-noetic-async-comm-0.2.1-py38h7c8cae0_9.tar.bz2
06/28/2021  11:24 PM           659,957 ros-noetic-audibot-description-0.2.1-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:25 PM            29,468 ros-noetic-audio-common-msgs-0.3.11-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:27 PM            46,088 ros-noetic-autoware-can-msgs-1.14.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:29 PM           110,887 ros-noetic-autoware-config-msgs-1.14.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:29 PM            28,010 ros-noetic-autoware-lanelet2-msgs-1.14.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:31 PM            92,501 ros-noetic-autoware-map-msgs-1.14.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:32 PM            48,844 ros-noetic-autoware-system-msgs-1.14.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:33 PM            55,240 ros-noetic-auv-msgs-0.1.1-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:39 PM            23,988 ros-noetic-baldor-0.1.3-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:46 PM            96,623 ros-noetic-calibration-msgs-0.10.15-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:47 PM            28,096 ros-noetic-can-msgs-0.8.5-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:48 PM           136,527 ros-noetic-capabilities-0.3.1-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:50 PM            82,179 ros-noetic-cartesian-control-msgs-0.1.0-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:51 PM            12,633 ros-noetic-cartesian-interface-0.1.3-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:52 PM            26,932 ros-noetic-cartesian-msgs-0.0.3-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:55 PM            53,703 ros-noetic-catkin-virtualenv-0.6.1-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:57 PM            90,940 ros-noetic-cob-actions-0.7.4-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:58 PM            62,252 ros-noetic-cob-android-msgs-0.1.9-py38h4b9bc1a_9.tar.bz2
06/28/2021  11:59 PM            29,528 ros-noetic-cob-android-resource-server-0.1.9-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:00 AM             9,713 ros-noetic-cob-android-settings-0.1.9-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:04 AM           369,280 ros-noetic-cob-cam3d-throttle-0.6.17-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:05 AM            46,524 ros-noetic-cob-control-mode-adapter-0.8.13-py38h7c8cae0_9.tar.bz2
06/29/2021  12:07 AM            40,537 ros-noetic-cob-control-msgs-0.8.13-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:08 AM           106,302 ros-noetic-cob-default-env-config-0.6.12-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:11 AM         8,809,548 ros-noetic-cob-description-0.7.4-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:11 AM             9,182 ros-noetic-cob-docker-control-0.6.10-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:21 AM            47,214 ros-noetic-cob-gazebo-tools-0.7.5-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:25 AM            40,710 ros-noetic-cob-interactive-teleop-0.6.21-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:28 AM           712,413 ros-noetic-cob-lookat-action-0.7.5-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:32 AM            20,276 ros-noetic-cob-moveit-config-0.7.5-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:33 AM            50,373 ros-noetic-cob-msgs-0.7.4-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:34 AM            15,151 ros-noetic-cob-navigation-config-0.6.11-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:36 AM            80,836 ros-noetic-cob-perception-msgs-0.6.17-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:36 AM             8,470 ros-noetic-cob-reflector-referencing-0.6.10-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:37 AM             9,026 ros-noetic-cob-safety-controller-0.6.10-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:41 AM            40,701 ros-noetic-cob-srvs-0.7.4-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:42 AM             8,485 ros-noetic-cob-supported-robots-0.6.15-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:44 AM            50,060 ros-noetic-cob-vision-utils-0.6.17-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:48 AM            77,572 ros-noetic-computer-status-msgs-2.0.0-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:52 AM        15,892,175 ros-noetic-control-box-rst-0.0.7-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:53 AM            13,418 ros-noetic-controller-manager-tests-0.19.5-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:55 AM            36,700 ros-noetic-costmap-cspace-msgs-0.8.0-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:58 AM           150,521 ros-noetic-cv-camera-0.5.0-py38h4b9bc1a_9.tar.bz2
06/29/2021  12:59 AM            59,805 ros-noetic-darknet-ros-msgs-1.1.5-py38h4b9bc1a_9.tar.bz2
06/29/2021  01:00 AM            52,247 ros-noetic-dataspeed-pds-msgs-1.0.6-py38h4b9bc1a_9.tar.bz2
[... cut ...]
06/29/2021  04:13 AM            30,992 ros-noetic-mouse-teleop-0.4.0-py38h4b9bc1a_9.tar.bz2
06/29/2021  04:14 AM            58,958 ros-noetic-move-base-sequence-0.0.1-py38h4b9bc1a_9.tar.bz2
06/29/2021  04:17 AM            61,746 ros-noetic-moveit-python-0.4.2-py38h4b9bc1a_9.tar.bz2
06/29/2021  04:19 AM             5,484 ros-noetic-moveit-resources-0.8.0-py38h4b9bc1a_9.tar.bz2
06/29/2021  04:21 AM             5,461 ros-noetic-moveit-runtime-1.1.5-py38h4b9bc1a_9.tar.bz2
```

I think the build stopped in ros-noetic-mpc-local-planner, unfortunately.

Tobias-Fischer, 2021-06-29:
Trigged a build of those packages + the osx packages that built successfully: https://github.com/RoboStack/ros-noetic/commit/388dee6a0ae0c5a3e54ade73dac0b2db383f27bb

Let's see ;)

Tobias-Fischer, 2021-11-22:
I think https://github.com/RoboStack/ros-noetic/issues/94 is enough

closed 2021-11-22 by Tobias-Fischer