RoboStack/ros-noetic #144: libuvc compile issues

Tobias-Fischer, 2021-06-11:
When trying to compile libuvc_camera, I get the following compile issues:
```
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `memcpy@GLIBC_2.14'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `clock_gettime@GLIBC_2.17'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `fcntl64@GLIBC_2.28'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `getrandom@GLIBC_2.25'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `getauxval@GLIBC_2.16'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `reallocarray@GLIBC_2.26'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `gettid@GLIBC_2.30'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `name_to_handle_at@GLIBC_2.14'
/home/tobias/mambaforge/conda-bld/ros_1623373869396/_build_env/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env/lib/libudev.so.0: undefined reference to `__explicit_bzero_chk@GLIBC_2.25'
```

Any ideas how to fix these?

wolfv, 2021-06-11:
ah, interesting. Any idea which package ships libudev? I thought libudev would be taken from the host system.
Maybe you could check `ldd libudev.so` and what the output is?

`clock_gettime` was usually fixed by adding a link with `rt` but I am not sure that is the right thing to do here.

With libudev there is also the problem that centos 6 ships with `libudev.so.0` and more recent Ubuntu / Fedora etc. versions ship with `libudev.so.1` and in the past that led to a couple of issues, I believe. But if we have a working version of libudev as conda package then maybe that isn't a problem anymore.

wolfv, 2021-06-11:
https://github.com/conda-forge/gnuradio-feedstock/issues/23

Tobias-Fischer, 2021-06-11:
```
ldd /home/tobias/mambaforge/conda-bld/ros_1623373869396/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/lib/libudev.so.0
	linux-vdso.so.1 (0x00007ffe8efeb000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f328b628000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f328b436000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f328b693000)
```

Looks okay to me?

wolfv, 2021-06-11:
Ah, ok, but it it's linking `libudev.so.0 -> libudev.so.1`. That could _potentially_ be the source of trouble.

Tobias-Fischer, 2021-06-11:
Yeah - there is no libudev.so.0 in newer systems ..

Tobias-Fischer, 2021-06-17:
Fixed - all that needed to do is pull in the libudev and libudev-devel cdt's ..

closed 2021-06-17 by Tobias-Fischer