RoboStack/ros-noetic #148: Saving environment variables for sourcing catkin_ws/devel/setup.sh

tadteo, 2021-06-28:
Hi,
I am trying to use ros with conda, however, in my local configuration in the .bashrc file I have added some lines to source the file   `source /home/$USER/catkin_ws/devel/setup.bash` automatically while opening the shell. I wanna do the same thing while using the ros environment within conda. 
I was following the tutorial from [conda docs](https://conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#saving-environment-variables), to do that, however, two problems arise.

1. the lines like `source /home/$USER/catkin_ws/devel/setup.bash` containing source are not working (hence I am not able to find the packages in my catkin_ws using rosrun or roslaunch).
2. if I try to modify the variable `$ROS_PACKAGE_PATH` the variable is overridden by the one in conda (`/home/my_name/anaconda3/envs/ros/share`). This is particularly useful when working for example with the PX4 stack since is useful to add the following path to the variable:
```
export $ROS_PACKAGE_PATH:/home/$USER/PX4-Autopilot:/home/$USER/PX4-Autopilot/Tools/sitl_gazebo
```

Some more info I am using Ubuntu 20.04 and ros noetic.

Tobias-Fischer, 2021-06-28:
Hi @tadteo,
That's a cool question. Some questions in turn:
1. have you (re)built the catkin_ws after sourcing the robostack environment?
2. `source /home/$USER/catkin_ws/devel/setup.bash` should work. What error do you get?
3. Can you try adding `source /home/$USER/catkin_ws/devel/setup.bash && export $ROS_PACKAGE_PATH:/home/$USER/PX4-Autopilot:/home/$USER/PX4-Autopilot/Tools/sitl_gazebo:$ROS_PACKAGE_PATH` to the end of `/Users/tobias/opt/anaconda3/envs/robostackenv/share/ros-noetic-catkin_activate.sh` and re-activate the environment?

tadteo, 2021-06-30:
Hi @Tobias-Fischer,

1.
I had recompiled the catkin_ws, but I had both ros on system-level and with conda.
I have then removed both ros and anaconda, and then reinstall it clean just with conda. Added the missing package and then I was able to compile the catkin_ws normally.

2. 
`source /home/$USER/catkin_ws/devel/setup.bash` didn't give any error it was just not sourcing the file. 

I'm trying now the point 3.
I will update you hopefully before the end of the day.

tadteo, 2021-06-30:
Hi again @Tobias-Fischer ,

I just tried point 3.

I tried both using as in the guide on conda docs so:
```
cd %CONDA_PREFIX%
mkdir .\etc\conda\activate.d
mkdir .\etc\conda\deactivate.d
type NUL > .\etc\conda\activate.d\env_vars.bat
type NUL > .\etc\conda\deactivate.d\env_vars.bat
```

and then modifying the file  `$CONDA_PREFIX/etc/conda/activate.d/env_vars.sh` with this content:
```
# !/usr/bin/env bash
source /home/$USER/catkin_ws/devel/setup.bash && export ROS_PACKAGE_PATH=/home/$USER/PX4-Autopilot:/home/$USER/PX4-Autopilot/Tools/sitl_gazebo:$ROS_PACKAGE_PATH
```

I have tried to do the same also adding a file in the folder/file you propose `/home/my_name/anaconda3/envs/ros/share/ros-noetic-catkin_activate.sh

Tried different combination with or without `source`, with or without `PATH` but none of them was actually working.

The path  `$ROS_PACKAGE_PATH` remain `/home/matteo/anaconda3/envs/ros/share`
and if trying to do `roslaunch my_package my_launchfile` is not working neither.

If instead I source the file manually after having activate the environment I can access my_package with ros.

Tobias-Fischer, 2021-06-30:
Hi @tadteo,
That is weird - the file `/home/my_name/anaconda3/envs/ros/share/ros-noetic-catkin_activate.sh` should already exist and it should be necessary for you to create it.

Can you please provide the output of `conda list`?

tadteo, 2021-07-01:
Hi @Tobias-Fischer 

You can find it attached here:
[conda_list_ros_env.txt](https://github.com/RoboStack/ros-noetic/files/6748393/conda_list_ros_env.txt)

I have also installed the `ros-noetic-desktop-full` package since I had to use gazebo and other packages as well, so it is a bit longer than normal, anyway the problem was there anyway.

Tobias-Fischer, 2021-07-01:
This looks okay. I just realized that you might have misread the file location of `ros-noetic-catkin_activate.sh`. It lives in `$CONDA_PREFIX/share/ros-noetic-catkin_activate.sh`. Could you please check whether it exists, and re-apply the suggestion above?

tadteo, 2021-07-02:
Hi @Tobias-Fischer ,

No, unfortunately, the file doesn't exist.

To double check, I have also created a second environment as in here (https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3)

I went to `$CONDA_PREFIX/share/` and printed the `ls` command, here the output:
[ros_env_share.txt](https://github.com/RoboStack/ros-noetic/files/6756192/ros_env_share.txt)

wolfv, 2021-07-02:
Weird, there should be a $prefix/etc/conda/activate.d/... folder that contains the activation scripts

tadteo, 2021-07-02:
Hi @wolfv 
Thank you for the input!

I finally found the `ros-noetic-catkin_activate.sh` file originally created whit the environment, it lays under `$CONDA_PREFIX/etc/conda/activate.d/` and not in `$CONDA_PREFIX/share/`.

So I have modified the `$CONDA_PREFIX/etc/conda/activate.d/ros-noetic-catkin_activate.sh` file adding the lines needed as @Tobias-Fischer suggested and now it works.

Thank you both for the help, I will close the issue now.

closed 2021-07-02 by tadteo