RoboStack/ros-noetic #149: Problem installing ROS OS on Mac

seyedrezabazyar, 2021-07-16:
I encountered this error during installation

```
(base) MyUsername ~ % conda create -n robostackenv python=3.8

Collecting package metadata (current_repodata.json): failed

UnavailableInvalidChannel: The channel is not accessible or is invalid.
  channel name: robstack
  channel url: https://conda.anaconda.org/robstack
  error code: 404

You will need to adjust your conda configuration to proceed.
Use `conda config --show channels` to view your configuration's current state,
and use `conda config --show-sources` to view config file locations.
```

AND

```
(base) MyUsername ~ % conda config --show channels
channels:
  - robostack
  - robstack
  - conda-forge
  - defaults
(base) MyUsername ~ % conda config --show-sources
==> /Users/seyedrezabazyar/.condarc <==
channel_priority: strict
channels:
  - robostack
  - robstack
  - conda-forge
  - defaults
```

How do I install the ROS OS on Mac operating system?

wolfv, 2021-07-16:
why did you add `robstack`? you should remove it and it should work :)

closed 2021-07-16 by wolfv

seyedrezabazyar, 2021-07-16:
The same error occurred while deleting
How do I uninstall robstack?

wolfv, 2021-07-16:
you need to open this file (`/Users/seyedrezabazyar/.condarc`) and remove the entry.

seyedrezabazyar, 2021-07-16:
The above problem was solved. Now there is another problem
I installed everything. Now there is another problem

```
(base) MyUsername ~ % conda activate ros              
(ros) MyUsername ~ % roscore
zsh: command not found: roscore
```

How do I install roscore?

seyedrezabazyar, 2021-07-16:
How to install ROS on Mac OS?  (Big Sur 11.4)

wolfv, 2021-07-16:
`conda install ros-noetic-ros-core`?

seyedrezabazyar, 2021-07-16:
> `conda install ros-noetic-ros-core`?

Again, the problem was not solved
Why is there no training resource for installing ROS on the Internet?

Tobias-Fischer, 2021-07-16:
There is :). See the readme in this repository, if you follow it closely, go you should get a working setup. There is also https://wiki.purduearc.com/wiki/tutorials/setup-ros

seyedrezabazyar, 2021-07-16:
> There is :). See the readme in this repository, if you follow it closely, go you should get a working setup. There is also https://wiki.purduearc.com/wiki/tutorials/setup-ros

Not installed on Mac :(

wolfv, 2021-07-16:
maybe you can run `conda create -n myrosenv -c robostack -c conda-forge ros-noetic-ros-core` and then `conda activate myrosenv` that should work (just tested locally).

seyedrezabazyar, 2021-07-16:
```
MyUsername ~ % roscore
zsh: command not found: roscore
yUsernameM ~ % conda activate myrosenv
(myrosenv) MyUsername ~ % roscore
... logging to /Users/seyedrezabazyar/.ros/log/3c9fbe44-e67f-11eb-a1a2-a0999b0af51d/roslaunch-SRB.local-10560.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://SRB.local:51804/
ros_comm version 1.15.9

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES

auto-starting new master
process[master]: started with pid [10568]
ROS_MASTER_URI=http://SRB.local:11311/

setting /run_id to 3c9fbe44-e67f-11eb-a1a2-a0999b0af51d
process[rosout-1]: started with pid [10575]
started core service [/rosout]
```
**1- why "conda activate myrosenv"???!!!  I would like to be activated with this command >>> «conda activate myrosenv»**
**2- Now how do I run the ROS operating system???**

Tobias-Fischer, 2021-07-17:
Re Q1: the two commands look exactly the same to me, so I'm not sure what you mean.
Re Q2: running `roscore` is running the ROS master. You can now continue with any ROS tutorial on the official ROS homepage.

seyedrezabazyar, 2021-07-17:
I apologize for the first question. I typed the question incorrectly
why "conda activate myrosenv"???!!! I would like to be activated with this command >>> «conda activate ros»

```
MyUsername ~ % conda activate ros
(ros) MyUsername ~ % roscore
zsh: command not found: roscore
(ros) MyUsername ~ % conda activate         
(base) MyUsername ~ % roscore
zsh: command not found: roscore
(base) MyUsername ~ % conda activate myrosenv
(myrosenv) MyUsername ~ % roscore
... logging to /Users/seyedrezabazyar/.ros/log/32f02842-e6c6-11eb-a9e8-a0xxxxxxxxxx/roslaunch-SRB.local-1914.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

RLException: Unable to contact my own server at [http://SRB.local:49426/].
This usually means that the network is not configured properly.

A common cause is that the machine cannot connect to itself.  Please check
for errors by running:

	ping srb.local

For more tips, please see

	http://wiki.ros.org/ROS/NetworkSetup

The traceback for the exception was written to the log file
```
Watch this video
`https://www.youtube.com/watch?v=zF7Pbq4Puvg`

I did everything like the video. But in the end the rviz command does not work

```
(myrosenv) MyUsername ~ % rviz
zsh: command not found: rviz
```

Are you sure you want to install it on your Mac? (MacBook Pro Intel processor - Big Sur 11.4)

wolfv, 2021-07-17:
When you do 'conda create -n xyz ...' the -n stands for name and gives a name to the environment. Depending on that parameter you need to activate the environment (in this case 'xyz')

seyedrezabazyar, 2021-07-17:
After 48 hours of trying, I did not get any results
MAC OS X is not officialy supported for ROS. Trying to install anything more than the "bare-bones" version of ROS will most probably result in a myriad of problems...

last question: How I completely remove all ros from my system?

wolfv, 2021-07-17:
You can install ros-noetic-rviz for a GUI application.
But it's hard to explain this when you're trying to learn many things at the same time (conda & ros)

To remove everything you can just delete the miniconda/miniforge folder

traversaro, 2021-07-17:
I did not followed all the issue in detail, but it may be worth to try running the same commands in the `bash` terminal instead of the `zsh` default shell of macOs, if everything works in bash but not in zsh it could be a zsh specific issue.

seyedrezabazyar, 2021-07-17:
The new version of Mac has many incompatibilities
I suggest you publish the installation tutorial on YouTube. Because many users like me had problems
I had to install Ubuntu Linux on a virtual machine so I could use ROS

Tobias-Fischer, 2021-07-17:
What do you mean by "new version of Mac"? Big Sur? There should not be any problems at all. If you closely follow the instructions in our readme it should work. We recommend following them 1:1 and once you are familiar with how things work you can then change environment names etc, but first go for 1:1 copy of commands.

Tobias-Fischer, 2021-07-17:
Also note that we are just a bunch of volunteers with limited time who develop RoboStack on the side to their full time jobs, RoboStack is not officially supported by ROS.