RoboStack/ros-noetic #155: Installing the Universal Robot driver

nberliner, 2021-08-02:
### Overview
I'm trying to get the [Universal Robot drivers](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) to work in the Robostack environment. I've seen #67 but know little to nothing about building packages.

As I only need the driver installed locally, I tried to follow the UR [build instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building) but am running into errors. I can build the drivers just fine when I am installing ROS via `apt` instead of using RoboStack.

### Setup
- Fresh install of Ubuntu 20.04 LTS in virtualbox
- Install miniconda and create a robostackenv (based on your [installation guide](https://github.com/RoboStack/ros-noetic#installation))
- Follow the UR [build instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)

### Issue 1
Running `rosdep install --from-paths src --ignore-src -y` after cloning the UR repository(s) fails with the following error
```console
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: No definition of [socat] for OS [robostack]
```
I could fix that by adding `socat` to the `robostack.yaml` file and adjust the `20-default.list` file to point to my fork (see [here](https://github.com/nberliner/ros-noetic/commit/33823d47ec26fd5e5fccb58922e6e022830bb5c0)). Is that the right way to fix that?

### Issue 2
After fixing the first issue I ran `catkin_make` with the following result;
```console
[ 90%] Linking CXX executable /home/niklas/catkin_ws/devel/lib/ur_robot_driver/robot_state_helper
/home/niklas/miniconda3/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lconsole_bridge::console_bridge
collect2: error: ld returned 1 exit status
make[2]: *** [Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/robot_state_helper.dir/build.make:157: /home/niklas/catkin_ws/devel/lib/ur_robot_driver/robot_state_helper] Error 1
make[1]: *** [CMakeFiles/Makefile2:7025: Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/robot_state_helper.dir/all] Error 2
[ 91%] Linking CXX shared library /home/niklas/catkin_ws/devel/lib/libur_robot_driver_plugin.so
/home/niklas/miniconda3/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/9.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lconsole_bridge::console_bridge
collect2: error: ld returned 1 exit status
make[2]: *** [Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/ur_robot_driver_plugin.dir/build.make:157: /home/niklas/catkin_ws/devel/lib/libur_robot_driver_plugin.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:7513: Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/ur_robot_driver_plugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j4 -l4" failed
```
While trying to understand what is going on with the console_bride library, I came across a stackoverflow [answer](https://stackoverflow.com/a/21647591) and checked the output of `ld -lconsole_bridge --verbose` which suggested that the library could not be found. However running a simple `find` gave me some results,
```console
niklas@niklas-VirtualBox:~$ sudo find / -iname '*console_bridge.*'
find: '/run/user/1000/doc': Permission denied
find: '/run/user/1000/gvfs': Permission denied
/home/niklas/miniconda3/envs/robostackenv/lib/libconsole_bridge.so.1.0
/home/niklas/miniconda3/envs/robostackenv/lib/pkgconfig/rosconsole_bridge.pc
/home/niklas/miniconda3/envs/robostackenv/lib/pkgconfig/console_bridge.pc
/home/niklas/miniconda3/envs/robostackenv/lib/libconsole_bridge.so
/home/niklas/miniconda3/envs/robostackenv/lib/librosconsole_bridge.so
/home/niklas/miniconda3/pkgs/console_bridge-1.0.1-h4bd325d_0/lib/libconsole_bridge.so.1.0
/home/niklas/miniconda3/pkgs/console_bridge-1.0.1-h4bd325d_0/lib/pkgconfig/console_bridge.pc
/home/niklas/miniconda3/pkgs/console_bridge-1.0.1-h4bd325d_0/lib/libconsole_bridge.so
/home/niklas/miniconda3/pkgs/ros-noetic-rosconsole-bridge-0.5.4-py38hcda8483_6/lib/pkgconfig/rosconsole_bridge.pc
/home/niklas/miniconda3/pkgs/ros-noetic-rosconsole-bridge-0.5.4-py38hcda8483_6/lib/librosconsole_bridge.so
```
I created a symlink to a path where `ld` is looking for the library and discovered that `libstdc++.so.6` wasn't found either. I fixed another symlink so that `libstdc++.so.6` could be found as well but `ld -lconsole_bridge --verbose` still complains with
```console
ld: warning: cannot find entry symbol _start; not setting start address
ld: /home/niklas/miniconda3/envs/robostackenv/bin/../x86_64-conda-linux-gnu/sysroot/lib64/./libstdc++.so.6: undefined reference to `aligned_alloc@GLIBC_2.16'
ld: /home/niklas/miniconda3/envs/robostackenv/bin/../x86_64-conda-linux-gnu/sysroot/lib64/./libstdc++.so.6: undefined reference to `clock_gettime@GLIBC_2.17'
ld: link errors found, deleting executable `a.out'
```

You probably can imagine that `catkin_make` still fails. There are definitely too many things I don't fully understand here and any help is much appreciated.

Tobias-Fischer, 2021-08-02:
Hi,
Many thanks for giving RoboStack a go :)!

I added `socat` to the robostack.yaml here, so no need for your fork anymore.

I was able to compile the UR Driver with the following changes:

```
diff --git a/ur_calibration/CMakeLists.txt b/ur_calibration/CMakeLists.txt
index 3563940..7951b67 100644
--- a/ur_calibration/CMakeLists.txt
+++ b/ur_calibration/CMakeLists.txt
@@ -11,6 +11,7 @@ find_package(catkin REQUIRED COMPONENTS
 find_package(Eigen3 REQUIRED)
 find_package(yaml-cpp REQUIRED)
 find_package(ur_client_library REQUIRED)
+find_package(console_bridge)
 
 # create alias for the '_INCLUDE_DIR' variable, to help Catkin export it
 set(YAML_CPP_INCLUDE_DIRS ${YAML_CPP_INCLUDE_DIR})
@@ -50,6 +51,7 @@ target_link_libraries(calibration_correction
   ${catkin_LIBRARIES}
   ${YAML_CPP_LIBRARIES}
   ur_client_library::urcl
+  console_bridge::console_bridge
 )
 
 install(DIRECTORY launch
diff --git a/ur_robot_driver/CMakeLists.txt b/ur_robot_driver/CMakeLists.txt
index e208173..838e895 100644
--- a/ur_robot_driver/CMakeLists.txt
+++ b/ur_robot_driver/CMakeLists.txt
@@ -8,6 +8,8 @@ if(NOT CMAKE_CONFIGURATION_TYPES AND NOT CMAKE_BUILD_TYPE)
   set(CMAKE_BUILD_TYPE RelWithDebInfo)
 endif()
 
+find_package(console_bridge REQUIRED)
+
 find_package(catkin REQUIRED
   COMPONENTS
     actionlib
@@ -92,7 +94,7 @@ add_dependencies(ur_robot_driver_plugin ${${PROJECT_NAME}_EXPORTED_TARGETS} ${ca
 add_library(urcl_log_handler
   src/urcl_log_handler.cpp
 )
-target_link_libraries(urcl_log_handler ${catkin_LIBRARIES} ur_client_library::urcl)
+target_link_libraries(urcl_log_handler ${catkin_LIBRARIES} ur_client_library::urcl console_bridge::console_bridge)
 
 add_executable(ur_robot_driver_node
   src/dashboard_client_ros.cpp
```

No need for any symlinks - I would suggest to remove them, you could run into issues down the track having them.

Let us know if it works.

nberliner, 2021-08-04:
Many thanks for your help (and for RoboStack :))!

Unfortunately your changes result in another error. Here is the output of the last few lines after running `catkin_make`
```console
[100%] Linking CXX executable /home/niklas/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node
[100%] Built target ur_robot_driver_node
make[2]: *** [Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/build.make:104: Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7819: Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j4 -l4" failed
``` 
Further up, there is an error shown although I don't know if it is related. There are quite a few `BOOST_PRAGMA_MESSAGE` messages showing up.
```console
/home/niklas/miniconda3/envs/robostackenv/include/ur_client_library/primary/primary_parser.h:76:13: error: expected ')' before 'PRIu32'
   76 |             URCL_LOG_WARN("Invalid sub-package size of %" PRIu32 " received!", sub_size);
      |             ^~~~~~~~~~~~~

```

Please let me know in case there is some more information I can provide. Could it be that I am not incorporating your changes correctly? I edited the `CMakeList.txt` files in `src/Universal_Robots_ROS_Driver/ur_calibration/` and `src/Universal_Robots_ROS_Driver/ur_robot_driver/` respectively.

Tobias-Fischer, 2021-08-04:
Hi, you don't need to build the `ur_client_library` from source - instead, simply `mamba install ros-noetic-ur-client-library` and this will be pulled in, instead of being built from source. The `ur_client_library` itself needed quite a few patches, so I wouldn't recommend building from source: https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-ur-client-library.patch

nberliner, 2021-08-11:
Hi Tobias,

I did a fresh install and made sure that the `ros-noetic-ur-client-library` library is installed prior to `catkin_make`ing the UR driver. Unfortunately with the same result. The `URCL_LOG_WARN` message is coming from the `ur_client_library` inside the robostackenv, no? Looking at it now, it seems as if that is the case.

From what I can tell, it is only the `ur_calibration` that fails.

I add the output starting from `ur_calibration` below.

Thanks!

```console
[ 96%] Building CXX object Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration.cpp.o
In file included from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/shared_ptr.hpp:17,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/forwards.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/common.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/ros.h:43,
                 from /home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:31,
                 from /home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration.cpp:28:
/home/niklas/miniconda3/envs/robostackenv/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
[ 97%] Building CXX object Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration_consumer.cpp.o
[ 98%] Building CXX object Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o
In file included from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/shared_ptr.hpp:17,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/forwards.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/common.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/ros.h:43,
                 from /home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:31,
                 from /home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration_consumer.h:34,
                 from /home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_consumer.cpp:28:
/home/niklas/miniconda3/envs/robostackenv/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /home/niklas/miniconda3/envs/robostackenv/include/boost/shared_ptr.hpp:17,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/forwards.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/common.h:37,
                 from /home/niklas/miniconda3/envs/robostackenv/include/ros/ros.h:43,
[... cut ...]
make[2]: *** [Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/build.make:104: Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7819: Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target ur_robot_driver_node
make: *** [Makefile:146: all] Error 2
Invoking "make -j4 -l4" failed
```

Tobias-Fischer, 2021-08-11:
Hi,

I did a bit of digging and found this: https://github.com/cocotb/cocotb/issues/1451

Could you please add `add_definitions(-D__STDC_FORMAT_MACROS)` somewhere near the top of the `/home/niklas/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/CMakeLists.txt`? That should solve your issues.

nberliner, 2021-08-11:
Magic! That fixes the error. Thanks again!!

closed 2021-08-11 by nberliner

Tobias-Fischer, 2021-08-11:
Great! It would be nice if you could open a PR for ur_calibration so the issue doesn't pop up for others in the future :)

wolfv, 2021-08-12:
This could also be something to patch in our catkin build so that it's applied to all ROS builds