RoboStack/ros-noetic #156: Package unavailable MacOS ARM64

momoai96, 2021-08-10:
Hey guys,

I followed your installation guide and set up a conda environment. 
I'm currently working on macOS Big Sur 11.5.1 Arm 64 with Apple Silicon. 

After setting up channels, I tried : "conda install ros-noetic-desktop", but received only an error: 

"PackagesNotFoundError: The following packages are not available from current channels:
  - ros-noetic-desktop"

What am I doing wrong?

Tobias-Fischer, 2021-08-10:
Hi @momoai96, many thanks for your interest in RoboStack! Unfortunately ros-noetic-desktop is not yet available for Apple Silicon - please refer to [this table](https://robostack.github.io/noetic.html) for available packages. You can still run the "standard" osx64 packages via Rosetta though, including ros-noetic-desktop.

We are currently stuck with some cross-compilation issues - if you are interested, have a look [here](https://github.com/mamba-org/boa/issues/158). @wolfv - do you have any suggestions how to get around the issue? Do we have an osx-arm64 machine for conda-forge that we could use to build packages natively?

momoai96, 2021-08-10:
Thanks for your fast answering! 
What are the steps to install Ros noetic with Rosetta?

Tobias-Fischer, 2021-08-11:
Hi, please refer to [this issue](https://github.com/conda-forge/miniforge/issues/165) to create a conda environment that uses Intel packages. Let us know if it works :)

waasnipun, 2021-09-21:
@momoai96 Did it work? I am still getting the same error even after switching to intel. Any help?

Tobias-Fischer, 2021-09-21:
Please also provide the command/steps that you used. It's very hard to debug the issue with just an error message.

momoai96, 2021-09-21:
@waasnipun 

yes, it worked really well!
No problems!

Just create a conda env based on rosetta to emulate osx-64. 
Afterwards you can install ros noetic. 
Run Roscore in a new terminal to work as usual!

waasnipun, 2021-09-24:
**First I ran this to emulate osx-64**

> CONDA_SUBDIR=osx-64 conda create -n robostackenv python=3.8  
conda env config vars set CONDA_SUBDIR=osx-64  
conda deactivate robostackenv
conda activate robostackenv
echo "CONDA_SUBDIR: $CONDA_SUBDIR" 

**Then ran**

> conda config --env --add channels conda-forge
conda config --env --add channels robostack
conda config --env --set channel_priority strict

> conda install ros-noetic-desktop

**This was the error**

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - ros-noetic-desktop

Current channels:

  - https://conda.anaconda.org/robostack/osx-arm64
  - https://conda.anaconda.org/robostack/noarch
  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

Tobias-Fischer, 2021-09-26:
I think you need to call `conda env config vars set CONDA_SUBDIR=osx-64` after `conda activate robostackenv`. At the moment, it is still looking for packages in the `osx-arm64` subdirectory.

In worst case, you can prepend `CONDA_SUBDIR=osx-64` to call conda/mamba calls (we really suggest using mamba instead of conda).

waasnipun, 2021-09-26:
`CONDA_SUBDIR=osx-64 mamba install ros-noetic-desktop` worked. Thanks

Tobias-Fischer, 2022-01-21:
Just an update: on ROS2 Galactic, we now have ros-galactic-desktop natively for the M1. If there is interest to get the same working on Noetic, I could point a volunteer to some relevant commits & patches to get the same working for Noetic.

Tobias-Fischer, 2022-02-01:
We now have a significant chuck of packages. Note that https://github.com/RoboStack/ros-galactic/issues/73 is still an issue.

sylarevan, 2022-02-01:
Hi there.

I'm trying to install ROS Noetic on my mac M1 (12.2). So far no success, as it seems to always miss a package during the installation of the `ros-noetic-desktop` package.
- with python=3.9: 
```
Encountered problems while solving:
  - package ros-noetic-desktop-1.5.0-py39heb88ffd_15 requires ros-noetic-viz, but none of the providers can be installed
```

- with python=3.8: 
```
Encountered problems while solving:
  - package ros-noetic-desktop-1.5.0-py38hb43b470_13 requires ros-noetic-joint-state-publisher-gui, but none of the providers can be installed
```

The environment is correctly configured to use Rosetta as per [the link](https://github.com/conda-forge/miniforge/issues/165) mentioned above.

Any idea on how to solve this? Thanks!

Tobias-Fischer, 2022-02-01:
Could you please specify the exact commands you are running so it's easier to reproduce the issue?

Tobias-Fischer, 2022-02-02:
Packages are now available natively on osx-arm64. Simply `mamba install ros-noetic-desktop` (without the CONDA_SUBDIR stuff) should work. Let me know.

closed 2022-02-02 by Tobias-Fischer

sylarevan, 2022-02-02:
Hi @Tobias-Fischer. I've switched to the osx-64 architecture since the package `ros-noetic-viz` was missing when installing `ros-noetic-desktop` yesterday with arm64 architecture. I can confirm this not the case anymore today!
Anyway, thanks!