RoboStack/ros-noetic #157: Add add_definitions(-D__STDC_FORMAT_MACROS) to our catkin

wolfv, 2021-08-12:
We could (globally) fix the problem with `add_definitions(-D__STDC_FORMAT_MACROS)` by adding it to our catkin patches so that it's set for all builds involving catkin under Linux.

nberliner, 2021-08-13:
I've opened a PR for now to have the fix in the UR repo as you suggested (see #155). It would be great though if it would be possible to fix that globally.

Thanks again

danzimmerman, 2022-07-12:
> Do we understand why this doesn't seem to be an issue for the ordinary ROS distribution channels?

I see now in local `vinca`/`boa` work that that this seems to be specified globally in the recipe build scripts...

Tobias-Fischer, 2022-07-12:
Indeed: https://github.com/RoboStack/vinca/blob/ae9e8fbbb7705925c289af372124ad621f295f6c/vinca/templates/build_catkin.sh.in#L51-L52

wolfv, 2022-07-12:
I think that the idea was to add this to the catkin package, though, to enable it globally in all CMake scripts based on catkin @Tobias-Fischer

wolfv, 2022-07-12:
(not just when using vinca & boa)

closed 2024-01-26 by Tobias-Fischer