RoboStack/ros-noetic #158: turtlesim_node window doesn't come out

vpegasus, 2021-08-24:
I installed  ros-noetic on macOS 10.15.1 (Catalina) successfully by following:  https://www.youtube.com/watch?v=zF7Pbq4Puvg&t=322s
When i use the command(with roscore) : `rosrun turtlesim turtlesim_node`; info appeared normally:
```
[ INFO] [1629808285.458262000]: Starting turtlesim with node name /turtlesim
[ INFO] [1629808285.463414000]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```
However, the window of turtle doesn't come out

Can someone help me?
 thx in advance!

Tobias-Fischer, 2021-08-24:
Thanks for reporting this bug @vpegasus! I just rebuild the package; please give it an hour or so to appear on the server. Then, please run `mamba update ros-noetic-turtlesim` which should update the package to the new version, and try again.

/cc @wolfv - is there someone in conda-forge who might be able to help us out with this? Presumably this will happen on conda-forge at some stage as well, as soon as the minimum versions are increased ..

vpegasus, 2021-08-25:
@Tobias-Fischer  Thanks!
I use `conda update ros-noetic-turtlesim` to update turtlesim ( I didn't use `mamba`, is that ok?).
But the problem is still here.

Tobias-Fischer, 2021-08-25:
Can you please show the output of `conda list`?

vpegasus, 2021-08-25:
ok, it shows below:

```
# packages in environment at /opt/anaconda3/envs/ROS:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        40.1.1               h694c41f_1    conda-forge
appnope                   0.1.2            py38h50d1736_1    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.0.1                h1224e73_6    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.0            py38h96a0964_1    conda-forge
boost                     1.74.0           py38h692b87f_3    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
bullet                    2.89             py38h721d79d_2    conda-forge
bullet-cpp                3.09                 hbd83c44_2    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.17.2               h0d85af4_0    conda-forge
c-compiler                1.2.0                h0d85af4_0    conda-forge
ca-certificates           2021.5.30            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
catkin_tools              0.7.1              pyhd8ed1ab_0    conda-forge
cctools                   949.0.1             hd9211c8_21    conda-forge
cctools_osx-64            949.0.1             h6407bdd_21    conda-forge
certifi                   2021.5.30        py38h50d1736_0    conda-forge
cffi                      1.14.6           py38h9688ba1_0    conda-forge
cfitsio                   3.470                h01dc385_7    conda-forge
clang                     11.1.0               h694c41f_1    conda-forge
clang-11                  11.1.0          default_he082bbe_1    conda-forge
clang_osx-64              11.1.0               hb91bd55_3    conda-forge
clangxx                   11.1.0          default_he082bbe_1    conda-forge
clangxx_osx-64            11.1.0               h7e1b574_3    conda-forge
cmake                     3.21.1               hdbd2f3e_0    conda-forge
compiler-rt               11.1.0               h654b07c_0    conda-forge
compiler-rt_osx-64        11.1.0               h8c5fa43_0    conda-forge
compilers                 1.2.0                h694c41f_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
cppzmq                    4.7.1                hf09c4ce_2    conda-forge
cryptography              3.4.7            py38h1fa4640_0    conda-forge
curl                      7.78.0               hb861fe1_0    conda-forge
cxx-compiler              1.2.0                h940c156_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dartsim                   6.10.1               he0c73e0_0    conda-forge
dbus                      1.13.6               ha13b53f_2    conda-forge
decorator                 5.0.9              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.17.1           py38h50d1736_0    conda-forge
eigen                     3.3.9                h926bf3e_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.9                h0d85af4_0    conda-forge
expat                     2.4.1                he49afe7_0    conda-forge
fcl                       0.6.1                h31ff308_3    conda-forge
ffmpeg                    4.3.2                h4dad6da_0    conda-forge
flann                     1.9.1             h4541d04_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.2.0                h6818c3e_0    conda-forge
freeimage                 3.18.0               h7d96ac2_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
freexl                    1.0.6                h0d85af4_0    conda-forge
[... 495 lines cut ...]
xorg-libxfixes            5.0.3             h01d97ff_1004    conda-forge
xorg-libxi                1.7.10               h01d97ff_0    conda-forge
xorg-libxinerama          1.1.4             he49afe7_1001    conda-forge
xorg-libxmu               1.1.3                h0b31af3_0    conda-forge
xorg-libxpm               3.5.13               h0b31af3_0    conda-forge
xorg-libxrandr            1.5.2                h0b31af3_1    conda-forge
xorg-libxrender           0.9.10            h01d97ff_1002    conda-forge
xorg-libxt                1.1.5             h01d97ff_1003    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                haf1e3a3_0    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                h1c7c35f_0    conda-forge
zlib                      1.2.11            h7795811_1010    conda-forge
zstd                      1.5.0                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge

```

Tobias-Fischer, 2021-08-25:
Hmmm .. I just checked locally and it does not work for me either. Any ideas @wolfv @traversaro? I already rebuilt with the old macos_min_version ..

traversaro, 2021-08-25:
I guess that other Qt/ROS applications work fine?

vpegasus, 2021-08-25:
@traversaro  according to my tests, windows for gazebo, rviz and rqt_graph can pop up normally.

Tobias-Fischer, 2021-08-31:
So I had a brief look at this - the issue is that the timer is not being started:
https://github.com/ros/ros_tutorials/blob/noetic-devel/turtlesim/src/turtle_frame.cpp#L58-L60

And thus everything falls apart. Not sure why though. The issue can be replicated in a catkin_ws.

GwilymNewton, 2021-09-13:
Just wanted to add a +1 to this, seeing the same issue. Let me know if I can give you any debug etc... to help.

Tobias-Fischer, 2021-09-13:
@GwilymNewton: Any debug help would be greatly appreciated! I think the best way would be to reduce the turtlesim code to a minimum failing example - getting rid of all the things we don't need to demonstrate the problem. If you create a fork for that, this would greatly help. As I said, debugging can be done in a catkin_ws as the problem happens then as well, this should be easier to debug. If you have any other ideas how to tackle the issues, please go ahead :)

GwilymNewton, 2021-09-13:
I can try! I was working my way through the ros tutorials when I hit the bug, so I dont really know what I am doing yet.

I was more wondering if there was a verbose version of rosrun I could give you the output from. Currently all I am seeing is:
```
[ INFO] [1629798206.783502000]: Starting turtlesim with node name /turtlesim
[ INFO] [1629798206.787438000]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
^C
```

So what you need me to do it fork the code over at:
https://github.com/ros/ros_tutorials/blob/noetic-devel/turtlesim/src/turtle_frame.cpp#L58-L60

try and reduce it down to minimum failing code, and then update you? I can probably do that, but not sure how to re-make code yet. I will see if I can work it out.

Tobias-Fischer, 2021-09-13:
Yes, that'd be great. Somewhere in the ROS tutorials you'll stumble over creating your own ROS package in a catkin workspace. Basically you would consider turtlesim as your ROS package ;).

oztahmet, 2021-10-06:
Hey, any news?

LowPower-Center, 2021-11-23:
I installed ros on my bigsur 11.6 . I'm having the same problem.And rqt rqt_graph didn't response(I don't how many tools can't run).How can I solve it?

Tobias-Fischer, 2022-01-28:
No updates I'm afraid - and not sure how to debug it either. If anyone has even a vague idea let us know

Tobias-Fischer, 2022-02-01:
Fixed in the latest builds by using a newer version of qt. Note that the rebuild is still ongoing though and there might be other issues by switching to the newer version.

closed 2022-02-01 by Tobias-Fischer

ibkvictor, 2022-02-08:
So, do we update just qt and how can I go about this. I use mamba-conda.

Hardik1306, 2022-10-04:
Run rosrun turtlesim turtlesim_node
in new terminal this will open gui