RoboStack/ros-noetic #159: pcl 1.11

baszalmstra, 2021-09-02:
The `ros-noetic-pcl-ros` package has a dependency on `pcl >=1.11. ..`. I ran into some issues with conversions from pcl types. pcl 1.11 switched from `boost::shared_ptr` to `std::shared_ptr`. ([See this post](https://answers.ros.org/question/358735/pcl-version-111-subscribe-to-pointcloud-fails/)) I was wondering why we didn't run into this issue before and figured out that that's because on ubuntu the latest version for `libpcl-dev` is still `1.10`.

Because `ros-noetic-pcl-ros` has a minimum requirement of `1.11` I also cant add a manual constraint for `1.10`. So my question is, why is there a minimum constraint on `pcl 1.11` while all ubuntu based installations work perfectly with `1.10`? Could we lower this constraint possibly?

Tobias-Fischer, 2021-09-02:
Hi @baszalmstra,
There are a few reasons why we opted for `pcl 1.11`. Firstly, `pcl 1.10` is not available on conda-forge (see https://anaconda.org/conda-forge/pcl/files); only 1.8, 1.9. 1.11 (and recently 1.12). So we needed to decide whether to go for 1.11 or 1.9. One of the ideas behind RoboStack is to be able to use more recent libraries than what is available in Ubuntu, so we opted for 1.11.

The changes required to get code to work with PCL 1.11 are relatively minor; see e.g. https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-laser-ortho-projector.patch

Do you have a particular issue with some package?

baszalmstra, 2021-09-06:
Nope! Thanks! That was the approach I took. I was mostly wondering about the reason behind the constraint.

closed 2021-09-06 by baszalmstra